mr-manuel/venus-os_dbus-serialbattery #172: Starts, stops, starts stops.. I am not skilled enough to figure this out.

kenac99, 2025-01-26:
### Describe the problem

the monitor starts, runs for a bit then stops, restarts, and so on.

### Driver version of the currently installed driver

1.5.20241119beta

### Driver version of the last known working driver

None.

### Venus OS device type

Raspberry Pi

### Venus OS version

v3.53

### BMS type

JKBMS (Heltec BMS)

### Cell count

7

### Battery count

1

### Connection type

Bluetooth

### Config file

```ini
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):
; MAX_BATTERY_CHARGE_CURRENT = 50.0
; MAX_BATTERY_DISCHARGE_CURRENT = 60.0

BLUETOOTH_BMS = Jkbms_Ble 98:DA:20:06:40:4F Fence BMS
MIN_CELL_VOLTAGE   = 2.900
MAX_CELL_VOLTAGE   = 4.20
FLOAT_CELL_VOLTAGE = 4.20
SOC_RESET_VOLTAGE = 4.199
EXTERNAL_CURRENT_SENSOR_DBUS_DEVICE = com.victronenergy.battery.ttyS2
EXTERNAL_CURRENT_SENSOR_DBUS_PATH = /Dc/0/Current
CVCM_ENABLE = True
CVL_ICONTROLLER_MODE   = True
CVL_ICONTROLLER_FACTOR = 0.2
CELL_VOLTAGES_WHILE_CHARGING	  = 4.20, 4.15, 4.10, 4.05
```

### Relevant log output

```shell
@40000000679598d31878c34c INFO:SerialBattery:
@40000000679598d3187cd644 INFO:SerialBattery:Starting dbus-serialbattery
@40000000679598d3188ce3a4 INFO:SerialBattery:Venus OS v3.53
@40000000679598d318910254 INFO:SerialBattery:dbus-serialbattery v1.5.20241119beta
@40000000679598e3264464f4 INFO:SerialBattery:Init of Jkbms_Ble at 98:DA:20:06:40:4F
@40000000679598e32648e934 INFO:SerialBattery:Test of Jkbms_Ble at 98:DA:20:06:40:4F
@40000000679598e808ef123c INFO:SerialBattery:BAT: JKBMS 15A 7S (20241126)
@40000000679598e808f97a4c INFO:SerialBattery:-- Connection established to Jkbms_Ble
@40000000679598e81238f484 INFO:SerialBattery:Reconnected to previously identified battery with DeviceInstance: 1
@40000000679598e81a46110c INFO:SerialBattery:Use DeviceInstance: 1
@40000000679598e81a59a0dc INFO:SerialBattery:Use dbus ServiceName: com.victronenergy.battery.ble_98da2006404f
@40000000679598e81bd3c58c INFO:SerialBattery:Battery JKBMS BLE connected to dbus from ble_98da2006404f
@40000000679598e81bd3dcfc INFO:SerialBattery:========== Settings ==========
@40000000679598e81bd3ec9c INFO:SerialBattery:> Connection voltage: 28.94 V | Current: -0.6 A | SoC: 99%
@40000000679598e81bd3f854 INFO:SerialBattery:> Cell count: 7 | Cells populated: 7
@40000000679598e81bd4040c INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
@40000000679598e81bd582dc INFO:SerialBattery:> MIN CELL VOLTAGE: 2.900 V | MAX CELL VOLTAGE: 4.200 V| FLOAT CELL VOLTAGE: 4.200 V
@40000000679598e81bdb5af4 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 50.0 A | MAX BATTERY DISCHARGE CURRENT: 60.0 A
@40000000679598e81be073a4 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 40.0 A | MAX BATTERY DISCHARGE CURRENT: 100.0 A (read from BMS)
@40000000679598e81be5133c INFO:SerialBattery:> CVCM:     True
@40000000679598e81be9ed6c INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
@40000000679598e81beeac44 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
@40000000679598e81bf3d494 INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
@40000000679598e81bf8c634 INFO:SerialBattery:> CHARGE FET: True | DISCHARGE FET: True | BALANCE FET: True
@40000000679598e81bfdb7d4 INFO:SerialBattery:Serial Number/Unique Identifier: 4092343037
@40000000679598e81c066e4c INFO:SerialBattery:Monitoring external current using: com.victronenergy.battery.ttyS2/Dc/0/Current
@40000000679598ea27b4f09c ./run: line 12:  5665 Segmentation fault      python /opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py Jkbms_Ble 98:DA:20:06:40:4F <-- Maybe?
@40000000679598ea2c22be84
@40000000679598ea2c24490c
@40000000679598ea2c2454c4 INFO:Preparing Bluetooth for connection to BMS
@40000000679598ea2c2615fc INFO:Bluetooth details
@40000000679598ea2d381e54 [CHG] Device 98:DA:20:06:40:4F RSSI: -59
@40000000679598ea2d3831dc [CHG] Device 98:DA:20:06:40:4F RSSI: -77
@40000000679598ee1b8df28c Attempting to disconnect from 98:DA:20:06:40:4F
@40000000679598ee1b8e09fc [CHG] Device C6:32:38:2E:B6:8F RSSI is nil
@40000000679598ee1b8e15b4 [DEL] Device C6:32:38:2E:B6:8F Fence
@40000000679598ee1b8e216c [CHG] Device F9:0E:DA:2C:39:C8 RSSI is nil
@40000000679598ee1b8e310c [DEL] Device F9:0E:DA:2C:39:C8 Fence
@40000000679598ee1b8e3cc4 [CHG] Device D1:9A:3A:68:7B:A0 RSSI is nil
@40000000679598ee1b8e487c [DEL] Device D1:9A:3A:68:7B:A0 B Shunt
```

### Any other information that may be helpful

It works for a moment then stops, reloads then starts the cycle again.

I am at a loss.

Thanks for whatever assistance you can offer.

Ken Compton
kacompton@radiation.tv

kenac99, 2025-01-26:
[image]

mr-manuel, 2025-01-26:
https://github.com/mr-manuel/venus-os_dbus-serialbattery/issues/86

closed 2025-01-26 by mr-manuel