mr-manuel/venus-os_dbus-serialbattery #179: End of charge / 100% SoC behaviour, system keeps charging/discharging

Bart-1992, 2025-02-05:
### Describe the problem

First of all thank you very much @mr-manuel for all your work and effort in the Victron integrations. 

When my system is at 100% SoC an erratic discharging/charging control loops starts to occur when in absorption charging, There is plenty of solar feedback into the grid available, thus discharging is absolutely uneccesary. 

See example here: 

https://github.com/user-attachments/assets/04c6711d-73d9-425c-8ca1-6993ae45c336

The charge/discharge currents are very low, between 0.8/2 amps. I would expect a constant constant absorption charge
I would expect a very low charging current, and an eventual 'idle' state where no charging is happening anymore or when the absorption time limit has exceeded ( victron setting ). However this is not the case, it keeps toggling between charge/discharge until there is no solar available anymore and permanent load is put on the system. 

I could find quite a lot of related issues/topics on the internet, but all without any solution or clear answer on how to prevent this behaviour. Or is this behaviour intended from the ESS system?

For example some related topic : 

https://github.com/Louisvdw/dbus-serialbattery/discussions/732
https://community.victronenergy.com/t/ess-continous-switching-between-charge-discharge-when-battery-full-and-negative-grid-power/983/15
https://communityarchive.victronenergy.com/questions/265899/multiplus-jumps-from-charge-to-discharge.html#comment-266436

I hope you have an answer to this behaviour, do I need to set some additional parameters? 

Currently these are my JKBMS configs ; 

[image]

My Victron MP II settings are : 

[image]

[image]

### Driver version of the currently installed driver

V1.6.20250123

### Venus OS device type

Raspberry Pi

### Venus OS version

v3.54

### BMS type

JKBMS (Heltec BMS)

### Cell count

16

### Battery count

1

### Connection type

Serial USB adapter to TTL

### Config file

```ini
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):
 MAX_BATTERY_CHARGE_CURRENT = 35.0
 MAX_BATTERY_DISCHARGE_CURRENT = 50.0

BMS_TYPE = Jkbms

; --------- Cell Voltages ---------
; Description:
;     Cell min/max voltages which are used to calculate the min/max battery voltage
; Example:
;     16 cells * 3.45V/cell = 55.2V max charge voltage. 16 cells * 2.90V = 46.4V min discharge voltage
MIN_CELL_VOLTAGE   = 2.95
; Max voltage (can seen as absorption voltage)
MAX_CELL_VOLTAGE   = 3.45
; Float voltage (can be seen as resting voltage)
FLOAT_CELL_VOLTAGE = 3.400

; --------- SOC reset voltage ---------
; Description:
;     May be needed to reset the SoC to 100% once in a while for some BMS, because of SoC drift.
;     Specify the cell voltage where the SoC should be reset to 100% by the BMS.
;       - JKBMS: SoC is reset to 100% if one cell reaches OVP (over voltage protection) voltage
;     As you have to adopt this value to your system, I reccomend to start with
;     OVP voltage - 0.030 (see Example).
;       - Try to increase (add) by 0.005 in steps, if the system does not switch to float mode, even if
;         the target voltage SOC_RESET_VOLTAGE * CELL_COUNT is reached.
;       - Try to decrease (lower) by 0.005 in steps, if the system hits the OVP too fast, before all
;         cells could be balanced and the system goes into protection mode multiple times.
; Example:
;     If OVP is 3.650, then start with 3.620 and increase/decrease by 0.005
; Note:
;     The value has to be higher as the MAX_CELL_VOLTAGE
;     You also have to set CELL_VOLTAGES_WHILE_CHARGING accordingly, if you set CCCM_CV_ENABLE to true
;         else the charging current will be reduced to 0 before the target voltage is reached and the
;         battery will never switch to float
SOC_RESET_VOLTAGE = 3.52

; --------- Charge Voltage limitation (affecting CVL) ---------
; Description:
;     Limit max charging voltage (MAX_CELL_VOLTAGE * cell count), switch from max voltage to float
;     voltage (FLOAT_CELL_VOLTAGE * cell count) and back
;     False: Max charging voltage is always kept
;     True: Max charging voltage is reduced based on charge mode
;         Step mode: After max voltage is reached for MAX_VOLTAGE_TIME_SEC it switches to float voltage. After
;             SoC is below SOC_LEVEL_TO_RESET_VOLTAGE_LIMIT it switches back to max voltage.
;         Linear mode: After max voltage is reachend and cell voltage difference is smaller or equal to
;             CELL_VOLTAGE_DIFF_KEEP_MAX_VOLTAGE_UNTIL it switches to float voltage after MAX_VOLTAGE_TIME_SEC
;             additional seconds.
;             After cell voltage difference is greater or equal to CELL_VOLTAGE_DIFF_TO_RESET_VOLTAGE_LIMIT
;             OR
;             SoC is below SOC_LEVEL_TO_RESET_VOLTAGE_LIMIT
;             it switches back to max voltage.
; Example when set to True:
;     Step mode:
;          The battery reached max voltage of 55.2V and hold it for 900 seconds, the the CVL is switched to
;          float voltage of 53.6V to don't stress the batteries. Allow max voltage of 55.2V again, if SoC is
;          once below 80%
;     Linear mode:
;          The battery reached max voltage of 55.2V and the max cell difference is 0.010V, then switch to float
;          voltage of 53.6V after 900 additional seconds to don't stress the batteries. Allow max voltage of
;          55.2V again if max cell difference is above 0.080V or SoC below 80%.
; Charge voltage control management enable (True/False).
CVCM_ENABLE = True

; -- CVL reset based on cell voltage diff (linear mode)
; Specify cell voltage diff where CVL limit is kept until diff is equal or lower
CELL_VOLTAGE_DIFF_KEEP_MAX_VOLTAGE_UNTIL        = 0.010
; Specify cell voltage diff where MAX_VOLTAGE_TIME_SEC restarts if diff is bigger
CELL_VOLTAGE_DIFF_KEEP_MAX_VOLTAGE_TIME_RESTART = 0.013
; Specify cell voltage diff where CVL limit is reset to max voltage, if value get above
; the cells are considered as imbalanced, if the cell diff exceeds 5% of the nominal cell voltage
; e.g. 3.2 V * 5 / 100 = 0.160 V
CELL_VOLTAGE_DIFF_TO_RESET_VOLTAGE_LIMIT        = 0.080

; -- CVL reset based on SoC option (step mode & linear mode)
; Specify how long the max voltage should be kept
;     Step mode: If reached then switch to float voltage
;     Linear mode: If cells are balanced keep max voltage for further MAX_VOLTAGE_TIME_SEC seconds
MAX_VOLTAGE_TIME_SEC = 900

; --------- Cell Voltage Current limitation (affecting CCL/DCL) ---------
; Description: Maximal charge / discharge current will be in-/decreased depending on min and max cell voltages
; Example:
;     18 cells * 3.55V/cell = 63.9V max charge voltage
;     18 cells * 2.70V/cell = 48.6V min discharge voltage
;     But in reality not all cells reach the same voltage at the same time. The (dis)charge current
;     will be (in-/)decreased, if even ONE SINGLE BATTERY CELL reaches the limits

; Charge current control management referring to cell-voltage enable (True/False).
CCCM_CV_ENABLE = False
; Discharge current control management referring to cell-voltage enable (True/False).
DCCM_CV_ENABLE = False

; Set steps to reduce battery current
; The current will be changed linear between those steps if LINEAR_LIMITATION_ENABLE is set to True
CELL_VOLTAGES_WHILE_CHARGING      = 3.45, 3.40, 3.35, 3.30
MAX_CHARGE_CURRENT_CV_FRACTION    =    0, 0.05,  0.5,    1

CELL_VOLTAGES_WHILE_DISCHARGING   = 2.70, 2.80, 2.90, 3.10
MAX_DISCHARGE_CURRENT_CV_FRACTION =    0,  0.1,  0.5,    1
```

### Relevant log output

```shell
2025-02-03 16:09:54.692203500 INFO:SerialBattery:Battery JKBMS connected to dbus from /dev/ttyUSB0
2025-02-03 16:09:54.692207500 INFO:SerialBattery:========== Settings ==========
2025-02-03 16:09:54.692210500 INFO:SerialBattery:> Connection voltage: 53.89 V | Current: -3.18 A | SoC: 100%
2025-02-03 16:09:54.692213500 INFO:SerialBattery:> Cell count: 16 | Cells populated: 16
2025-02-03 16:09:54.692217500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2025-02-03 16:09:54.692220500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.950 V | MAX CELL VOLTAGE: 3.450 V| FLOAT CELL VOLTAGE: 3.400 V
2025-02-03 16:09:54.692282500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 35.0 A | MAX BATTERY DISCHARGE CURRENT: 50.0 A
2025-02-03 16:09:54.692286500 INFO:SerialBattery:> CVCM:     True
2025-02-03 16:09:54.692289500 INFO:SerialBattery:> CCCM CV:  False | DCCM CV:  False
2025-02-03 16:09:54.692292500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2025-02-03 16:09:54.692295500 INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
2025-02-03 16:09:54.692392500 INFO:SerialBattery:> CHARGE FET: True | DISCHARGE FET: True | BALANCE FET: True
2025-02-03 16:09:54.692397500 INFO:SerialBattery:Serial Number/Unique Identifier: 409232A1219
```

onemorename, 2025-02-05:
Same here, noticed today when the sun was finally shining again after a long time.

mr-manuel, 2025-02-05:
This is the default behavior of an ESS system, when the battery is full. If you want to know more details, then your question fits better to the Victron community.

closed 2025-02-05 by mr-manuel