mr-manuel/venus-os_dbus-serialbattery #182: When multiple serial batteries are defined in coexistence with one CAN battery, the CAN appears twice.

rogergrant99, 2025-02-08:
### Describe the problem

When there are multiple devices defined by the BATTERY_ADDRESSES =  in config.ini the same applies to the CAN portion of the detection code, therefore if two serial batteries are defined and one CAN the CAN appears twice.

This diff code fixes this issue.

[multiple-canbus-batteries.txt](https://github.com/user-attachments/files/18722530/multiple-canbus-batteries.txt)

### Driver version of the currently installed driver

v2.0.20250207beta

### Venus OS device type

Cerbo GX

### Venus OS version

v3.54

### BMS type

JKBMS Inverter (JKBMS PB Model)

### Cell count

4

### Battery count

3

### Connection type

Serial USB adapter to RS485

### Config file

```ini
[DEFAULT]

; --------- Set logging level ---------
; ERROR: Only errors are logged
; WARNING: Errors and warnings are logged
; INFO: Errors, warnings, and info messages are logged
; DEBUG: Errors, warnings, info, and debug messages are logged
LOGGING = INFO

; --------- Battery Current Limits ---------
; +++ Limits apply to each individual battery/BMS. +++
; +++ If you have multiple batteries, you need a battery aggregator. +++
MAX_BATTERY_CHARGE_CURRENT    = 100.0
MAX_BATTERY_DISCHARGE_CURRENT = 100.0

; --------- Cell Voltages ---------
; Description:
;     Minimum and maximum cell voltages used for:
;         - Calculating the minimum and maximum battery voltage
;         - Triggering the SoC reset when SoC calculation is enabled
; Example:
;     16 cells * 3.45 V/cell = 55.2 V max charge voltage
;     16 cells * 2.90 V/cell = 46.4 V min discharge voltage
; Minimum voltage (can be seen as cut-off voltage)
MIN_CELL_VOLTAGE   = 2.800
; Maximum voltage (can be seen as absorption voltage)
MAX_CELL_VOLTAGE   = 3.600
; Float voltage (can be seen as resting voltage)
;FLOAT_CELL_VOLTAGE = 3.375

; --------- SoC Reset Voltage (must match BMS settings) ---------
[... 450 lines cut ...]
INVERT_CURRENT_MEASUREMENT = 1

; -- JKBMS CAN settings
; Predefined cell count for Jkbms_can.
; The cell count should be auto-detected by identifying the highest cell number,
; but this process may be slow, causing cell voltage updates to be delayed in VenusOS.
; Use this workaround if you experience problems with cell voltage updates.
JKBMS_CAN_CELL_COUNT = 4

; -- ESC GreenMeter and Lipro device settings
GREENMETER_ADDRESS  = 1
LIPRO_START_ADDRESS = 2
LIPRO_END_ADDRESS   = 4
LIPRO_CELL_COUNT    = 15

; -- Seplos V3 settings
; Use min/max cell voltage, CVL, CCL, and DCL values from the BMS.
SEPLOS_USE_BMS_VALUES = False
```

### Relevant log output

```shell
NONE
```

mr-manuel, 2025-02-09:
Please describe more detailed how you connected the batteries (one per adapter or daisy chain), what device addresses you gave them (app and dip switches) and post the log of all batteries.

rogergrant99, 2025-02-12:
I have two JKBMS connected via RS-485 daisy chained and one Renegy REGO connected to the CAN1 port of the Cerbo, I also have a wake speed connected to the CAN0 port of the Cerbo. 

For the Wakespeed to work I have to use CAN0  on the Cerbo .

I use CAN1 for the REGO but CAN1 on the Cerbo does not support RV-C, so I wrote a BMS driver based on JKBMS_Can that decodes the RV-C protocol and it works as it should if the "BATTERY_ADDRESSES"  is null.

When the existing version of the code parse the "BATTERY_ADDRESSES" constant for CAN batteries it will use the two defined JKBMS serial definition (0x01,0x02) and fail because of a duplicate serial.

Here is a copy of the log:

2025-02-12 00:51:29.722797500 INFO:SerialBattery:
2025-02-12 00:51:29.723613500 INFO:SerialBattery:Starting dbus-serialbattery
2025-02-12 00:51:29.727007500 INFO:SerialBattery:Venus OS v3.54 running on Cerbo GX
2025-02-12 00:51:29.727993500 INFO:SerialBattery:dbus-serialbattery v2.0.20250207beta
2025-02-12 00:51:30.731166500 INFO:SerialBattery:Detected CAN Bus bitrate: 250 kbps
2025-02-12 00:51:32.736211500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2025-02-12 00:51:32.737429500 INFO:SerialBattery:Testing Rego_Can at address "\x01"
2025-02-12 00:51:44.220295500 INFO:SerialBattery:-- Connection established to Rego_Can
2025-02-12 00:51:44.221132500 INFO:SerialBattery:Successful battery connection at can1 and this address 0x01
2025-02-12 00:51:44.222573500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2025-02-12 00:51:44.223531500 INFO:SerialBattery:Testing Rego_Can at address "\x02"
2025-02-12 00:51:55.710779500 INFO:SerialBattery:-- Connection established to Rego_Can
2025-02-12 00:51:55.712622500 INFO:SerialBattery:Successful battery connection at can1 and this address 0x02
2025-02-12 00:52:06.507527500 INFO:SerialBattery:Reconnected to previously identified battery with DeviceInstance: 83
2025-02-12 00:52:07.232161500 INFO:SerialBattery:Use DeviceInstance: 83
2025-02-12 00:52:07.234804500 INFO:SerialBattery:Use dbus ServiceName: com.victronenergy.battery.can1__0x01
2025-02-12 00:52:18.215306500 INFO:SerialBattery:Reconnected to previously identified battery with DeviceInstance: 83
2025-02-12 00:52:18.883390500 INFO:SerialBattery:Use DeviceInstance: 83
2025-02-12 00:52:18.885443500 ERROR:SerialBattery:** DRIVER STOPPED! Another battery with the same serial number/unique identifier "REGOCAN4S_400Ah" found! **
2025-02-12 00:52:18.886128500 ERROR:SerialBattery:Please check that the batteries have unique identifiers.
2025-02-12 00:52:18.886861500 ERROR:SerialBattery:Change the battery capacities to be unique. 

In my original post I included a diff of the code changes that made it work correctly for me. I understand that when two batteries are found on the CAN bus the capacity of one of the two has to be increased to differentiate them but the root of the issue is when one CAN battery is defined while there are two serial batteries on the same serial port the single CAN battery will show up twice.

mr-manuel, 2025-02-13:
> I wrote a BMS driver based on JKBMS_Can

Better would be, if you fix your driver to use correctly the battery address, so it does not recognize two batteries. You cannot use a fixed recognition pattern.

I won't change the whole driver for this.

closed 2025-02-14 by mr-manuel

rogergrant99, 2025-02-20:
After trying to figure out how I could modify the driver to only show one battery I realized that CAN does not have battery identifier, so if you have multiple CAN batteries they have to support a Master/Slave protocol, so it seems that CAN by design can have only one BMS.... That's why in the existing code the logs show:

ERROR:SerialBattery:Please check that the batteries have unique identifiers.
ERROR:SerialBattery:Change the battery capacities to be unique.
ERROR:SerialBattery:Example for batteries with 280 Ah:
ERROR:SerialBattery:- Battery 1: 279 Ah
ERROR:SerialBattery:- Battery 2: 280 Ah
ERROR:SerialBattery:- Battery 3: 281 Ah

So if the BATTERY_ADDRESSES contains more than one value, the code will call the BMS driver in succession with the listed values, as per line 384 of bus-serialbattery.py:

addresses = [None] if len(BATTERY_ADDRESSES) == 0 else BATTERY_ADDRESSES  # use default address, if not configured

So if multiple serial batteries are defined and only one CAN battery is present the problem will develop.

I will adjust the driver to have only one instance but I believe the problem would show even for a Jkbms_can battery.

rogergrant99, 2025-02-20:
I was getting confused address detection so I did some tests, here are the results, let me know your thoughts. 

1- I did a test with one JKBMS on CAN1 and two RS-485 JKBMS.

2- I set the variable "BATTERY_ADDRESSES = 0X01 , 0X02" and the two RS-485 show up correctly but the CAN does not.

3- I redid the same test with the variable  "BATTERY_ADDRESSES =  " and one RS-485 shows up and one CAN shows up. 

I include the CAN logs of both tests:

2025-02-20 23:17:15.347282500 INFO:SerialBattery:Starting dbus-serialbattery
2025-02-20 23:17:15.352850500 INFO:SerialBattery:Venus OS v3.54 running on Cerbo GX
2025-02-20 23:17:15.354540500 INFO:SerialBattery:dbus-serialbattery v2.0.20250207beta
2025-02-20 23:17:16.981767500 INFO:SerialBattery:Detected CAN Bus bitrate: 500 kbps
2025-02-20 23:17:18.987441500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2025-02-20 23:17:18.988493500 INFO:SerialBattery:Testing Jkbms_Can
2025-02-20 23:17:18.992263500 INFO:SerialBattery:-- Connection established to Jkbms_Can
2025-02-20 23:17:18.993202500 INFO:SerialBattery:Successful battery connection at can1 and this address None
2025-02-20 23:17:46.925602500 INFO:SerialBattery:Use DeviceInstance: 4
2025-02-20 23:17:46.928289500 INFO:SerialBattery:Use dbus ServiceName: com.victronenergy.battery.can1
2025-02-20 23:17:46.992202500 INFO:SerialBattery:Polling interval: 1.000 s
2025-02-20 23:17:47.005992500 INFO:SerialBattery:Battery JKBMS CAN V2 connected to dbus from can1
2025-02-20 23:17:47.006914500 INFO:SerialBattery:========== Settings ==========
2025-02-20 23:17:47.008072500 INFO:SerialBattery:> Connection voltage: 13.363000000000001 V | Current: -1.1999999999999886 A | SoC: 100% | SoC calc: 100%
2025-02-20 23:17:47.008935500 INFO:SerialBattery:> Cell count: 4 | Cells populated: 4
2025-02-20 23:17:47.010263500 INFO:SerialBattery:> CHARGE MODE: Linear
2025-02-20 23:17:47.015470500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.800 V | MAX CELL VOLTAGE: 3.550 V| FLOAT CELL VOLTAGE: 3.525 V
2025-02-20 23:17:47.022730500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 120.0 A | MAX BATTERY DISCHARGE CURRENT: 120.0 A
2025-02-20 23:17:47.031744500 INFO:SerialBattery:> CVCM:     True
2025-02-20 23:17:47.035654500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2025-02-20 23:17:47.036569500 INFO:SerialBattery:> CCCM T:   False | DCCM T:   False
2025-02-20 23:17:47.044166500 INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
2025-02-20 23:17:47.053687500 INFO:SerialBattery:> CHARGE FET: True | DISCHARGE FET: True | BALANCE FET: None
2025-02-20 23:17:47.054885500 INFO:SerialBattery:Serial Number/Unique Identifier: can1

2025-02-20 23:20:32.873934500 INFO:SerialBattery:Starting dbus-serialbattery
2025-02-20 23:20:32.877307500 INFO:SerialBattery:Venus OS v3.54 running on Cerbo GX
2025-02-20 23:20:32.878017500 INFO:SerialBattery:dbus-serialbattery v2.0.20250207beta
2025-02-20 23:20:34.144461500 INFO:SerialBattery:Detected CAN Bus bitrate: 500 kbps
2025-02-20 23:20:36.149075500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2025-02-20 23:20:36.150099500 INFO:SerialBattery:Testing Jkbms_Can at address "\x01"
2025-02-20 23:20:36.152529500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2025-02-20 23:20:36.654498500 INFO:SerialBattery:-- Testing BMS: 2 of 3 rounds
2025-02-20 23:20:36.655929500 INFO:SerialBattery:Testing Jkbms_Can at address "\x01"
2025-02-20 23:20:36.657670500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2025-02-20 23:20:37.159835500 INFO:SerialBattery:-- Testing BMS: 3 of 3 rounds
2025-02-20 23:20:37.160947500 INFO:SerialBattery:Testing Jkbms_Can at address "\x01"
2025-02-20 23:20:37.164294500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2025-02-20 23:20:37.666337500 WARNING:SerialBattery:No battery connection at can1 and this address 0x01
2025-02-20 23:20:37.667305500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2025-02-20 23:20:37.668116500 INFO:SerialBattery:Testing Jkbms_Can at address "\x02"
2025-02-20 23:20:37.670096500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2025-02-20 23:20:38.172037500 INFO:SerialBattery:-- Testing BMS: 2 of 3 rounds
2025-02-20 23:20:38.173129500 INFO:SerialBattery:Testing Jkbms_Can at address "\x02"
2025-02-20 23:20:38.175360500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2025-02-20 23:20:38.676960500 INFO:SerialBattery:-- Testing BMS: 3 of 3 rounds
2025-02-20 23:20:38.677892500 INFO:SerialBattery:Testing Jkbms_Can at address "\x02"
2025-02-20 23:20:38.679458500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2025-02-20 23:20:39.181390500 WARNING:SerialBattery:No battery connection at can1 and this address 0x02
2025-02-20 23:20:39.182689500 INFO:SerialBattery:Found no devices on can bus, retrying with 250 kbps
2025-02-20 23:20:39.273167500 INFO:SerialBattery:CAN Bus can1 is up with bitrate 250 kbps
2025-02-20 23:20:41.276378500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2025-02-20 23:20:41.277566500 INFO:SerialBattery:Testing Jkbms_Can at address "\x01"
2025-02-20 23:20:41.282837500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2025-02-20 23:20:41.787751500 INFO:SerialBattery:-- Testing BMS: 2 of 3 rounds
2025-02-20 23:20:41.789372500 INFO:SerialBattery:Testing Jkbms_Can at address "\x01"
2025-02-20 23:20:41.791661500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2025-02-20 23:20:42.293807500 INFO:SerialBattery:-- Testing BMS: 3 of 3 rounds
2025-02-20 23:20:42.294919500 INFO:SerialBattery:Testing Jkbms_Can at address "\x01"
[... cut ...]
2025-02-20 23:20:44.328478500 INFO:SerialBattery:Found no devices on can bus, retrying with 500 kbps
2025-02-20 23:20:44.654155500 INFO:SerialBattery:CAN Bus can1 is up with bitrate 500 kbps
2025-02-20 23:20:46.663230500 ERROR:SerialBattery:ERROR >>> No battery connection at can1 and this bus addresses: 0x01, 0x02
2025-02-20 23:20:46.663244500 INFO:SerialBattery:Exit signal received, exiting gracefully...
2025-02-20 23:20:46.663247500 INFO:SerialBattery:Stopped dbus-serialbattery with exit code 1

mr-manuel, 2025-02-21:
> Publish your code to GitHub and post some CAN frames here, then I will try to help you

rogergrant99, 2025-02-22:
The point I was trying to make on my last post was that I was using the Jkbms_Can driver and not my driver, and I got similar "BATTERY_ADDRESSES = " conflict between CAN and Serial.

I forked the project on my repo called RV-C  git@github.com:rogergrant99/venus-os_dbus-serialbattery.git

I will post frames, what format do you prefer ?

rogergrant99, 2025-02-22:
Here are some from DEBUG

2025-02-22 15:00:36.074310500 DEBUG:SerialBattery:[can1] Received: ID=0x19ffff8f, Daten=bytearray(b'\x19\x02\x16\x00\x16"\x1d\x07')
2025-02-22 15:00:36.172031500 DEBUG:SerialBattery:[can1] Received: ID=0x19fec78f, Daten=bytearray(b'\x01x\x00\x00\x00\xff\xff\xff')
2025-02-22 15:00:36.174228500 DEBUG:SerialBattery:[can1] Received: ID=0x19fea58f, Daten=bytearray(b'\x01x\x15\x90\x01\x16\x00\x00')
2025-02-22 15:00:36.188446500 DEBUG:SerialBattery:[can1] Received: ID=0x19fffb8f, Daten=bytearray(b'\x01x\xc8\\\x01\xae\xff\xff')
2025-02-22 15:00:36.192416500 DEBUG:SerialBattery:[can1] Received: ID=0x19fffc8f, Daten=bytearray(b'\x01x\xb3#\xae\x95\t\x01')
2025-02-22 15:00:36.202106500 DEBUG:SerialBattery:[can1] Received: ID=0x19fffd8f, Daten=bytearray(b'\x01x\n\x01\\\x8d5w')
2025-02-22 15:00:36.682013500 DEBUG:SerialBattery:[can1] Received: ID=0x19fffb8f, Daten=bytearray(b'\x01x\xc8\\\x01\xae\xff\xff')
2025-02-22 15:00:36.692763500 DEBUG:SerialBattery:[can1] Received: ID=0x19fffc8f, Daten=bytearray(b'\x01x\xb3#\xae\x95\t\x01')
2025-02-22 15:00:36.703389500 DEBUG:SerialBattery:[can1] Received: ID=0x19fffd8f, Daten=bytearray(b'\x01x\n\x01\x94\x8c5w')
2025-02-22 15:00:37.071601500 DEBUG:SerialBattery:[can1] Received: ID=0x19ffff8f, Daten=bytearray(b'\x19\x02\x16\x00\x16"\x1e\x07')
2025-02-22 15:00:37.172355500 DEBUG:SerialBattery:[can1] Received: ID=0x19fea58f, Daten=bytearray(b'\x01x\x15\x90\x01\x1a\x00\x00')
2025-02-22 15:00:37.183030500 DEBUG:SerialBattery:[can1] Received: ID=0x19fffb8f, Daten=bytearray(b'\x01x\xc8\\\x01\xae\xff\xff')
2025-02-22 15:00:37.191794500 DEBUG:SerialBattery:[can1] Received: ID=0x19fffc8f, Daten=bytearray(b'\x01x\xb3#\xae\x95\t\x01')
2025-02-22 15:00:37.201628500 DEBUG:SerialBattery:[can1] Received: ID=0x19fffd8f, Daten=bytearray(b'\x01x\n\x010\x8c5w')

mr-manuel, 2025-02-23:
It seems, that you used the old JKBMS_CAN driver from a `v1.x` release to create your own driver. Please use the new JKBMS_CAN driver from the `v2.x` release as template.

From the data you posted it could be that the `8f` in the CAN frame ID is the BMS ID and I would recommend to use it as such.

Frame ID's:
```
0x19fea58f
0x19fec78f
0x19fffb8f
0x19fffc8f
0x19fffd8f
0x19ffff8f
```

In the JKBMS_CAN code you see, that the `device_address` is added to the `frame_id`, which would mean, that only one battery address would match the `frame_id`. Like `0x00` matches `0x19fea58f`, `0x01` matches `0x19fea590`, `0x02` matches `0x19fea591` and so on.

https://github.com/mr-manuel/venus-os_dbus-serialbattery/blob/62caeb9fee52d54986737a7b1048aa788a3ba5fd/dbus-serialbattery/bms/jkbms_can.py#L196-L200

rogergrant99, 2025-02-23:
That's right, i started this on V1 but I upgraded to the beta version for the temperature bug I had encountered, I did not notice the change in the bms driver file. I am redoing the code based on V2 and I'm hitting a problem that I can't seem to resolve. I have all the frames defined and I extract the data in the appropriate variables but I seem to have a calculation issue and I can't really track it down. This is the error I'm getting... What are your thoughts ?

Thanks

2025-02-23 21:58:30.202507500 Traceback (most recent call last):
2025-02-23 21:58:30.202516500   File "/data/apps/dbus-serialbattery/dbus-serialbattery.py", line 505, in <module>
2025-02-23 21:58:30.202520500     main()
2025-02-23 21:58:30.202521500   File "/data/apps/dbus-serialbattery/dbus-serialbattery.py", line 462, in main
2025-02-23 21:58:30.202525500     battery[key_address].set_calculated_data()
2025-02-23 21:58:30.202527500   File "/data/apps/dbus-serialbattery/battery.py", line 1915, in set_calculated_data
2025-02-23 21:58:30.202531500     self.soc_calc = self.get_soc()
2025-02-23 21:58:30.202665500   File "/data/apps/dbus-serialbattery/battery.py", line 1902, in get_soc
2025-02-23 21:58:30.202669500     return self.soc_calculation()
2025-02-23 21:58:30.202672500   File "/data/apps/dbus-serialbattery/battery.py", line 622, in soc_calculation
2025-02-23 21:58:30.202675500     self.soc_calc_capacity_remain = self.capacity * self.soc_calc / 100 if self.soc_calc > 0 else 0
2025-02-23 21:58:30.202679500 TypeError: unsupported operand type(s) for *: 'NoneType' and 'float'

rogergrant99, 2025-02-24:
I did the changes using the Jkbms_can as a staring base and it all works fine now. 

The code should work for any RV-C house battery. So if you want to include it in the project it's fine by me. 

Thank you for your help.

mr-manuel, 2025-02-24:
Nice to hear. If you want to add it for the community and maintain it, feel free to open a PR. Here is a list of what is needed: https://mr-manuel.github.io/venus-os_dbus-serialbattery_docs/general/supported-bms#add-by-opening-a-pull-request

mr-manuel, 2025-03-01:
Is the cell type always the same? Do you have min and max cell voltages? If you have min/max cell voltages I would use them: https://github.com/mr-manuel/venus-os_dbus-serialbattery/blob/c9a621264be9f1dbac97e312def6191b57e7a9e6/dbus-serialbattery/bms/jkbms_can.py#L360

rogergrant99, 2025-03-01:
In order to make 1 cell only work I have to make min and max the same otherwise I get cell differences. Also the control_voltage does not get calculated for one cell, it appears to be the max voltage divided by 4.... 

self.cell_min_voltage = 14.2
self.cell_max_voltage = 14.2

[image]

mr-manuel, 2025-03-11:
> Is the cell type always the same?

If yes you can use the default 4, 8 and 16 cell blocks and recognize the cell count. Normally you should get at least min and max cell voltage from a battery pack with the cell ID. Strange that you don‘t have it here.

rogergrant99, 2025-03-15:
> Normally you should get at least min and max cell voltage from a battery pack with the cell ID

This Rego battery has probably 4 cells but the RV-C implementation only gives the whole battery voltage as well as max and min battery voltage, there is no provision to see the individual cell voltages, therefore I am trying to treat is a a one cell battery, if I simulate 4 cells by dividing the whole battery voltage by 4 then it all works fine, I get the problem if I give a cell count of one.

mr-manuel, 2025-03-15:
Then just provide 2 cells and treat it as a two cell battery, one is min cell voltage and one the max cell voltage.

mr-manuel, 2025-03-15:
Use min voltage for as cell 1 and the highes cell voltage for cell 4.

Then do: (`battery voltage` - `highest cell voltage` - `lowest cell voltage`) / (`cell count` - 2) -> cell voltage of all other cells between the first and last.

E.G.: (12.8 V - 3.25 V -3.15 V) / (4 - 2) = 3.2 V/cell

That way all should work.

mr-manuel, 2025-05-09:
Try to install the large image or the beta.

rogergrant99, 2025-05-11:
I tried installing the v2.0.20250324rc venus-data-tar.gz from local file, I also removed my config file and started with the defaults, all is good with my two serial jkbms batteries but when I enable the can port (only change), I then get the same error…

2025-05-11 15:26:30.818393500 INFO:SerialBattery:

2025-05-11 15:26:30.819241500 INFO:SerialBattery:Starting dbus-serialbattery

2025-05-11 15:26:30.823702500 INFO:SerialBattery:Venus OS v3.55 running on Cerbo GX

2025-05-11 15:26:30.827768500 INFO:SerialBattery:dbus-serialbattery v2.0.20250324rc

2025-05-11 15:26:31.095009500 Traceback (most recent call last):

2025-05-11 15:26:31.095017500   File "/data/apps/dbus-serialbattery/dbus-serialbattery.py", line 526, in <module>

2025-05-11 15:26:31.095021500     main()

2025-05-11 15:26:31.095022500   File "/data/apps/dbus-serialbattery/dbus-serialbattery.py", line 362, in main

2025-05-11 15:26:31.095026500     from bms.daly_can import Daly_Can

2025-05-11 15:26:31.095028500   File "/data/apps/dbus-serialbattery/bms/daly_can.py", line 24, in <module>

2025-05-11 15:26:31.095031500     from can import Message, CanOperationError

2025-05-11 15:26:31.095165500   File "/data/apps/dbus-serialbattery/ext/can/__init__.py", line 81, in <module>

2025-05-11 15:26:31.095168500     from . import util  # isort:skip

2025-05-11 15:26:31.095171500   File "/data/apps/dbus-serialbattery/ext/can/util.py", line 36, in <module>

2025-05-11 15:26:31.095174500     from .interfaces import VALID_INTERFACES

2025-05-11 15:26:31.095176500   File "/data/apps/dbus-serialbattery/ext/can/interfaces/__init__.py", line 7, in <module>

2025-05-11 15:26:31.095206500     from can._entry_points import read_entry_points

2025-05-11 15:26:31.095209500   File "/data/apps/dbus-serialbattery/ext/can/_entry_points.py", line 3, in <module>

2025-05-11 15:26:31.095212500     from dataclasses import dataclass

2025-05-11 15:26:31.095214500 ModuleNotFoundError: No module named 'dataclasses'

^C

Anything else I could try ? Is it a python module problem ?

Thanks

Roger

mr-manuel, 2025-05-11:
Read my message again. Try VENUS OS large or beta, not the driver itself, but it's never wrong to also test the latest nightly.

rogergrant99, 2025-05-11:
My bad, after installing the large Venus OS all is back to normal.

Thank you.

Roger