mr-manuel/venus-os_dbus-serialbattery #185: Why no integration with setuphelper ?

spacecabbie, 2025-02-12:
### Is your feature request related to a problem? Please describe.

Greetings,

Just was wondering if you are aware of https://github.com/kwindrem/SetupHelper 
It would be a good match perhaps.

mr-manuel, 2025-02-13:
https://github.com/Louisvdw/dbus-serialbattery/issues/67

mr-manuel, 2025-02-13:
SetupHelper additionally adds a lot of additional maintenance work and you anyway need SSH access for troubleshooting, should the update go wrong or you have to update the config for compatibility.

There is a bigger project in progress were I will integrate this driver.

closed 2025-02-13 by mr-manuel