mr-manuel/venus-os_dbus-serialbattery #186: LiTime_Ble

Drizzt321, 2025-02-14:
### Describe the bug

I'm playing around with VenusOS on a RPi Zero 2W, had my LiTime 280Ah battery connected with a branch version (https://github.com/mr-manuel/venus-os_dbus-serialbattery/pull/59) that got working. Upon starting up (after a long time) the RPi, didn't get any battery info. I decided to updated the driver since I know it had been merged, so I went ahead and updated to stable. That didn't work, tried beta, and that didn't work either. I've included relevant snippets from `/data/log/dbus-blebattery.0/current` for both stable & beta outputs, I'm seeing python stacktrace. 

### How to reproduce

Follow install instructions from https://mr-manuel.github.io/venus-os_dbus-serialbattery_docs/general/install on install of VenusOS on RPi Zero 2W, set `config.ini` to the attached below values. Bluetooth is turned off as per CLI recommendation. 

### Expected behavior

Expect LiTime LFP battery to show up in VenusOS (GUI v2).

### Driver version of the currently installed driver

v1.6.20250131, v2.0.20250207beta

### Driver version of the last known working driver

Branch for PR https://github.com/mr-manuel/venus-os_dbus-serialbattery/pull/59

### Venus OS device type

Please select

### Venus OS version

v3.54

### BMS type

LiTime

### Cell count

4

### Battery count

1

### Connection type

Bluetooth

### Config file

```ini
MAX_BATTERY_CHARGE_CURRENT    = 100.0
MAX_BATTERY_DISCHARGE_CURRENT = 200.0

BMS_TYPE=LiTime_Ble
BLUETOOTH_BMS=LiTime_Ble C8:47:80:1C:B1:E1
MIN_CELL_VOLTAGE   = 3.000
MAX_CELL_VOLTAGE   = 3.450
FLOAT_CELL_VOLTAGE = 3.375
```

### Relevant log output

```shell
Stable

@4000000067aed1b22a4864ec Device C8:47:80:1C:B1:E1 (public)
@4000000067aed1b22a487c5c       Alias: L-12280BNN200-A01039
@4000000067aed1b22a48842c       Paired: no
@4000000067aed1b22a488bfc       Trusted: no
@4000000067aed1b22a4893cc       Blocked: no
@4000000067aed1b22a489b9c       Connected: no
@4000000067aed1b22a489f84       LegacyPairing: no
@4000000067aed1b22a48a754       UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
@4000000067aed1b22a5716a4 
@4000000067aed1b520ce375c INFO:SerialBattery:
@4000000067aed1b520d520e4 INFO:SerialBattery:Starting dbus-serialbattery
@4000000067aed1b520f26ce4 INFO:SerialBattery:Venus OS v3.54 running on Raspberry Pi Zero 2 W Rev 1.0
@4000000067aed1b520f8ecf4 INFO:SerialBattery:dbus-serialbattery v1.6.20250131
@4000000067aed1b5213be964 Traceback (most recent call last):
@4000000067aed1b5213bfcec   File "/opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py", line 400, in <module>
@4000000067aed1b5213c1074     main()
@4000000067aed1b5213c1844   File "/opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py", line 270, in main
@4000000067aed1b5213c2bcc     class_ = eval(port)
@4000000067aed1b5213c339c   File "<string>", line 1, in <module>
@4000000067aed1b5213c3f54 NameError: name 'LiTime_Ble' is not defined
tail: /data/log/dbus-blebattery.0/current has been replaced; following end of new file
@4000000067aed1b53742147c [CHG] Device 5A:D2:9A:08:58:EF TxPower is nil
@4000000067aed1b537f297fc [CHG] Device 5A:D2:9A:08:58:EF RSSI is nil
@4000000067aed1b537f2af6c [CHG] Device 76:3F:93:9D:33:D6 TxPower is nil
@4000000067aed1b537f2bf0c [CHG] Device 76:3F:93:9D:33:D6 RSSI is nil
@4000000067aed1b537f2ceac [CHG] Device 6C:AD:F9:BD:C9:F5 RSSI is nil
@4000000067aed1b537f2de4c [CHG] Device C7:DA:40:8C:2A:5F RSSI is nil
@4000000067aed1b537f50ce4 [CHG] Device 61:7A:3E:E5:1F:7B RSSI is nil
@4000000067aed1b537f51c84 [CHG] Device 24:5A:4C:6A:A2:7A RSSI is nil
@4000000067aed1b537f52c24 [CHG] Device 4C:E0:A1:FC:AA:72 TxPower is nil
@4000000067aed1b537f537dc [CHG] Device 4C:E0:A1:FC:AA:72 RSSI is nil
@4000000067aed1b537f5477c [CHG] Controller D8:3A:DD:77:24:88 Discovering: no
@4000000067aed1b537f5b8c4 Successful disconnected

Current beta

@4000000067aed124325e1c94 INFO:SerialBattery:
@4000000067aed12432680f74 INFO:SerialBattery:Starting dbus-serialbattery
@4000000067aed124328a3d74 INFO:SerialBattery:Venus OS v3.54 running on Raspberry Pi Zero 2 W Rev 1.0
@4000000067aed1243292962c INFO:SerialBattery:dbus-serialbattery v2.0.20250207beta
@4000000067aed12509e2d87c Traceback (most recent call last):
@4000000067aed12509e2f3d4   File "/data/apps/dbus-serialbattery/dbus-serialbattery.py", line 503, in <module>
@4000000067aed12509e3075c     main()
@4000000067aed12509e30b44   File "/data/apps/dbus-serialbattery/dbus-serialbattery.py", line 333, in main
@4000000067aed12509e31ecc     testbms = class_("ble_" + ble_address.replace(":", "").lower(), 9600, ble_address)
@4000000067aed12509e33254 TypeError: Can't instantiate abstract class LiTime_Ble with abstract methods get_settings
```

mr-manuel, 2025-02-14:
Thanks for reporting. The problem was introduced with https://github.com/mr-manuel/venus-os_dbus-serialbattery/commit/0a2a09a75edb6248d8ee2847d71397c9549772ab and fixed with https://github.com/mr-manuel/venus-os_dbus-serialbattery/commit/55f58ac8cb3c0d5d61aec1a7d888e29060a4059d.

Please install the latest nightly and report back.

mr-manuel, 2025-02-14:
Side note: LiTime was introduced with v2, so it definitely won't work with v1. See https://github.com/mr-manuel/venus-os_dbus-serialbattery/blob/master/CHANGELOG.md

closed 2025-02-14 by mr-manuel

Drizzt321, 2025-02-14:
@mr-manuel success with current nightly!

Good to know about v2, I'll stick with beta/nightly until v2 becomes stable.