mr-manuel/venus-os_dbus-serialbattery #190: dbus-serial-battery startet nicht mehr

ray90408, 2025-02-19:
### Describe the bug

I changed the config.ini while Venus was running and restarted it. After that, serial-battery no longer ran. Even a new installation did not help.
Why did I do this? After two almost full cells (3.4V) differed by 20mV and the charging current was constantly adjusted by the driver, causing the charging power to fluctuate greatly, I wanted to calm things down a bit with new settings.

### How to reproduce

1. config.ini 
CELL_VOLTAGES_WHILE_CHARGING   = 3.50, 3.44, 3.42, 3.35
MAX_CHARGE_CURRENT_CV_FRACTION =    0, 0.05,  0.8,    1

changed

;CELL_VOLTAGES_WHILE_CHARGING   = 3.50, 3.44, 3.42, 3.35
;MAX_CHARGE_CURRENT_CV_FRACTION =    0, 0.05,  0.8,    1
2.Restart
Driver no longer starts or is no longer visible in Venus

### Expected behavior

I expected the default.ini settings to be set so that the charging current is reduced to 3.45V at the end phase of charging, so that serialbattery does not constantly set the permissible charging current differently due to the changing cell voltage differences. (Charging power jumps up and down by 1000W)

### Driver version of the currently installed driver

v2.0.20250214dev

### Driver version of the last known working driver

v1.6.20250131

### Venus OS device type

Raspberry Pi 4

### Venus OS version

v3.54

### BMS type

JKBMS (Heltec BMS)

### Cell count

16

### Battery count

1

### Connection type

Serial USB adapter to RS485

### USB hub

Yes, no external power supply

### Config file

```ini
[DEFAULT]

MAX_BATTERY_CHARGE_CURRENT = 70.0
MAX_BATTERY_DISCHARGE_CURRENT = 90.0

MIN_CELL_VOLTAGE   = 3.000
MAX_CELL_VOLTAGE   = 3.450

FLOAT_CELL_VOLTAGE = 3.375
[... 228 lines cut ...]
2025-02-19 14:59:09.991922500     from utils import logger, publish_config_variables
2025-02-19 14:59:09.991925500   File "/opt/victronenergy/dbus-serialbattery/utils.py", line 44, in <module>
2025-02-19 14:59:09.992016500     config.read([default_config_file_path, custom_config_file_path])
2025-02-19 14:59:09.992020500   File "/usr/lib/python3.8/configparser.py", line 697, in read
2025-02-19 14:59:09.992023500     self._read(fp, filename)
2025-02-19 14:59:09.992026500   File "/usr/lib/python3.8/configparser.py", line 1017, in _read
2025-02-19 14:59:09.992029500     for lineno, line in enumerate(fp, start=1):
2025-02-19 14:59:09.992032500   File "/usr/lib/python3.8/codecs.py", line 322, in decode
2025-02-19 14:59:09.992060500     (result, consumed) = self._buffer_decode(data, self.errors, final)
2025-02-19 14:59:09.992064500 UnicodeDecodeError: 'utf-8' codec can't decode byte 0xe4 in position 242: invalid continuation byte
2025-02-19 14:59:25.358231500 *** CCGX booted (0) ***
2025-02-19 15:16:50.376657500 *** CCGX booted (0) ***
2025-02-19 15:21:58.302780500 *** CCGX booted (0) ***

tail -F -n 100 /data/log/serial-starter/current | grep dbus-serialbattery | tai64nlocal
keine Anzeige

root@raspberrypi4:~# head /data/var/lib/serial-starter/*
==> /data/var/lib/serial-starter/ttyUSB0 <==
gps

==> /data/var/lib/serial-starter/ttyUSB1 <==
mkx
root@raspberrypi4:~#
```

mr-manuel, 2025-02-19:
```
/opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py
```

Indicated, that somehow Venus OS still tries to run driver version v1. Did you reboot after installing v2?

What is the output of `lsusb`? if there is no entry in the `serial-starter` log, it's very likely USB related. Since you are using a USB hub, it's even more likely that the USB hub the the point of failure. What, if you connect the USB to serial adapter directly to the GX device?

mr-manuel, 2025-02-19:
Additionally it seems that you have some coding problems. Your config.ini has some strange characters and the log also reports `UnicodeDecodeError: 'utf-8' codec can't decode byte 0xe4 in position 242: invalid continuation byte`.

How did you install the driver?

Freensman, 2025-02-23:
Hello, I tried again to install an get serial battery running today. Now after 3 hours of install and uninstall I found this post. 
Not sure when this started. I think the first installation it worked. Than I found the post that the new Daly BMS don't work with UART, so I reinstall again and tried now with RS485 (incl. settings in config.ini.

I see the same as in the upper post. 

I tried to install the BETA version 2, but this did not work as well, is in a different folder "apps".
So changed back to latest 1.6 and  installed it that time via USB stick. Same, Same 

I run Venus 3.54 (20250128120635)

ls -l /dev/ttyUSB*.  -> seams fine I see /dev/ttyUSB0 and would also .....USB1

root@einstein:~# tail -F -n 100 /data/log/dbus-serialbattery.ttyUSB0/current | tai64nlocal

2025-02-23 14:34:55.076281500   File "/opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py", line 13, in <module>
2025-02-23 14:34:55.076285500     from dbushelper import DbusHelper
2025-02-23 14:34:55.076288500   File "/opt/victronenergy/dbus-serialbattery/dbushelper.py", line 8, in <module>
2025-02-23 14:34:55.076291500     from utils import logger, publish_config_variables
2025-02-23 14:34:55.076294500   File "/opt/victronenergy/dbus-serialbattery/utils.py", line 421, in <module>
2025-02-23 14:34:55.076423500     BATTERY_CAPACITY: float = get_float_from_config("DEFAULT", "BATTERY_CAPACITY")
2025-02-23 14:34:55.076427500   File "/opt/victronenergy/dbus-serialbattery/utils.py", line 82, in get_float_from_config
2025-02-23 14:34:55.076431500     return float(config[group][option])
2025-02-23 14:34:55.076433500 ValueError: could not convert string to float: ''
2025-02-23 14:35:38.202765500 *** CCGX booted (0) ***
2025-02-23 14:36:25.675948500 Traceback (most recent call last):
2025-02-23 14:36:25.675958500   File "/opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py", line 13, in <module>
2025-02-23 14:36:25.675962500     from dbushelper import DbusHelper
2025-02-23 14:36:25.675964500   File "/opt/victronenergy/dbus-serialbattery/dbushelper.py", line 8, in <module>
2025-02-23 14:36:25.675968500     from utils import logger, publish_config_variables
2025-02-23 14:36:25.675970500   File "/opt/victronenergy/dbus-serialbattery/utils.py", line 421, in <module>
2025-02-23 14:36:25.676118500     BATTERY_CAPACITY: float = get_float_from_config("DEFAULT", "BATTERY_CAPACITY")
2025-02-23 14:36:25.676122500   File "/opt/victronenergy/dbus-serialbattery/utils.py", line 82, in get_float_from_config
2025-02-23 14:36:25.676126500     return float(config[group][option])
2025-02-23 14:36:25.676128500 ValueError: could not convert string to float: ''
2025-02-23 14:36:55.715920500 Traceback (most recent call last):
2025-02-23 14:36:55.715929500   File "/opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py", line 13, in <module>
2025-02-23 14:36:55.715933500     from dbushelper import DbusHelper
2025-02-23 14:36:55.715935500   File "/opt/victronenergy/dbus-serialbattery/dbushelper.py", line 8, in <module>
2025-02-23 14:36:55.715939500     from utils import logger, publish_config_variables
2025-02-23 14:36:55.715941500   File "/opt/victronenergy/dbus-serialbattery/utils.py", line 421, in <module>
2025-02-23 14:36:55.716090500     BATTERY_CAPACITY: float = get_float_from_config("DEFAULT", "BATTERY_CAPACITY")
2025-02-23 14:36:55.716095500   File "/opt/victronenergy/dbus-serialbattery/utils.py", line 82, in get_float_from_config
2025-02-23 14:36:55.716099500     return float(config[group][option])
2025-02-23 14:36:55.716101500 ValueError: could not convert string to float: ''
2025-02-23 14:37:18.044350500 Traceback (most recent call last):
2025-02-23 14:37:18.044359500   File "/opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py", line 13, in <module>
2025-02-23 14:37:18.044363500     from dbushelper import DbusHelper
2025-02-23 14:37:18.044365500   File "/opt/victronenergy/dbus-serialbattery/dbushelper.py", line 8, in <module>
2025-02-23 14:37:18.044368500     from utils import logger, publish_config_variables
2025-02-23 14:37:18.044371500   File "/opt/victronenergy/dbus-serialbattery/utils.py", line 421, in <module>
2025-02-23 14:37:18.044513500     BATTERY_CAPACITY: float = get_float_from_config("DEFAULT", "BATTERY_CAPACITY")
2025-02-23 14:37:18.044518500   File "/opt/victronenergy/dbus-serialbattery/utils.py", line 82, in get_float_from_config
2025-02-23 14:37:18.044522500     return float(config[group][option])
2025-02-23 14:37:18.044524500 ValueError: could not convert string to float: ''
2025-02-23 14:37:41.115717500 Traceback (most recent call last):
2025-02-23 14:37:41.115728500   File "/opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py", line 13, in <module>
2025-02-23 14:37:41.115732500     from dbushelper import DbusHelper
2025-02-23 14:37:41.115734500   File "/opt/victronenergy/dbus-serialbattery/dbushelper.py", line 8, in <module>
2025-02-23 14:37:41.115738500     from utils import logger, publish_config_variables
2025-02-23 14:37:41.115740500   File "/opt/victronenergy/dbus-serialbattery/utils.py", line 421, in <module>
2025-02-23 14:37:41.115898500     BATTERY_CAPACITY: float = get_float_from_config("DEFAULT", "BATTERY_CAPACITY")
2025-02-23 14:37:41.115903500   File "/opt/victronenergy/dbus-serialbattery/utils.py", line 82, in get_float_from_config
2025-02-23 14:37:41.115907500     return float(config[group][option])
2025-02-23 14:37:41.115909500 ValueError: could not convert string to float: ''
[... cut ...]
2025-02-23 14:39:58.623764500     BATTERY_CAPACITY: float = get_float_from_config("DEFAULT", "BATTERY_CAPACITY")
2025-02-23 14:39:58.623768500   File "/opt/victronenergy/dbus-serialbattery/utils.py", line 82, in get_float_from_config
2025-02-23 14:39:58.623772500     return float(config[group][option])
2025-02-23 14:39:58.623774500 ValueError: could not convert string to float: ''
^[[1;2A2025-02-23 14:40:21.661828500 Traceback (most recent call last):

No plan what further to do. Like to get twi new DALY BMS running, at the moment just one connected via RS485

Thanks a lot for your help.

Silvio

mr-manuel, 2025-02-23:
@Freensman please open a new issue, so I have all data I need from you, since you have different hardware