mr-manuel/venus-os_dbus-serialbattery #193: Need help for Update driver - less memory - cerbo gx 3.54 large

Approved-OK, 2025-02-21:
### Describe the problem

Update driver failed due to Memory-leak.

root@einstein:~# df
Filesystem           1K-blocks      Used Available Use% Mounted on
/dev/root              1200384   1097248     24828  98% /          <<<<----------------------------------------
devtmpfs                465376         4    465372   0% /dev
tmpfs                   515040      1016    514024   0% /run
tmpfs                   515040       656    514384   0% /var/volatile
/dev/mmcblk1p5         1134336    296468    762196  28% /data
/dev/mmcblk0p1        31248384       832  31247552   0% /run/media/mmcblk0p1
/dev/sda1             15561600      5824  15555776   0% /run/media/sda1
tmpfs                   515040      1016    514024   0% /service

how is it possible to resize /dev/root or find whats wrong.
of course run the resizefs command, but did not help.

root@einstein:~# bash /opt/victronenergy/swupdate-scripts/resize2fs.sh
resizing /dev/mmcblk1p3
resize2fs 1.45.7 (28-Jan-2021)
The filesystem is already 327680 (4k) blocks long.  Nothing to do!

Installed the last update manualy - did work but a lot of work...

is there a solution for this problem ?

thx for help

### Installation method

Install script from the documentation

### Driver version of the currently installed driver

v1.6.20250131

### Venus OS device type

Cerbo GX

### Venus OS version

v3.54 large

### BMS type

Daly BMS

### Cell count

16

### Battery count

2

### Connection type

Serial USB adapter to RS485

### USB hub

Yes, no external power supply

### Config file

```ini
thats not the problem - 1.6.20250131 2 batteries seriell rs485 with one usb cti-connector 
thx great job so far...
```

### Relevant log output

```shell
root@einstein:~# bash /opt/victronenergy/swupdate-scripts/resize2fs.sh
resizing /dev/mmcblk1p3
resize2fs 1.45.7 (28-Jan-2021)
The filesystem is already 327680 (4k) blocks long.  Nothing to do!
```

mr-manuel, 2025-02-21:
Install v2

Approved-OK, 2025-02-22:
it worked, thanks that's worth a beer :-)

closed 2025-02-22 by Approved-OK