mr-manuel/venus-os_dbus-serialbattery #194: dbus-serialbattery not running

Freensman, 2025-02-23:
### Describe the problem

Hello, I tried again to install an get serial battery running today. Now after 3 hours of install and uninstall I found this post.
Not sure when this started. I think the first installation it worked. Than I found the post that the new Daly BMS don't work with UART, so I reinstall again and tried now with RS485 (incl. settings in config.ini.)

I also tried to install the BETA version 2, but this did not work as well, is in a different folder "apps".
So changed back to latest 1.6 and installed it that time via USB stick. 

No plan what further to do. Like to get two new DALY BMS running, at the moment just one connected via RS485

Thanks a lot for your help.

Silvio

### Installation method

None

### Driver version of the currently installed driver

v1.6.20250131

### Venus OS device type

Cerbo GX

### Venus OS version

3.54

### BMS type

Daly BMS

### Cell count

8

### Battery count

later 2

### Connection type

Serial USB adapter to RS485

### Config file

```ini
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):
; MAX_BATTERY_CHARGE_CURRENT = 50.0
; MAX_BATTERY_DISCHARGE_CURRENT = 60.0

; --------- Set logging level ---------
; ERROR: Only errors are logged
; WARNING: Errors and warnings are logged
; INFO: Errors, warnings, and info messages are logged
; DEBUG: Errors, warnings, info, and debug messages are logged
LOGGING = INFO

; --------- CAN BMS ---------
; Description:
;     Specify the CAN port(s) where the BMS is connected. Leave empty to disable.
;     Show available CAN ports with: canshow
; Available CAN BMS:
[... 148 lines cut ...]
2025-02-23 14:39:35.851885500   File "/opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py", line 13, in <module>
2025-02-23 14:39:35.851889500     from dbushelper import DbusHelper
2025-02-23 14:39:35.851891500   File "/opt/victronenergy/dbus-serialbattery/dbushelper.py", line 8, in <module>
2025-02-23 14:39:35.851894500     from utils import logger, publish_config_variables
2025-02-23 14:39:35.851897500   File "/opt/victronenergy/dbus-serialbattery/utils.py", line 421, in <module>
2025-02-23 14:39:35.852039500     BATTERY_CAPACITY: float = get_float_from_config("DEFAULT", "BATTERY_CAPACITY")
2025-02-23 14:39:35.852044500   File "/opt/victronenergy/dbus-serialbattery/utils.py", line 82, in get_float_from_config
2025-02-23 14:39:35.852048500     return float(config[group][option])
2025-02-23 14:39:35.852050500 ValueError: could not convert string to float: ''
2025-02-23 14:39:58.623607500 Traceback (most recent call last):
2025-02-23 14:39:58.623616500   File "/opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py", line 13, in <module>
2025-02-23 14:39:58.623620500     from dbushelper import DbusHelper
2025-02-23 14:39:58.623623500   File "/opt/victronenergy/dbus-serialbattery/dbushelper.py", line 8, in <module>
2025-02-23 14:39:58.623626500     from utils import logger, publish_config_variables
2025-02-23 14:39:58.623629500   File "/opt/victronenergy/dbus-serialbattery/utils.py", line 421, in <module>
2025-02-23 14:39:58.623764500     BATTERY_CAPACITY: float = get_float_from_config("DEFAULT", "BATTERY_CAPACITY")
2025-02-23 14:39:58.623768500   File "/opt/victronenergy/dbus-serialbattery/utils.py", line 82, in get_float_from_config
2025-02-23 14:39:58.623772500     return float(config[group][option])
2025-02-23 14:39:58.623774500 ValueError: could not convert string to float: ''
^[[1;2A2025-02-23 14:40:21.661828500 Traceback (most recent call last):
```

### Any other information that may be helpful

ls -l /dev/ttyUSB*.  -> seams fine I see /dev/ttyUSB0 and would also .....USB1

mr-manuel, 2025-02-23:
> ```ini
> ; Battery capacity (in amp-hours), if the BMS does not support reading it.
> BATTERY_CAPACITY = 
> ; Invert Battery Current. Default is non-inverted. Set to -1 to invert.
> INVERT_CURRENT_MEASUREMENT =
> ```

Set a value for both this options. Normally they are not empty and therefore you get this errors in all versions of the driver.

Freensman, 2025-02-23:
Thanks a lot, you are absolutely right. I did not read the default_config.ini carefully. I expected blank is okee like with other settings. Now the driver is working. 

Unfortunately the new 100balance BMS from Daly is not working with UART --> discussions/180

And it seams I also have the one mentioned problem (Issue #163)  over RS485. It just don't recognize the BMS, its trying with different ModBus Ports but it don't find the BMS.

I have played around with board numbers, but can not change it right now, as the only working WIN computer is in the office.
So need to test that next week. 

Thanks for your help till now.

closed 2025-02-23 by mr-manuel