mr-manuel/venus-os_dbus-serialbattery #196: Devision by zero in battery.py get_midvoltage() line 1645

asmcc, 2025-02-25:
### Describe the bug

During testing with my new DEYE integration I gain followed ERROR:

`2025-02-25 22:15:19.936708500 DEBUG:SerialBattery:CAN Message Data BMS alarms: 0000000000000000
2025-02-25 22:15:19.938108500 DEBUG:SerialBattery:bat_stat_fet = 00110001
2025-02-25 22:15:19.939998500 DEBUG:SerialBattery:bms_check = 0000000101001001, bat_check = 0000000000100011, intercan_check = 0000000000000000, self.init_check = 0000000011111111
2025-02-25 22:15:19.943209500 DEBUG:SerialBattery:bms_check = 0000000101001101, bat_check = 0000000000100011, intercan_check = 0000000000000000, self.init_check = 0000000011111111
2025-02-25 22:15:19.958726500 DEBUG:SerialBattery:[can8] Received: ID=0x650, Daten=bytearray(b'B3200757')
2025-02-25 22:15:19.969380500 DEBUG:SerialBattery:[can8] Received: ID=0x110, Daten=bytearray(b'\x00\x00\x00\x00\x00\x00\x001')
2025-02-25 22:15:20.004154500 Traceback (most recent call last):
2025-02-25 22:15:20.004165500   File "/data/apps/dbus-serialbattery/dbushelper.py", line 883, in publish_battery
2025-02-25 22:15:20.004171500     self.publish_dbus()
2025-02-25 22:15:20.004174500   File "/data/apps/dbus-serialbattery/dbushelper.py", line 911, in publish_dbus
2025-02-25 22:15:20.004179500     midpoint, deviation = self.battery.get_midvoltage()
2025-02-25 22:15:20.004183500   File "/data/apps/dbus-serialbattery/battery.py", line 1645, in get_midvoltage
2025-02-25 22:15:20.004362500     abs((half2voltage - half1voltage) / (half2voltage + half1voltage) * 100),
2025-02-25 22:15:20.004370500 ZeroDivisionError: division by zero
2025-02-25 22:15:20.004373500 DEBUG:SerialBattery:Polling data took 0.126 seconds
2025-02-25 22:15:25.866576500 INFO:SerialBattery:
2025-02-25 22:15:25.868206500 INFO:SerialBattery:Starting dbus-serialbattery
2025-02-25 22:15:25.872744500 INFO:SerialBattery:Venus OS v3.60~25 running on Raspberry Pi 2 Model B Rev 1.1 
2025-02-25 22:15:25.873924500 INFO:SerialBattery:dbus-serialbattery v2.0.20250214dev
2025-02-25 22:15:27.025911500 DEBUG:SerialBattery:Interface can8 is already up
2025-02-25 22:15:27.167086500 INFO:SerialBattery:Detected CAN Bus bitrate: 500 kbps
2025-02-25 22:15:27.170490500 DEBUG:SerialBattery:Wait shortly to make sure that all needed data is in the cache
2025-02-25 22:15:27.328452500 DEBUG:SerialBattery:[can8] Received: ID=0x150, Daten=bytearray(b'\x12\x02#\x00\xfe\x02\xb8\x03')
2025-02-25 22:15:27.406167500 DEBUG:SerialBattery:[can8] Received: ID=0x35c, Daten=bytearray(b'\x08\x00\x00\x00\x00\x00\x00\x00')
`
I assume, that during the initialisation in seldom cases you can gain here zero values for both "half1voltage" and "half2voltage", if cell.voltage is not initialised or all cell voltages are zero.

In case of a such zero devision dbus-serialbattery starts again. Theoretically so named "boot loops" are here possible.

### How to reproduce

I assume, it is not so simple to reproduce this behaviour, since you must reproduce an accurate timing during the initialisation. But you can reproduce it indirectly, if you call get_midvoltage before cell.voltage is initialised or if cell.voltage is initialised by zero values

### Expected behavior

Restart of dbus-serialbattery (see log above). Looping is teoretically possible.

### Installation method

Install script from the documentation

### Driver version of the currently installed driver

v2.0.20250214dev see log

### Driver version of the last known working driver

v2.0.20250214dev see log

### Venus OS device type

⚠️ Choose your option ⚠️

### Venus OS version

3.60 see Log

### BMS type

JKBMS (Heltec BMS)

### Cell count

16

### Battery count

1

### Connection type

⚠️ Choose your option ⚠️

### USB hub

⚠️ Choose your option ⚠️

### Config file

```ini
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):
MAX_BATTERY_CHARGE_CURRENT = 200
MAX_BATTERY_DISCHARGE_CURRENT = 200

CAN_PORT = can8
BMS_TYPE = Deye_Can
TEMPERATURE_SOURCE_BATTERY = 1
TEMPERATURE_1_NAME = "Average Cell Temperature"
TEMPERATURE_2_NAME = "Maximal Cell Temperature"
TEMPERATURE_3_NAME = "Minimal Cell Temperature"
TEMPERATURE_4_NAME = "Heating Film Temperature"
GUI_PARAMETERS_SHOW_ADDITIONAL_INFO = True
PUBLISH_BATTERY_DATA_AS_JSON = False

CCCM_CV_ENABLE = False
DCCM_CV_ENABLE = False
CCCM_T_ENABLE = False
DCCM_T_ENABLE = False

MIDPOINT_ENABLE = True

;INVERT_CURRENT_MEASUREMENT = -1

LOGGING = DEBUG
```

### Relevant log output

```shell
2025-02-25 22:15:20.004165500   File "/data/apps/dbus-serialbattery/dbushelper.py", line 883, in publish_battery
2025-02-25 22:15:20.004171500     self.publish_dbus()
2025-02-25 22:15:20.004174500   File "/data/apps/dbus-serialbattery/dbushelper.py", line 911, in publish_dbus
2025-02-25 22:15:20.004179500     midpoint, deviation = self.battery.get_midvoltage()
2025-02-25 22:15:20.004183500   File "/data/apps/dbus-serialbattery/battery.py", line 1645, in get_midvoltage
2025-02-25 22:15:20.004362500     abs((half2voltage - half1voltage) / (half2voltage + half1voltage) * 100),
2025-02-25 22:15:20.004370500 ZeroDivisionError: division by zero
2025-02-25 22:15:20.004373500 DEBUG:SerialBattery:Polling data took 0.126 seconds
2025-02-25 22:15:25.866576500 INFO:SerialBattery:
2025-02-25 22:15:25.868206500 INFO:SerialBattery:Starting dbus-serialbattery
2025-02-25 22:15:25.872744500 INFO:SerialBattery:Venus OS v3.60~25 running on Raspberry Pi 2 Model B Rev 1.1
```

### Any other information that may be helpful

Each possible zero division should be intercepted. I would calculate (half2voltage + half1voltage) separatelly, then check it against zero and after that and only if it is not zero, then this calculation with division.
Yes, a little more code, but then it is secure.

mr-manuel, 2025-02-26:
You should already fetch all needed values with the `test_connection()` function in your driver. Do you have pushed the code to GitHub? Found nothing in your repo.

asmcc, 2025-02-26:
You mean a PR for this division zero or my DEYE integration? My DEYE integration is not fully ready yet and I did not posted it yet as fork for later integration here, since it is in testing now.
Only the required files with changes are posted as preview under [https://github.com/asmcc/my48vdc/tree/main/SerialBattery](https://github.com/asmcc/my48vdc/tree/main/SerialBattery).
And yes, I make usualy initialisation first and wait until all required messages are received. But here is something occurs, that this function was called without or with zero cell voltages.
Yes, it is very seldom, that such a case may occur, but a potential division by zero in the code is not good.

closed 2025-03-01 by mr-manuel