mr-manuel/venus-os_dbus-serialbattery #197: in V2 white Gui v1 after switching OS-Firmware

Didi1965, 2025-02-27:
### Describe the bug

After installing serialbattery v2.0.20250207beta in Venus-OS V3.54 allready looks good in GUI v1; 
Swap to Venus-OS Beta V3.60~25 allready looks good in GUI v1 too

!!Swap back to Venus-OS V3.54 again, suddenly i have got white Screen in GUI v1!!
>After disabling Serialbattery, the GUI v1 restarts and looks good again,
>After enabling Serialbattery, the GUI v1 restarts an remains white Screen too. 
>I have no GUI-Mod installed

 Inv1.6.20250131 all looks good

Hope this feedback helps
Thanks

### How to reproduce

see before

### Expected behavior

..

### Installation method

Install script from the documentation

### Driver version of the currently installed driver

v2.0.20250207beta

### Driver version of the last known working driver

v1.6.20250131

### Venus OS device type

Raspberry Pi 4

### Venus OS version

V3.54 + V3.60~25

### BMS type

Daly BMS

### Cell count

16

### Battery count

3

### Connection type

Serial USB adapter to RS485

### USB hub

Yes, with external power supply

### Config file

```ini
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.
[... 547 lines cut ...]
USE_BMS_DVCC_VALUES = False

; -- LltJbd settings
; SoC low levels
; Note:
;     SOC_LOW_WARNING can be used to calculate the Time-To-Go, even if you are not using an LltJbd BMS.
SOC_LOW_WARNING = 15
SOC_LOW_ALARM   = 10

; -- Daly settings
; Invert Battery Current. Default is non-inverted. Set to -1 to invert.
INVERT_CURRENT_MEASUREMENT = -1

; -- ESC GreenMeter and Lipro device settings
GREENMETER_ADDRESS  = 1
LIPRO_START_ADDRESS = 2
LIPRO_END_ADDRESS   = 4
LIPRO_CELL_COUNT    = 15
```

### Relevant log output

```shell
not needed, no functionally Problems with serialbattery, its only GUI v1 Problem
```

mr-manuel, 2025-02-27:
Currently there are that much changes ongoing in GUIv1 and GUIv2 that it's hard to keep the right version for every different Venus OS. I will fix that once Venus OS 3.60 was released.

Didi1965, 2025-02-27:
Thanks, for your feedback! 
Yes really, there are lots of changes, ..... I feel for you :-)
BR

mr-manuel, 2025-02-27:
Could you post me the output of

```bash
cat /var/log/gui/*| tai64nlocal
```

when you get the whitescreen? That would show the error. Only the content from the last `*** Starting gui ***` to the bottom is needed.

Didi1965, 2025-03-04:
Sorry, i was not at home last days.
Now i went back to OS 3.54 and had the white GUI v1 Screen again. But no log File, see below?!
root@einstein:~# cat /var/log/gui/*| tai64nlocal
cat: can't open '/var/log/gui/*': No such file or directory
root@einstein:~#

mr-manuel, 2025-03-04:
Because in the issue you said you have a Raspberry Pi, but from the output you pasted here you have a Cerbo GX.

Here the command for the Cerbo GX:

```bash
cat /var/log/start-gui/* | tai64nlocal
```

Sebbo69, 2025-03-22:
same Problem on Rpi4. White Screen after New QML files were installed and the GUI was restarted.

mr-manuel, 2025-03-24:
Can you post the output of the gui installation script when you have the above error?

```bash
/data/apps/dbus-serialbattery/custom-gui-install.sh
```

Didi1965, 2025-03-24:
root@raspberrypi4:~# /data/apps/dbus-serialbattery/custom-gui-install.sh
The app "dbus-serialbattery_gui" was already added to the directory "/opt/victronenergy/gui" in the config file.
The app "dbus-serialbattery_gui" was already added to the directory "/var/www/venus" in the config file.

Installing QML files for GUI V1...
Copying PageBattery.qml...
Copying PageBatteryCellVoltages.qml...
Copying PageBatteryParameters.qml...
Copying PageBatterySettings.qml...
Copying PageLynxIonIo.qml...
Venus OS v3.54 is older than v3.60~18. Fixing QtQuick version...
done.
New QML files were installed and the GUI was restarted.

root@raspberrypi4:~#

mr-manuel, 2025-03-24:
I was able to get a more detailed log now also on one system.

```
2025-03-24 07:32:19.307515500 *** starting start-gui ***
2025-03-24 07:32:19.328636500 method return time=1742801539.328413 sender=:1.70 -> destination=:1.1030 serial=9798 reply_serial=2
2025-03-24 07:32:19.328646500    int32 0
2025-03-24 07:32:19.383405500 *** headless device=0
2025-03-24 07:32:19.383797500 *** Waiting for localsettings...
2025-03-24 07:32:19.403007500 *** Localsettings is up, continuing...
2025-03-24 07:32:19.440083500 *** Starting gui, with VNC enabled)
2025-03-24 07:32:19.488734500 QVNCServer created on port 5900
2025-03-24 07:32:19.494287500 Reloading input devices: "LinuxInput:/dev/input/event0 LinuxInput:/dev/input/event1 LinuxInput:/dev/input/event2"
2025-03-24 07:32:19.497477500 "using /etc/venus for runtime features"
2025-03-24 07:32:19.498092500 "running on einstein"
2025-03-24 07:32:19.508999500 Connecting to deprecated signal QDBusConnectionInterface::serviceOwnerChanged(QString,QString,QString)
2025-03-24 07:32:19.698826500 Creating settings
2025-03-24 07:32:19.973004500 [VeQItemExportedDbusServices] Registered service "debug.victronenergy.gui"
2025-03-24 07:32:20.307923500 language is "en"
2025-03-24 07:32:20.313491500 accessLevelChanged:  2
2025-03-24 07:32:20.313618500 setAccessLevel:  2
2025-03-24 07:32:20.341558500 LocalThemeDaemonClient: Looking for assets in "/opt/victronenergy/themes/ccgx"
2025-03-24 07:32:20.342070500 Screen: 800x480 110x60mm 185x203dpi
2025-03-24 07:32:20.483537500 TimeZone::valueChanged:  "Europe/Berlin"
2025-03-24 07:32:20.483840500 ScreenSaver::dbusDisplayOffTimeChanged:  600
2025-03-24 07:32:20.483965500 ScreenSaver::enable
2025-03-24 07:32:20.484006500 ScreenSaver::screenOn
2025-03-24 07:32:23.293855500 file:///opt/victronenergy/gui/qml/main.qml:12:15: Type PageMain unavailable
2025-03-24 07:32:23.293864500           initialPage: PageMain {}
2025-03-24 07:32:23.293866500                        ^
2025-03-24 07:32:23.295015500 file:///opt/victronenergy/gui/qml/PageMain.qml:40:26: Type PageSettings unavailable
2025-03-24 07:32:23.295022500                                   subpage: Component { PageSettings {} }
2025-03-24 07:32:23.295025500                                                        ^
2025-03-24 07:32:23.295952500 file:///opt/victronenergy/gui/qml/PageSettings.qml:50:5: Type PageSettingsSystem unavailable
2025-03-24 07:32:23.295961500                                   PageSettingsSystem {
2025-03-24 07:32:23.295963500                                   ^
2025-03-24 07:32:23.296849500 file:///opt/victronenergy/gui/qml/PageSettingsSystem.qml:157:5: Type PageSettingsBatteries unavailable
2025-03-24 07:32:23.296857500                                   PageSettingsBatteries {
2025-03-24 07:32:23.296860500                                   ^
2025-03-24 07:32:23.297560500 file:///opt/victronenergy/gui/qml/PageSettingsBatteries.qml:1:1: module "QtQuick" version 2.0 is not installed
2025-03-24 07:32:23.297566500      import QtQuick 2
2025-03-24 07:32:23.297568500      ^
```

The very strange thing is, that `PageSettingsBatteries.qml` is not part of `dbus-serialbattery`. Have you installed also other drivers?

Execute this command and then reboot. That will delete the changed files and make a clean install of the GUI modifications:

```bash
rm -rf /data/apps/overlay-fs/data/gui/upper/qml

reboot
```

Didi1965, 2025-03-24:
Now it looks good :-)

in the current Test-Raspi installation i have over Setuphelper installed following packages:
Gui-Mods, RpiDisplaySetup, VeCanSetup, ShutdownMonitor, RpiTemperature 
and
dbus-aggregate-batteries

mr-manuel, 2025-03-24:
> Google Search says: https://github.com/stgnet/victron-gui-qml/blob/main/PageSettingsBatteries.qml

This isn't it. I would close it, since the solution is described above and it's very rare that it happens.

closed 2025-03-24 by mr-manuel

Albarge, 2025-03-25:
I just had the white screen after testing V3.60~49 with GUI Mods, Battery Aggregator and Dbus-Serial, but GUI Mods wouldn't install, and the missing opkg packages etc meant a few functions missing, so I went back to V3.54 but had white screen on the Ekrano in GuiV1 - I could ssh in though and I could see GuiV2 using the beta tablet viewer but the Ekrano was booting to GuiV1 with the white screen although I could change it to GuiV2 but not access package manager to remove anything.

I guess it's not related to Dbus-serial but the script above cleared the problem for me too and got rid of the white screen, so thanks again Mr-Manuel.

mr-manuel, 2025-11-25:
Resetting the root password, if GUIv1 does not work anymore, can be done through GUIv2. Sample URL: http://venus.local/gui-v2 or http://192.168.1.123/gui-v2

1. [Set access level to Superuser](https://www.victronenergy.com/live/ccgx:root_access#set_access_level_to_superuser)
2. [Create a temporary root password](https://www.victronenergy.com/live/ccgx:root_access#create_a_temporary_root_password)
3. [Enable sshd and log in](https://www.victronenergy.com/live/ccgx:root_access#enable_sshd_and_log_in)