mr-manuel/venus-os_dbus-serialbattery #202: Pace BMS: temperatures are not being read correctly

gerd-git, 2025-03-06:
### Describe the bug

Hello everyone,

I have got the Pace BMS running via RS232 with venus-os_dbus-serialbattery. Beta Build 'v2.0.20250207beta'.
However, the temperatures are not being read correctly. 
All sensor values are at about 29°C, which is definitely not correct. PbmsTools shows the correct values.

/data/log/dbus-serialbattery.ttyUSB0/current
(...)
@4000000067c8adf41e0084e4 DEBUG:SerialBattery:Temp sensor count: 6
@4000000067c8adf41e00f62c DEBUG:SerialBattery:Temperature [0]: 29.2
@4000000067c8adf41e0101e4 DEBUG:SerialBattery:Temperature [1]: 29.16
@4000000067c8adf41e010d9c DEBUG:SerialBattery:Temperature [2]: 29.18
@4000000067c8adf41e011954 DEBUG:SerialBattery:Temperature [3]: 29.17
@4000000067c8adf41e01250c DEBUG:SerialBattery:Temperature [4]: 29.22
@4000000067c8adf41e0130c4 DEBUG:SerialBattery:Temperature [5]: 29.3
(...)

[image]

### How to reproduce

Pace BMS running via RS232
Beta Build 'v2.0.20250207beta'
see temperatures

### Expected behavior

The correct temperatures should be read out

### Installation method

Install script from the documentation

### Driver version of the currently installed driver

v2.0.20250207beta

### Venus OS device type

⚠️ Choose your option ⚠️

### Venus OS version

v2.0.20250207beta

### BMS type

Pace BMS

### Cell count

16

### Battery count

1

### Connection type

Serial USB adapter to RS485

### Config file

```ini
[DEFAULT]

; --------- Set logging level ---------
; ERROR: Only errors are logged
[... 538 lines cut ...]

; -- Daly settings
; Invert Battery Current. Default is non-inverted. Set to -1 to invert.
INVERT_CURRENT_MEASUREMENT = 1

; -- ESC GreenMeter and Lipro device settings
GREENMETER_ADDRESS  = 1
LIPRO_START_ADDRESS = 2
LIPRO_END_ADDRESS   = 4
LIPRO_CELL_COUNT    = 15
```

### Relevant log output

```shell
(...)
@4000000067c8adf41e0084e4 DEBUG:SerialBattery:Temp sensor count: 6
@4000000067c8adf41e00f62c DEBUG:SerialBattery:Temperature [0]: 29.2
@4000000067c8adf41e0101e4 DEBUG:SerialBattery:Temperature [1]: 29.16
@4000000067c8adf41e010d9c DEBUG:SerialBattery:Temperature [2]: 29.18
@4000000067c8adf41e011954 DEBUG:SerialBattery:Temperature [3]: 29.17
@4000000067c8adf41e01250c DEBUG:SerialBattery:Temperature [4]: 29.22
@4000000067c8adf41e0130c4 DEBUG:SerialBattery:Temperature [5]: 29.3
(...)
```

mr-manuel, 2025-03-06:
@KoljaWindeler

gerd-git, 2025-03-07:
Thanks Kolja, that solves the problem.

I have inserted the line `v = (v-27.3)*10` in pace.py:

`nano /data/apps/dbus-serialbattery/bms/pace.py`

```
(...)
        for i in range(0, temperature_sensor_count):
            v = int(status_data[85 + i * 4 : 85 + i * 4 + 4], 16) / 100
--->        v = (v-27.3)*10
            logger.debug("Temperature [" + str(i) + "]: " + str(v))
            if i < 4:  # 0,1,2,3 are internal temps
                self.to_temperature(i + 1, v)
            if i == 4:  # mosfet
                self.to_temperature(0, v)
(...)
```

and it works now:

`cat /data/log/dbus-serialbattery.ttyUSB0/current`
```
(...)
@4000000067cab3a11a9f5c6c DEBUG:SerialBattery:Temp sensor count: 6
@4000000067cab3a11aa7a96c DEBUG:SerialBattery:Temperature [0]: 17.89999999999999
@4000000067cab3a11aafe6cc DEBUG:SerialBattery:Temperature [1]: 18.099999999999987
@4000000067cab3a11ab81874 DEBUG:SerialBattery:Temperature [2]: 17.799999999999976
@4000000067cab3a11ac0906c DEBUG:SerialBattery:Temperature [3]: 18.000000000000007
@4000000067cab3a11ac975c4 DEBUG:SerialBattery:Temperature [4]: 18.599999999999994
@4000000067cab3a11ad28de4 DEBUG:SerialBattery:Temperature [5]: 19.60000000000001
(...)
```
And yes, the screenshot above was not at the same time as the log output. But this one fits now:

[image]

Then another question: 
Pace supplies six temperature sensors, but only five are analysed. If I understand the code correctly, ENV_T is not read. Can SerialBattery not read six temperatures?

mr-manuel, 2025-03-07:
@KoljaWindeler thanks, I will add that fix.

mr-manuel, 2025-03-07:
@gerd-git can you try the latest nightly and report back, if it's fixed?

closed 2025-03-07 by mr-manuel

gerd-git, 2025-03-07:
Hello Manuel, 

I installed nightly:
```
root@raspberrypi2:~# wget -O /tmp/install.sh https://raw.githubusercontent.com/mr-manuel/venus-os_dbus-serialbattery/master/dbus-serialbattery/install.sh
(...)
root@raspberrypi2:~# bash /tmp/install.sh
(...)
Select which version you want to install from m-rmanuel's repo and enter the corresponding number: 3
> Selected: nightly build "v2.0.20250307dev" (newest features and fixes, bugs possible
(...)
```

But after reboot, pace bms is not found anymore:
`root@raspberrypi2:~# dbus-spy`
```
Services
com.victronenergy.acload.http_id71                                     Shelly WW
com.victronenergy.adc
com.victronenergy.battery.aggregator                                       Aggre
com.victronenergy.battery.socketcan_can0                                  JK CAN
com.victronenergy.digitalinputs
com.victronenergy.fronius
com.victronenergy.grid.cg_BW0590032001D                                     EM24
com.victronenergy.hub4
com.victronenergy.logger
com.victronenergy.modbusclient.tcp
com.victronenergy.modbustcp
com.victronenergy.packageManager                                 Package Manager
com.victronenergy.platform                                                 Venus
com.victronenergy.settings
com.victronenergy.shutdown                                       ShutdownMonitor
com.victronenergy.solarcharger.ttyUSB1                                    OST 60
com.victronenergy.solarcharger.ttyUSB3                                   WEST 70
com.victronenergy.solarcharger.ttyUSB4                                   SUD 100
com.victronenergy.system
com.victronenergy.temperature.RpiCpu06                                    PI-CPU
com.victronenergy.temperature.http_id71                                Shelly WW
com.victronenergy.vebus.ttyUSB2                                            MP II
```

Back to v2.0.20250207beta:

```
root@raspberrypi2:~# wget -O /tmp/install.sh https://raw.githubusercontent.com/mr-manuel/venus-os_dbus-serialbattery/master/dbus-serialbattery/install.sh
(...)
root@raspberrypi2:~# bash /tmp/install.sh
(...)
Select which version you want to install from m-rmanuel's repo and enter the corresponding number: 2
> Selected: beta build "v2.0.20250207beta" (no errors after 72 h runtime, long time testing needed)
(...)
root@raspberrypi2:~# reboot
```

Pace is found again:
`root@raspberrypi2:~# dbus-spy`
```
Services
com.victronenergy.acload.http_id71                                     Shelly WW
com.victronenergy.adc
com.victronenergy.battery.aggregator                                       Aggre
com.victronenergy.battery.socketcan_can0                                  JK CAN
com.victronenergy.battery.ttyUSB1                                       PACE 232
com.victronenergy.digitalinputs
com.victronenergy.fronius
com.victronenergy.grid.cg_BW0590032001D                                     EM24
com.victronenergy.hub4
com.victronenergy.logger
com.victronenergy.modbusclient.tcp
com.victronenergy.modbustcp
com.victronenergy.packageManager                                 Package Manager
com.victronenergy.platform                                                 Venus
com.victronenergy.settings
com.victronenergy.shutdown                                       ShutdownMonitor
com.victronenergy.solarcharger.ttyUSB0                                    OST 60
com.victronenergy.solarcharger.ttyUSB3                                   WEST 70
com.victronenergy.solarcharger.ttyUSB4                                   SUD 100
com.victronenergy.system
com.victronenergy.temperature.RpiCpu06                                    PI-CPU
com.victronenergy.temperature.http_id71                                Shelly WW
com.victronenergy.vebus.ttyUSB2                                            MP II
```

But with wrong temperatures:
```
@4000000067cac0600e16cdcc DEBUG:SerialBattery:Temp sensor count: 6
@4000000067cac0600e1f805c DEBUG:SerialBattery:Temperature [0]: 29.09
@4000000067cac0600e2af5f4 DEBUG:SerialBattery:Temperature [1]: 29.11
@4000000067cac0600e33f6a4 DEBUG:SerialBattery:Temperature [2]: 29.08
@4000000067cac0600e3cd814 DEBUG:SerialBattery:Temperature [3]: 29.1
@4000000067cac0600e45c154 DEBUG:SerialBattery:Temperature [4]: 29.21
@4000000067cac0600e4e970c DEBUG:SerialBattery:Temperature [5]: 29.26

```

So Nightly seems to have other problem?

mr-manuel, 2025-03-07:
Please try again. If it does not work again, please check the driver logs.

gerd-git, 2025-03-07:
In installed nightly build v2.0.20250307dev again and it now seems to work.

```
Services
com.victronenergy.acload.http_id71                                     Shelly WW
com.victronenergy.adc
com.victronenergy.battery.aggregator                                       Aggre
com.victronenergy.battery.socketcan_can0                                  JK CAN
com.victronenergy.battery.ttyUSB1                                       PACE 232
com.victronenergy.digitalinputs
com.victronenergy.fronius
com.victronenergy.grid.cg_BW0590032001D                                     EM24
com.victronenergy.hub4
com.victronenergy.logger
com.victronenergy.modbusclient.tcp
com.victronenergy.modbustcp
com.victronenergy.packageManager                                 Package Manager
com.victronenergy.platform                                                 Venus
com.victronenergy.settings
com.victronenergy.shutdown                                       ShutdownMonitor
com.victronenergy.solarcharger.ttyUSB0                                    OST 60
com.victronenergy.solarcharger.ttyUSB3                                   WEST 70
com.victronenergy.solarcharger.ttyUSB4                                   SUD 100
com.victronenergy.system
com.victronenergy.temperature.RpiCpu06                                    PI-CPU
com.victronenergy.temperature.http_id71                                Shelly WW
```

```
@4000000067cacf131cd48bec DEBUG:SerialBattery:Temp sensor count: 6
@4000000067cacf131cdc4864 DEBUG:SerialBattery:Temperature [0]: 18.0
@4000000067cacf131ce43b8c DEBUG:SerialBattery:Temperature [1]: 18.3
@4000000067cacf131cec1f14 DEBUG:SerialBattery:Temperature [2]: 17.9
@4000000067cacf131cf421dc DEBUG:SerialBattery:Temperature [3]: 18.2
@4000000067cacf131cfc5b54 DEBUG:SerialBattery:Temperature [4]: 20.4
@4000000067cacf131d046dbc DEBUG:SerialBattery:Temperature [5]: 19.9

```

Thank you !!!

commit 3b3f6878 (2025-03-07): Fix #202 Pace BMS wrong temperature
commit b6cab7b2 (2025-03-07): Fix #202 Pace BMS wrong temperature