mr-manuel/venus-os_dbus-serialbattery #204: JKBMS won't connect via Bluetooth

rver2024, 2025-03-09:
### Describe the bug

I haven't been testing the dbus-serialbattery drive for a little while now, and recently installed the latest beta build, v2.0.20250207beta and found out that my JKBMS won't connect to my Cerbo GX via Bluetooth, which it can connect on the older version driver (sorry don't remember which version). I also tried the latest nightly build and got the same result. 

### How to reproduce

Steps to reproduce the issue:
1. Install the latest nighly or beta build to Cerbo GX
2. change the /data/apps/dbus-serialbattery/config.ini to enable JKbms_Ble
3. reboot Cerbo GX
4. wait for JKbms to connect but it never does

### Expected behavior

I'd would like to be able to connect my JKBMS to connect to the Cerbo GX for battery monitoring. Thanks!

### Installation method

Install script from the documentation

### Driver version of the currently installed driver

v2.0.20250207beta

### Driver version of the last known working driver

sorry don't remember, already uninstalled

### Venus OS device type

Cerbo GX

### Venus OS version

v3.54

### BMS type

JKBMS (Heltec BMS)

### Cell count

8

### Battery count

1

### Connection type

Bluetooth

### USB hub

⚠️ Choose your option ⚠️

### Config file

```ini
[DEFAULT]

; --------- Set logging level ---------
; ERROR: Only errors are logged
; WARNING: Errors and warnings are logged
; INFO: Errors, warnings, and info messages are logged
; DEBUG: Errors, warnings, info, and debug messages are logged
LOGGING = INFO

; --------- Battery Current Limits ---------
; +++ Limits apply to each individual battery/BMS. +++
[... 533 lines cut ...]
; Invert Battery Current. Default is non-inverted. Set to -1 to invert.
INVERT_CURRENT_MEASUREMENT = 1

; -- ESC GreenMeter and Lipro device settings
GREENMETER_ADDRESS  = 1
LIPRO_START_ADDRESS = 2
LIPRO_END_ADDRESS   = 4
LIPRO_CELL_COUNT    = 15
```

### Relevant log output

```shell
2025-03-09 23:14:20.253633500 INFO:SerialBattery:
2025-03-09 23:14:20.254390500 INFO:SerialBattery:Starting dbus-serialbattery
2025-03-09 23:14:20.257723500 INFO:SerialBattery:Venus OS v3.54 running on Cerbo GX
2025-03-09 23:14:20.258374500 INFO:SerialBattery:dbus-serialbattery v2.0.20250207beta
2025-03-09 23:14:20.264360500 Traceback (most recent call last):
2025-03-09 23:14:20.264368500   File "/data/apps/dbus-serialbattery/dbus-serialbattery.py", line 503, in <module>
2025-03-09 23:14:20.264372500     main()
2025-03-09 23:14:20.264373500   File "/data/apps/dbus-serialbattery/dbus-serialbattery.py", line 330, in main
2025-03-09 23:14:20.264377500     class_ = eval(port)
2025-03-09 23:14:20.264378500   File "<string>", line 1, in <module>
2025-03-09 23:14:20.264380500 NameError: name 'JKbms_Ble' is not defined
```

BipedalPrimate, 2025-03-10:
Not sure if it is relevant but you don't appear to have specified the Bluetooth MAC address in config.ini at:

BLUETOOTH_BMS = JKbms_Ble xx:xx:xx:xx:xx:xx

rver2024, 2025-03-10:
Hi @BipedalPrimate, I do have the correct mac address for the JKBMS in the config file in my system, I just masked it here since it is irrelevant.

mr-manuel, 2025-03-10:
> ~~~
> NameError: name 'JKbms_Ble' is not defined
> ~~~

You habe a typo in your config. It‘s `Jkbms_Ble`. You wrote the second letter uppercase.

rver2024, 2025-03-11:
Hi @mr-manuel, sorry for making such low level mistake. I corrected it the config.ini file, but the JKBMS is still not connected to the Cerbo GX. I tried it on the v2.0.20250207beta, and currently on the latest nightly v2.0.20250307dev, both behave the same. I updated the log output, please help take a look. Thank you very much!

mr-manuel, 2025-03-11:
Install the Venus OS large image or the latest Venus OS beta and see https://github.com/mr-manuel/venus-os_dbus-serialbattery/issues/86#issuecomment-2710027233

In Venus OS 3.5x there is a little bug that this class is missing in the normal image.

rver2024, 2025-03-11:
Hi @mr-manuel Thanks a lot for your help again! I am making some progress after installing the latest Venus OS v3.60~39beta large image. The JKBMS can connect to the Cerbo GX via bluetooth but it connects and disconnects every few seconds. I updated log here, please help take a look again.

mr-manuel, 2025-03-11:
Just post the log here, else you modify the complete context of this issue.

rver2024, 2025-03-11:
2025-03-11 13:29:19.789942500 INFO:SerialBattery:
2025-03-11 13:29:19.790659500 INFO:SerialBattery:Starting dbus-serialbattery
2025-03-11 13:29:19.793006500 INFO:SerialBattery:Venus OS v3.60~39 running on Cerbo GX
2025-03-11 13:29:19.793441500 INFO:SerialBattery:dbus-serialbattery v2.0.20250307dev
2025-03-11 13:29:20.085628500 INFO:SerialBattery:Init of Jkbms_Ble at xx:xx:xx:xx:xx:xx
2025-03-11 13:29:20.086188500 INFO:SerialBattery:Test of Jkbms_Ble at xx:xx:xx:xx:xx:xx
2025-03-11 13:29:23.590526500 INFO:SerialBattery:BAT: JKBMS 11.XW 8S (20231207)
2025-03-11 13:29:23.591807500 INFO:SerialBattery:-- Connection established to Jkbms_Ble
2025-03-11 13:29:25.986615500 INFO:SerialBattery:Reconnected to previously identified battery with DeviceInstance: 1
2025-03-11 13:29:26.239002500 INFO:SerialBattery:Use DeviceInstance: 1
2025-03-11 13:29:26.240799500 INFO:SerialBattery:Use dbus ServiceName: com.victronenergy.battery.ble_c8478005f092
2025-03-11 13:29:26.288011500 INFO:SerialBattery:Battery JKBMS BLE connected to dbus from ble_c8478005f092
2025-03-11 13:29:26.292255500 INFO:SerialBattery:========== Settings ==========
2025-03-11 13:29:26.293920500 INFO:SerialBattery:> Connection voltage: 26.6 V | Current: 0.0 A | SoC: 96%
2025-03-11 13:29:26.294691500 INFO:SerialBattery:> Cell count: 8 | Cells populated: 8
2025-03-11 13:29:26.295254500 INFO:SerialBattery:> CHARGE MODE: Linear
2025-03-11 13:29:26.295853500 INFO:SerialBattery:> MIN CELL VOLTAGE: 3.000 V | MAX CELL VOLTAGE: 3.550 V| FLOAT CELL VOLTAGE: 3.375 V
2025-03-11 13:29:26.296405500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 50.0 A | MAX BATTERY DISCHARGE CURRENT: 60.0 A
2025-03-11 13:29:26.296945500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 50.0 A | MAX BATTERY DISCHARGE CURRENT: 200.0 A (read from BMS)
2025-03-11 13:29:26.297461500 INFO:SerialBattery:> CVCM:     True
2025-03-11 13:29:26.298112500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2025-03-11 13:29:26.298670500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2025-03-11 13:29:26.299193500 INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
2025-03-11 13:29:26.299719500 INFO:SerialBattery:> CHARGE FET: True | DISCHARGE FET: True | BALANCE FET: True
2025-03-11 13:29:26.300239500 INFO:SerialBattery:Serial Number/Unique Identifier: 3090145024
2025-03-11 13:29:31.229691500 INFO:SerialBattery:
2025-03-11 13:29:31.230411500 INFO:SerialBattery:Starting dbus-serialbattery
2025-03-11 13:29:31.232852500 INFO:SerialBattery:Venus OS v3.60~39 running on Cerbo GX
2025-03-11 13:29:31.233330500 INFO:SerialBattery:dbus-serialbattery v2.0.20250307dev
2025-03-11 13:29:31.535490500 INFO:SerialBattery:Init of Jkbms_Ble at xx:xx:xx:xx:xx:xx
2025-03-11 13:29:31.536034500 INFO:SerialBattery:Test of Jkbms_Ble at xx:xx:xx:xx:xx:xx
2025-03-11 13:29:35.039730500 INFO:SerialBattery:BAT: JKBMS 11.XW 8S (20231207)
2025-03-11 13:29:35.040844500 INFO:SerialBattery:-- Connection established to Jkbms_Ble
2025-03-11 13:29:37.479951500 INFO:SerialBattery:Reconnected to previously identified battery with DeviceInstance: 1
2025-03-11 13:29:37.746977500 INFO:SerialBattery:Use DeviceInstance: 1
2025-03-11 13:29:37.748777500 INFO:SerialBattery:Use dbus ServiceName: com.victronenergy.battery.ble_c8478005f092
2025-03-11 13:29:37.785888500 INFO:SerialBattery:Battery JKBMS BLE connected to dbus from ble_c8478005f092
2025-03-11 13:29:37.785899500 INFO:SerialBattery:========== Settings ==========
2025-03-11 13:29:37.785901500 INFO:SerialBattery:> Connection voltage: 26.6 V | Current: 0.0 A | SoC: 96%
2025-03-11 13:29:37.785904500 INFO:SerialBattery:> Cell count: 8 | Cells populated: 8
2025-03-11 13:29:37.785907500 INFO:SerialBattery:> CHARGE MODE: Linear
2025-03-11 13:29:37.785909500 INFO:SerialBattery:> MIN CELL VOLTAGE: 3.000 V | MAX CELL VOLTAGE: 3.550 V| FLOAT CELL VOLTAGE: 3.375 V
2025-03-11 13:29:37.786136500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 50.0 A | MAX BATTERY DISCHARGE CURRENT: 60.0 A
2025-03-11 13:29:37.786142500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 50.0 A | MAX BATTERY DISCHARGE CURRENT: 200.0 A (read from BMS)
2025-03-11 13:29:37.786197500 INFO:SerialBattery:> CVCM:     True
2025-03-11 13:29:37.787325500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2025-03-11 13:29:37.794644500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2025-03-11 13:29:37.794652500 INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
2025-03-11 13:29:37.794655500 INFO:SerialBattery:> CHARGE FET: True | DISCHARGE FET: True | BALANCE FET: True
2025-03-11 13:29:37.794658500 INFO:SerialBattery:Serial Number/Unique Identifier: 3090145024
2025-03-11 13:29:42.740475500 INFO:SerialBattery:
2025-03-11 13:29:42.741173500 INFO:SerialBattery:Starting dbus-serialbattery
2025-03-11 13:29:42.743515500 INFO:SerialBattery:Venus OS v3.60~39 running on Cerbo GX
2025-03-11 13:29:42.743959500 INFO:SerialBattery:dbus-serialbattery v2.0.20250307dev
2025-03-11 13:29:43.043565500 INFO:SerialBattery:Init of Jkbms_Ble at xx:xx:xx:xx:xx:xx
2025-03-11 13:29:43.044125500 INFO:SerialBattery:Test of Jkbms_Ble at xx:xx:xx:xx:xx:xx
2025-03-11 13:29:46.547672500 INFO:SerialBattery:BAT: JKBMS 11.XW 8S (20231207)
2025-03-11 13:29:46.548873500 INFO:SerialBattery:-- Connection established to Jkbms_Ble
2025-03-11 13:29:48.913363500 INFO:SerialBattery:Reconnected to previously identified battery with DeviceInstance: 1
2025-03-11 13:29:49.171579500 INFO:SerialBattery:Use DeviceInstance: 1
[... cut ...]
2025-03-11 13:30:12.579206500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2025-03-11 13:30:12.579840500 INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
2025-03-11 13:30:12.581165500 INFO:SerialBattery:> CHARGE FET: True | DISCHARGE FET: True | BALANCE FET: True
2025-03-11 13:30:12.582262500 INFO:SerialBattery:Serial Number/Unique Identifier: 3090145024
2025-03-11 13:30:14.430639500 ./run: line 16:  3983 Segmentation fault      python /data/apps/dbus-serialbattery/dbus-serialbattery.py Jkbms_Ble xx:xx:xx:xx:xx:xx

rver2024, 2025-03-11:
sorry I already updated the log in the original post. I posted here again but got some strikethroughs on some of the lines, not sure what happenned. Thanks!

rver2024, 2025-03-11:
Tried to install the ble-testing but encountered error. I tried a couple of times. Please see below:

*** Welcome to the dbus-serialbattery installer from mr-manuel! ***

Fetch available version numbers... done.

** Currently installed version: v2.0.20250307dev **

1) stable release "v1.6.20250131:" (stable, most up to date)
2) beta build "v2.0.20250207beta:" (no errors after 72 h runtime, long time testing needed)
3) nightly build "v2.0.20250307dev" (newest features and fixes, bugs possible)
4) specific branch (specific feature testing)
5) specific version
6) local tar file
7) quit

Select which version you want to install from m-rmanuel's repo and enter the corresponding number: 4
> Selected: specific branch (specific feature testing)

Config.ini backed up to /data/apps/dbus-serialbattery_config.ini.backup

1) ble-testing:          3) new-pid-controllers:
2) master:               4) v1.x.x:

Select the branch you want to install and enter the corresponding number: 1
> Selected branch: ble-testing:
--2025-03-11 14:35:45--  https://github.com/mr-manuel/venus-os_dbus-serialbattery/archive/refs/heads/ble-testing:.zip
Resolving github.com... 140.82.114.4
Connecting to github.com|140.82.114.4|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://codeload.github.com/mr-manuel/venus-os_dbus-serialbattery/zip/refs/heads/ble-testing: [following]
--2025-03-11 14:35:46--  https://codeload.github.com/mr-manuel/venus-os_dbus-serialbattery/zip/refs/heads/ble-testing:
Resolving codeload.github.com... 140.82.113.9
Connecting to codeload.github.com|140.82.113.9|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2025-03-11 14:35:46 ERROR 404: Not Found.

ERROR: Error during downloading the ZIP file. Please try again.
Config.ini restored to /data/apps/dbus-serialbattery/config.ini

rver2024, 2025-03-11:
Is the comma supposed to be in the ble-testing:.zip? in below link:
https://github.com/mr-manuel/venus-os_dbus-serialbattery/archive/refs/heads/ble-testing:.zip

rver2024, 2025-03-11:
could be it since I could download the zip file on my pc if I removed the colon, otherwise I got a 404 error

mr-manuel, 2025-03-11:
Yep, found it now to. Fixed it right now.

rver2024, 2025-03-11:
Hi @mr-manuel, Looks like your fix is in and I was able to install the ble-testing branch to my Cerbo GX and indeed the JKBMS connects to Cerbo GX via bluetooth fine, I will test the bluetooth connection longer to see this ble-testing version improves the connection's reliability, I remember the bluetooth connection would randomly drop off in previous versions when I tested them a few months ago. Thanks!

itsdaveit, 2025-03-11:
@mr-manuel i just wanted to let you know, that i had the same issue, after upgrading from 1.2 or so, two weeks ago. The current ble-testing seems to work again. I use it with the internal bluetooth module.

Thanks for your effort!

```
2025-03-11 19:24:25.896665500 *** CCGX booted (0) ***
2025-03-11 19:25:30.474814500 INFO:SerialBattery:
2025-03-11 19:25:30.475498500 INFO:SerialBattery:Starting dbus-serialbattery
2025-03-11 19:25:30.485051500 INFO:SerialBattery:Venus OS v3.54 running on Cerbo-S GX
2025-03-11 19:25:30.485870500 INFO:SerialBattery:dbus-serialbattery v2.0.20250310-ble-testing
2025-03-11 19:25:33.410251500 INFO:SerialBattery:Init of Jkbms_Ble at C8:47:8C:E8:E0:C2
2025-03-11 19:25:34.988017500 INFO:SerialBattery:|- Try to connect to Jkbms_Ble at C8:47:8C:E8:E0:C2
2025-03-11 19:25:37.118681500 INFO:SerialBattery:|- Device connected, check if it's really a JKBMS
2025-03-11 19:25:37.596512500 INFO:SerialBattery:   |- Device info received
2025-03-11 19:25:38.372447500 INFO:SerialBattery:Processing frame with settings info
2025-03-11 19:25:38.607619500 INFO:SerialBattery:   |- Settings received
2025-03-11 19:25:39.601587500 INFO:SerialBattery:   |- Cell info received
2025-03-11 19:25:39.604495500 INFO:SerialBattery:-- Connection established to Jkbms_Ble
2025-03-11 19:25:40.870758500 INFO:SerialBattery:Reconnected to previously identified battery with DeviceInstance: 1
2025-03-11 19:25:41.331582500 INFO:SerialBattery:Use DeviceInstance: 1
2025-03-11 19:25:41.334870500 INFO:SerialBattery:Use dbus ServiceName: com.victronenergy.battery.ble_c8478ce8e0c2
2025-03-11 19:25:41.483809500 INFO:SerialBattery:Polling interval: 1.000 s
2025-03-11 19:25:41.486761500 INFO:SerialBattery:Battery JKBMS BLE connected to dbus from ble_c8478ce8e0c2
2025-03-11 19:25:41.487811500 INFO:SerialBattery:========== Settings ==========
2025-03-11 19:25:41.489061500 INFO:SerialBattery:> Connection voltage: 53.02 V | Current: 0.0 A | SoC: 82%
2025-03-11 19:25:41.490844500 INFO:SerialBattery:> Cell count: 16 | Cells populated: 16
2025-03-11 19:25:41.490854500 INFO:SerialBattery:> CHARGE MODE: Linear
2025-03-11 19:25:41.492058500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.900 V | MAX CELL VOLTAGE: 3.451 V| FLOAT CELL VOLTAGE: 3.450 V
2025-03-11 19:25:41.524137500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 140.0 A | MAX BATTERY DISCHARGE CURRENT: 140.0 A
2025-03-11 19:25:41.524299500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 150.0 A | MAX BATTERY DISCHARGE CURRENT: 120.0 A (read from BMS)
2025-03-11 19:25:41.526408500 INFO:SerialBattery:> CVCM:     True
2025-03-11 19:25:41.541340500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2025-03-11 19:25:41.542299500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2025-03-11 19:25:41.544729500 INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
2025-03-11 19:25:41.544739500 INFO:SerialBattery:> CHARGE FET: True | DISCHARGE FET: True | BALANCE FET: True
2025-03-11 19:25:41.544743500 INFO:SerialBattery:Serial Number/Unique Identifier: 2092447184 ```

rver2024, 2025-03-11:
I tested the ble-testing version for a few hours, it has worked pretty good, but it connects and disconnects to Cerbo GX every 60 odd seconds, looking at the log, it seems the JKBMS would reconnect after finished processing frame with settings info, not sure it was designed this way or not. Here are 3 sections of log which would repeat again and again every 60 odd seconds.

```
2025-03-11 21:26:10.560929500 INFO:SerialBattery:|- Try to connect to Jkbms_Ble at xx:xx:xx:xx:xx:xx
2025-03-11 21:26:12.196058500 INFO:SerialBattery:|- Device connected, check if it's really a JKBMS
2025-03-11 21:26:13.448358500 INFO:SerialBattery:Processing frame with settings info
2025-03-11 21:27:00.531232500 INFO:SerialBattery:current: 0.0 - voltage: 26.57 - temp MOS: 24.9 - temp1: 21.6 - temp2: 21                   .3 - last update: 0s - cell voltages: [3.322, 3.322, 3.324, 3.318, 3.323, 3.322, 3.322, 3.322]
2025-03-11 21:27:15.583238500 INFO:SerialBattery:|- Try to connect to Jkbms_Ble at xx:xx:xx:xx:xx:xx
2025-03-11 21:27:16.550831500 INFO:SerialBattery:|- Device connected, check if it's really a JKBMS
2025-03-11 21:27:17.848359500 INFO:SerialBattery:Processing frame with settings info
2025-03-11 21:28:00.614233500 INFO:SerialBattery:current: 0.0 - voltage: 26.57 - temp MOS: 24.9 - temp1: 21.6 - temp2: 21.3 - last update: 0s - cell voltages: [3.323, 3.323, 3.323, 3.318, 3.324, 3.323, 3.323, 3.322]
2025-03-11 21:28:20.563844500 INFO:SerialBattery:|- Try to connect to Jkbms_Ble at xx:xx:xx:xx:xx:xx
2025-03-11 21:28:21.595453500 INFO:SerialBattery:|- Device connected, check if it's really a JKBMS
2025-03-11 21:28:22.866334500 INFO:SerialBattery:Processing frame with settings info
2025-03-11 21:29:00.697577500 INFO:SerialBattery:current: 0.0 - voltage: 26.58 - temp MOS: 24.9 - temp1: 21.6 - temp2: 21.3 - last update: 0s - cell voltages: [3.323, 3.322, 3.324, 3.318, 3.324, 3.324, 3.323, 3.323]
```