mr-manuel/venus-os_dbus-serialbattery #210: Modified Temp 3/4 for charge allow on JK-BMS JK-B2A20S20P-HC

mr-manuel, 2025-03-14:
### Discussed in https://github.com/mr-manuel/venus-os_dbus-serialbattery/discussions/200

<div type='discussions-op-text'>

<sup>Originally posted by **rvisscher95** March  2, 2025</sup>
Hello,

I was struggling to get the JK-BMS JK-B2A20S20P-HC being allowed to charge from the d-bus script. The reason was a third temperature sensor being read at 0C. This problem was not solved by the beta update [v2.0.20250207beta] in my case. 

I solved the issue by adjusting the battery.py script such that the battery min and max values are also made up only of the temperature sensor list from the config.ini file : TEMPERATURE_SOURCE_BATTERY. 

I changed the existing lines at the get max/min temperatures functions to the approach proposed in the beta update:

```
temperature_map = {1: self.temperature_1, 2: self.temperature_2, 3: self.temperature_3, 4: self.temperature_4}
# Calculate the average temperature from the sensors given in the list
temperatures = [temperature_map.get(sensor) for sensor in utils.TEMPERATURE_SOURCE_BATTERY if temperature_map.get(sensor) is not None]
```
If I there is another way of doing this properly, I would really like to know. 

Kind regards,
Robert Visscher
</div>

mr-manuel, 2025-03-14:
@rvisscher95 @fritz-de can you please try the nightly from today?

closed 2025-03-14 by mr-manuel

eicke123, 2025-03-15:
@mr-manuel 
Hi I'm new here and not a python native. But I tried this version on my Raspberry with the latest venus os. 
This line didn't work with the installed python version 3.8.18 
` def get_filtered_temperature_map(self) -> dict[int, float]:`

I fixed it for my testing with:
`    def get_filtered_temperature_map(self) -> Dict[int, float]:`
and added Dict in line 2.

Best 
Christoph

fritz-de, 2025-03-15:
> [@rvisscher95](https://github.com/rvisscher95) [@fritz-de](https://github.com/fritz-de) can you please try the nightly from today?

@mr-manuel: The changes are working exept the "Dict"-bug Christoph reported. Thanks a lot. This was my frist trip to the github-community. Next time i will try to generate bug-report. :-)
@eicke123 : Due to I am a python-rookyie, I would not have been able to get the nightly started without your hint. Thx

mr-manuel, 2025-03-16:
@eicke123 thanks. I tested it only with the latest Venus OS beta image, which has Python 3.12 where it works. I fixed it now that it works with the old and new version. Please reinstall the nightly.

commit bd40a9e3 (2025-03-14): Fix #210
commit 18d610ce (2025-03-16): Fix #210