mr-manuel/venus-os_dbus-serialbattery #212: dBus driver don't see 2x Daly BMS 100

Freensman, 2025-03-16:
### Describe the problem

I did multiple attemps to get the dbus serial working within many hours, tonight after uninstall, delete and reinstall I could make it work shorty that one of the two DalyBMS could be seen. I had connected just one, this was visible. As I connected the other one, just one was seen. changing to just the other one (visile on GX) but not on dbus serial.

Looked into details it always had some errors:

tail -F -n 100 /data/log/dbus-canbattery.*/current | tai64nlocal

2025-03-13 19:12:45.515255500 ERROR:SerialBattery:Exception occurred: CanError('Failed to transmit: [Errno 105] No buffer space available') of type <class 'can.CanError'> in /opt/victronenergy/dbus-serialbattery/bms/daly_can.py line #97

More in the Loging.

Log 1 - as it worked with first BMS
Log 2 - with the second BMS
Log 3 - back with the first BMS

At the end, first BMS was visible, but as the second was added it was gone, even GX showed the second one.

Now two days later, the db-bus serial is a gain not visible.

I have no plan where to look

Thanks for your help about this.

[image]

### Installation method

Install script from the documentation

### Driver version of the currently installed driver

v1.6.20250131

### Venus OS device type

Cerbo GX

### Venus OS version

v3.54

### BMS type

Daly BMS

### Cell count

8

### Battery count

2

### Connection type

CAN

### USB hub

⚠️ Choose your option ⚠️

### Config file

```ini
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

[... 661 lines cut ...]
2025-03-13 19:14:50.009664500 INFO:SerialBattery:Venus OS v3.54 running on Cerbo GX
2025-03-13 19:14:50.010314500 INFO:SerialBattery:dbus-serialbattery v1.6.20250131
2025-03-13 19:14:51.707249500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2025-03-13 19:14:51.711949500 INFO:SerialBattery:Testing Daly_Can
2025-03-13 19:14:51.717762500 ERROR:SerialBattery:Exception occurred: CanError('Failed to transmit: [Errno 105] No buffer space available') of type <class 'can.CanError'> in /opt/victronenergy/dbus-serialbattery/bms/daly_can.py line #97
2025-03-13 19:14:51.718446500 INFO:SerialBattery:Testing Jkbms_Can
2025-03-13 19:14:51.719821500 DEBUG:SerialBattery:Can bus init
2025-03-13 19:14:51.725506500 DEBUG:SerialBattery:bustype: socketcan, channel: can0, bitrate: None
2025-03-13 19:14:51.726145500 DEBUG:SerialBattery:Can bus init done
2025-03-13 19:14:52.728477500 INFO:SerialBattery:No CAN Message received
2025-03-13 19:14:53.230377500 INFO:SerialBattery:-- Testing BMS: 2 of 3 rounds
2025-03-13 19:14:53.231256500 INFO:SerialBattery:Testing Daly_Can
2025-03-13 19:14:53.232787500 DEBUG:SerialBattery:bus shutdown
2025-03-13 19:14:53.240477500 ERROR:SerialBattery:Exception occurred: CanError('Failed to transmit: [Errno 105] No buffer space available') of type <class 'can.CanError'> in /opt/victronenergy/dbus-serialbattery/bms/daly_can.py line #97
2025-03-13 19:14:53.264043500 INFO:SerialBattery:Testing Jkbms_Can
2025-03-13 19:14:53.265393500 DEBUG:SerialBattery:Can bus init
2025-03-13 19:14:53.271256500 DEBUG:SerialBattery:bustype: socketcan, channel: can0, bitrate: None
2025-03-13 19:14:53.272073500 DEBUG:SerialBattery:Can bus init done
^Z
[2]+  Stopped(SIGTSTP)        tail -F -n 100 /data/log/dbus-canbattery.*/current | tai64nlocal
```

### Any other information that may be helpful

Both

Freensman, 2025-03-16:
I tried so many things over the last month, with different versions etc. 

This time I did not tried another time the nighty build. 

But just did now. Can not even find the folder / open the config.ini file anymore. 
There seams to be no dbus-serialbattery folder in the /data/etc/ anymore. So not sure what that time got wrong. 
Also no possibility to uninstall.  

So no Idea what to do as start with all different installs or uninstalls again. Like I did all the other times. 
To me it feels a bit of a like a stroke of luck.

May you have an idea, as you know what has been changed in between versions?

Thanks

mr-manuel, 2025-03-16:
It would help, if people read the text when opening an issue, then I don't have to repeat it over and over. Here again:

🚨 IMPORTANT 🚨

- Please carefully read all the instructions below including the descriptions! Very often we have to request the data again, because it was not provided in the first place. <-- not done

- Please try the nightly build first, if you are not already using it. It's very likely that the bug is already fixed. See [here](https://mr-manuel.github.io/venus-os_dbus-serialbattery_docs/general/install#nightly-build) how to install it. <-- done after requesting it again

- Did you reboot at least once after you saw the issue? If not, reboot and check if the issue still exists.

- Please read the complete [changelog](https://github.com/mr-manuel/dbus-serialbattery/blob/master/CHANGELOG.md). <-- not done

 - Make sure to provide the logfiles in the correct format with a READABLE timestamp. <-- OK

mr-manuel, 2025-03-17:
> Failed to transmit: [Errno 105] No buffer space available

This means you CAN controller fails to send CAN frames. Make sure you have a working CAN bus, proper termination, same bitrate, at least two CAN stations.

https://mr-manuel.github.io/venus-os_dbus-serialbattery_docs/general/connect#can-daisy-chain-possible

Freensman, 2025-03-17:
Thanks for this, it seams drivers are changing fast and I have to follow carefully version numbers and change log. I will try to do so. 
This may got a bit neglected after so many hours try. 

regarding your last comment. As you can see in the screenshot, the venus gx sees the battery always without any problem. one time it could see the battery directly and the dbus-serial. but normaly it sees just one battery and as soon as the second gets connected the one that dominates get visible / stays visible. So I doubt a bit that it has something to do with the baud rate, termination ect. as it otherwise would not be seen from the gx at all? Also can see communication working under "devices" in Service / CAN BMS. 

I will do some more testing tonight. Thanks for help

Freensman, 2025-03-17:
I just gave it another try. As you mentioned the Errno 105 with CAN relation I tested the Ve.CAN port with the bms with 250kbits. (before with BMS-CAN with 500kits)
And I can see one battery on dbus serial and one in gx, so means that the DALY BMS is changing its baud rate automaticaly.  There are no setting for it. I saw that before once but it did not came back.  This for sure is out of your reach, but may you are aware of this.

db-serial can just see one bms even if I change the bms, it keeps showing the frozen datas from the first bms, but the second bms is visible on gx. On gx I can see the bms change but not in dbus serial.

If I have two bms connected just one is visible on gx but changing from bms1 to bms2  and back (never saw this before). 
On db- serial just one bms is visible.

How would / should two bms be displayed in db-serial? As two batteries / devices?

I also can just see 1 BMS online in details. 

I'm also confused about the explanation with multiple bms over RS485 and modbus addresses. There it mentions this is just required for RS485m but there is a column for CAN addresses?

Log with two bms connected:

tail: /data/log/dbus-canbattery.can0/current has been replaced; following end of new file

==> /data/log/dbus-canbattery.can0/current <==
2025-03-17 14:44:51.505977500 DEBUG:SerialBattery:[can0] Received: ID=0x377, Daten=bytearray(b'\x01\x00\x00\x00\x00\x00\x00\x00')
2025-03-17 14:44:51.505986500 DEBUG:SerialBattery:[can0] Received: ID=0x379, Daten=bytearray(b'\xc8\x00\x00\x00\x00\x00\x00\x00')
2025-03-17 14:44:51.776729500 DEBUG:SerialBattery:[can0] Received: ID=0x305, Daten=bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00')
2025-03-17 14:44:51.778317500 DEBUG:SerialBattery:[can0] Received: ID=0x307, Daten=bytearray(b'\x124VxVIC\x00')
2025-03-17 14:44:51.894121500 DEBUG:SerialBattery:[can0] Received: ID=0x351, Daten=bytearray(b' \x01\xa6\x0e\xa6\x0e\xe4\x00')
2025-03-17 14:44:51.896112500 DEBUG:SerialBattery:[can0] Received: ID=0x355, Daten=bytearray(b'F\x00d\x00X\x1b\x00\x00')
2025-03-17 14:44:51.897740500 DEBUG:SerialBattery:[can0] Received: ID=0x356, Daten=bytearray(b'F\n\x00\x00\xc8\x00\x00\x00')
2025-03-17 14:44:51.899176500 DEBUG:SerialBattery:[can0] Received: ID=0x35a, Daten=bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00')
2025-03-17 14:44:51.900553500 DEBUG:SerialBattery:[can0] Received: ID=0x35e, Daten=bytearray(b'LVaiiey\x00')
2025-03-17 14:44:51.901916500 DEBUG:SerialBattery:[can0] Received: ID=0x35f, Daten=bytearray(b'\x00\x00\x01\x01\x8d\x00\x00\x00')
2025-03-17 14:44:51.903322500 DEBUG:SerialBattery:[can0] Received: ID=0x370, Daten=bytearray(b'Lithium\x00')
2025-03-17 14:44:51.904886500 DEBUG:SerialBattery:[can0] Received: ID=0x371, Daten=bytearray(b'Valley\x00\x00')
2025-03-17 14:44:51.906256500 DEBUG:SerialBattery:[can0] Received: ID=0x373, Daten=bytearray(b'\xde\x0c\xe0\x0c$\x01&\x01')
2025-03-17 14:44:51.907692500 DEBUG:SerialBattery:[can0] Received: ID=0x374, Daten=bytearray(b'\x04\x00\x01\x00\x01\x00\x02\x00')
2025-03-17 14:44:51.909120500 DEBUG:SerialBattery:[can0] Received: ID=0x375, Daten=bytearray(b'\x01\x00\x00\x00\x00\x00\x00\x00')
2025-03-17 14:44:51.910455500 DEBUG:SerialBattery:[can0] Received: ID=0x376, Daten=bytearray(b'\x01\x00\x00\x00\x00\x00\x00\x00')
2025-03-17 14:44:51.911777500 DEBUG:SerialBattery:[can0] Received: ID=0x377, Daten=bytearray(b'\x02\x00\x00\x00\x00\x00\x00\x00')
2025-03-17 14:44:51.913122500 DEBUG:SerialBattery:[can0] Received: ID=0x379, Daten=bytearray(b'\x8d\x00\x00\x00\x00\x00\x00\x00')
2025-03-17 14:44:52.333163500 DEBUG:SerialBattery:[can0] Received: ID=0x18904001, Daten=bytearray(b'\x01\x07\x00\x00u0\x02\xbc')
2025-03-17 14:44:52.334616500 DEBUG:SerialBattery:[can0] Received: ID=0x18914001, Daten=bytearray(b'\x0c\xe0\x01\x0c\xde\x04\xff\xff')
2025-03-17 14:44:52.336027500 DEBUG:SerialBattery:[can0] Received: ID=0x18924001, Daten=bytearray(b'=\x02;\x01\xff\xff\xff\xff')
2025-03-17 14:44:52.337508500 DEBUG:SerialBattery:[can0] Received: ID=0x18934001, Daten=bytearray(b'\x00\x01\x01>\x00\x02(X')
2025-03-17 14:44:52.339015500 DEBUG:SerialBattery:[can0] Received: ID=0x18944001, Daten=bytearray(b'\x08\x02\x00\x00\x00\x00\x00@')
2025-03-17 14:44:52.340607500 DEBUG:SerialBattery:[can0] Received: ID=0x18a64001, Daten=bytearray(b'\x01\x0c\xe0\x0c\xdf\x0c\xdf\x00')
2025-03-17 14:44:52.342012500 DEBUG:SerialBattery:[can0] Received: ID=0x18a74001, Daten=bytearray(b'\x02\x0c\xde\x0c\xe0\x0c\xdf\x00')
2025-03-17 14:44:52.343384500 DEBUG:SerialBattery:[can0] Received: ID=0x18a84001, Daten=bytearray(b'\x03\x0c\xde\x0c\xe0\x0c\xdf\x00')
2025-03-17 14:44:52.344726500 DEBUG:SerialBattery:[can0] Received: ID=0x18974001, Daten=bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00')
2025-03-17 14:44:52.366656500 DEBUG:SerialBattery:[can0] Received: ID=0x18984001, Daten=bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00')
2025-03-17 14:44:52.412739500 DEBUG:SerialBattery:Data check: 2
2025-03-17 14:44:52.425102500 DEBUG:SerialBattery:logged to dbus [70.0]
2025-03-17 14:44:52.426561500 DEBUG:SerialBattery:Cells:[1]3.296V [2]3.295V [3]3.295V [4]3.294V [5]3.296V [6]3.295V [7]3.294V [8]3.296V 
2025-03-17 14:44:52.427734500 DEBUG:SerialBattery:Polling data took 0.120 seconds
2025-03-17 14:44:52.462382500 DEBUG:SerialBattery:[can0] Received: ID=0x351, Daten=bytearray(b'\x1e\x01\xa6\x0e\xa6\x0e\xd4\x00')
2025-03-17 14:44:52.464053500 DEBUG:SerialBattery:[can0] Received: ID=0x355, Daten=bytearray(b"d\x00d\x00\x10\'\x00\x00")
[... cut ...]
2025-03-17 14:45:02.913920500 DEBUG:SerialBattery:[can0] Received: ID=0x375, Daten=bytearray(b'\x01\x00\x00\x00\x00\x00\x00\x00')
2025-03-17 14:45:02.915719500 DEBUG:SerialBattery:[can0] Received: ID=0x376, Daten=bytearray(b'\x01\x00\x00\x00\x00\x00\x00\x00')
2025-03-17 14:45:02.917564500 DEBUG:SerialBattery:[can0] Received: ID=0x377, Daten=bytearray(b'\x02\x00\x00\x00\x00\x00\x00\x00')
2025-03-17 14:45:02.919544500 DEBUG:SerialBattery:[can0] Received: ID=0x379, Daten=bytearray(b'\x8d\x00\x00\x00\x00\x00\x00\x00')