mr-manuel/venus-os_dbus-serialbattery #215: dBus: not all values are shown

BeSt-SP, 2025-03-18:
### Describe the problem

Hi! I am currently trying to use the driver with 14x Daly BMS via CAN, on a Cerbo GX. After the first startup, I got the following error:

`@4000000067d9220b2652dc14 INFO:SerialBattery:Use dbus ServiceName: com.victronenergy.battery.can0__0x05
@4000000067d9220b27dba6c4 Traceback (most recent call last):
@4000000067d9220b27de2f34   File "/data/apps/dbus-serialbattery/dbus-serialbattery.py", line 508, in <module>
@4000000067d9220b28049b24     main()
@4000000067d9220b280bf20c   File "/data/apps/dbus-serialbattery/dbus-serialbattery.py", line 458, in main
@4000000067d9220b282d8b9c     if not helper[key_address].setup_vedbus():
@4000000067d9220b28369804            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
@4000000067d9220b283995a4   File "/data/apps/dbus-serialbattery/dbushelper.py", line 698, in setup_vedbus
@4000000067d9220b28677ba4     for i in range(1, self.battery.cell_count + 1):
@4000000067d9220b2871726c                       ~~~~~~~~~~~~~~~~~~~~~~~~^~~
@4000000067d9220b2874c9e4 TypeError: unsupported operand type(s) for +: 'NoneType' and 'int'`

I managed to fix this by changing the sleep value in daly_can.py:155 to 0.2. The driver is now able to communicate with all BMS (see the log for an example). However, I have the strange issue now where the data is not correctly displayed on the dBus, and only 5 BMS are shown in the UI. Using dbus-spy, I was able to find out that BMS 4-9 do not have all the datapoints that should be sent on the dbus:
[image]

[image]

Could it be that the dBus service is overloaded somehow? I am really stuck here

### Installation method

None

### Driver version of the currently installed driver

2.0.20250317dev

### Venus OS device type

Cerbo GX

### Venus OS version

v3.60~39

### BMS type

Daly BMS

### Cell count

16

### Battery count

14

### Connection type

CAN

### Config file

```ini
[DEFAULT]

; --------- Set logging level ---------
; ERROR: Only errors are logged
; WARNING: Errors and warnings are logged
; INFO: Errors, warnings, and info messages are logged
; DEBUG: Errors, warnings, info, and debug messages are logged
LOGGING = DEBUG

; --------- Battery Current Limits ---------
; +++ Limits apply to each individual battery/BMS. +++
; +++ If you have multiple batteries, you need a battery aggregator. +++
[... 708 lines cut ...]
@4000000067d923020dbe0b24 DEBUG:SerialBattery:[can0] Received: ID=0x18974006, Daten=bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00')
@4000000067d923020dd7636c DEBUG:SerialBattery:[can0] Received: ID=0x18984006, Daten=bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00')
@4000000067d92302161e0684 DEBUG:SerialBattery:Data check: 2
@4000000067d92302183d9ac4 DEBUG:SerialBattery:logged to dbus [44.9]
@4000000067d92302193f2474 DEBUG:SerialBattery:Cells:[1]3.325V [2]3.325V [3]3.325V [4]3.326V [5]3.326V [6]3.325V [7]3.325V [8]3.325V [9]3.326V [10]3.325V [11]3.325V [12]3.326V [13]3.324V [14]3.326V [15]3.325V [16]3.324V 
@4000000067d92302194ba3ac DEBUG:SerialBattery:updating battery idx 0x07
@4000000067d9230219ff5794 DEBUG:SerialBattery:[can0] Received: ID=0x18904007, Daten=bytearray(b'\x02\x13\x00\x00u,\x03\x89')
@4000000067d923021a39ff34 DEBUG:SerialBattery:[can0] Received: ID=0x18914007, Daten=bytearray(b'\x0c\xfe\x0b\x0c\xfb\x02\x03\x89')
@4000000067d923021a8b094c DEBUG:SerialBattery:[can0] Received: ID=0x18924007, Daten=bytearray(b'@\x01>\x02\xfb\x02\x03\x89')
@4000000067d923021ae4162c DEBUG:SerialBattery:[can0] Received: ID=0x18934007, Daten=bytearray(b'\x00\x01\x01$\x00\x03\xdd\xd8')
@4000000067d923021b5dda0c DEBUG:SerialBattery:[can0] Received: ID=0x18944007, Daten=bytearray(b'\x10\x02\x00\x00\x00\x00\x00\xff')
@4000000067d923021b9b506c DEBUG:SerialBattery:[can0] Received: ID=0x18a64007, Daten=bytearray(b'\x01\x0c\xfc\x0c\xfb\x0c\xfc\xff')
@4000000067d923021bbdefb4 DEBUG:SerialBattery:[can0] Received: ID=0x18a74007, Daten=bytearray(b'\x02\x0c\xfc\x0c\xfc\x0c\xfc\xff')
@4000000067d923021bd2cb8c DEBUG:SerialBattery:[can0] Received: ID=0x18a84007, Daten=bytearray(b'\x03\x0c\xfb\x0c\xfd\x0c\xfd\xff')
@4000000067d923021bf5d834 DEBUG:SerialBattery:[can0] Received: ID=0x18a94007, Daten=bytearray(b'\x04\x0c\xfc\x0c\xfe\x0c\xfd\xff')
```

### Please confirm that you have completed the following steps before submitting your support request:

- [x] I have carefully read all the instructions above including the descriptions.
- [x] I have tried the `nightly` build first, if I was not already using it. This means the logs I provide are from the `nightly` build.
- [x] I have rebooted at least once after I saw the issue.
- [x] I have read the complete [changelog](https://github.com/mr-manuel/dbus-serialbattery/blob/master/CHANGELOG.md).
- [x] I have provided the logfiles in the correct format with a READABLE timestamp.
- [ ] I have clicked all checkboxes because I did not read the instructions.

mr-manuel, 2025-03-18:
I unchecked the checkboxes where you cheated. Once you fixed it I can proceed to help you.

mr-manuel, 2025-03-18:
Your device takes 1 second to process 4 batteries and probably 3.5 seconds to process all 14. Very likely your Cerbo GX device is too weak to handle all of the 14 batteries.

Did you check the CPU load?

- https://mr-manuel.github.io/venus-os_dbus-serialbattery_docs/faq/#how-to-troubleshoot-high-cpu-load
- https://mr-manuel.github.io/venus-os_dbus-serialbattery_docs/faq/#why-do-i-have-a-high-cpu-usage-after-installing-the-driver

BeSt-SP, 2025-03-18:
Yes, I noticed that it takes quite a while for the driver to process a battery. I did check the CPU using top: 

[image]

It's sitting at around 30%, so I guess that should not be a problem. I do also have a Ekrano here which should have two more CPU cores, I'll check if the issue is also present there.

mr-manuel, 2025-03-18:
Interesting, that with CAN the CPU load is that low. Can you contact me on Discord?

mr-manuel, 2025-03-19:
The dbus and CPU is overloaded. Increasing the `POLL_INTERVAL` to `60` would help, but is not that nice.

closed 2025-03-19 by mr-manuel