mr-manuel/venus-os_dbus-serialbattery #219: UnboundLocalError: local variable referenced before assignment

sy-filis, 2025-03-20:
### Describe the bug

After some time the service stop working due to the refering unassigned variable `penalty_sum` in file https://github.com/mr-manuel/venus-os_dbus-serialbattery/blob/f769aebeac3eb7a51b5f9c4e26ac3dd422d42eff/dbus-serialbattery/battery.py#L870

### How to reproduce

Maybe change from bulk to float.

### Expected behavior

No error 😂

### Installation method

Install script from the documentation

### Driver version of the currently installed driver

v2.0.20250207beta

### Venus OS device type

Cerbo GX

### Venus OS version

3.54

### BMS type

JKBMS Inverter (JKBMS PB Model)

### Cell count

16

### Battery count

1

### Connection type

Serial USB adapter to TTL

### Config file

```ini
[DEFAULT]

; --------- Set logging level ---------
; ERROR: Only errors are logged
; WARNING: Errors and warnings are logged
; INFO: Errors, warnings, and info messages are logged
; DEBUG: Errors, warnings, info, and debug messages are logged
LOGGING = DEBUG

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

MAX_BATTERY_CHARGE_CURRENT    = 200.0
MAX_BATTERY_DISCHARGE_CURRENT = 200.0

MIN_CELL_VOLTAGE   = 3.000
MAX_CELL_VOLTAGE   = 3.500
FLOAT_CELL_VOLTAGE = 3.375

; --------- SoC Reset Voltage (must match BMS settings) ---------
; +++ This section is independent and unrelated to the "SoC calculation" section below. +++
; This is used to reset the SoC to 100% periodically to correct SoC drift.
; Description:
[... 192 lines cut ...]
TELEMETRY = False
```

### Relevant log output

```shell
2025-03-19 20:05:27.613102500 DEBUG:SerialBattery:bytearray: \x4e\x57\x01\x21\x00\x00\x00\x00\x06\x00\x01\x79\x30\x01\x0c\xfd\x02\x0d\x02\x03\x0d\x02\x04\x0d\x00\x05\x0c\xff\x06\x0d\x00\x07\x0c\xff\x08\x0c\xfe\x09\x0d\x00\x0a\x0c\xff\x0b\x0c\xfc\x0c\x0c\xff\x0d\x0c\xfe\x0e\x0c\xfe\x0f\x0c\xff\x10\x0c\xff\x80\x00\x0d\x81\x00\x0b\x82\x00\x0b\x83\x14\xcc\x84\x06\xe7\x85\x5e\x86\x02\x87\x00\xae\x89\x00\x02\x20\xcf\x8a\x00\x10\x8b\x00\x00\x8c\x00\x03\x8e\x16\xd0\x8f\x10\xe0\x90\x0e\x42\x91\x0d\xac\x92\x00\x05\x93\x0a\x8c\x94\x0a\xbe\x95\x00\x05\x96\x01\x2c\x97\x00\xc8\x98\x01\x2c\x99\x00\xc8\x9a\x00\x78\x9b\x0d\xac\x9c\x00\x0a\x9d\x01\x9e\x00\x5a\x9f\x00\x46\xa0\x00\x64\xa1\x00\x64\xa2\x00\x14\xa3\x00\x46\xa4\x00\x46\xa5\xff\xec\xa6\xff\xf6\xa7\xff\xec\xa8\xff\xf6\xa9\x10\xaa\x00\x00\x03\x20\xab\x01\xac\x01\xad\x05\x91\xae\x01\xaf\x00\xb0\x00\x0a\xb1\x14\xb2\x31\x32\x33\x34\x35\x36\x00\x00\x00\x00\xb3\x00\xb4\x49\x6e\x70\x75\x74\x20\x55\x73\xb5\x32\x32\x31\x32\xb6\x00\x11\xdd\x9f\xb7\x31\x30\x2e\x58\x57\x5f\x53\x31\x30\x2e\x31\x30\x5f\x5f\x5f\xb8\x00\xb9\x00\x00\x03\x20\xba\x49\x6e\x70\x75\x74\x20\x55\x73\x65\x72\x64\x61\x5a\x75\x6b\x20\x46\x56\x20\x42\x4d\x53\x00\x00\xc0\x01\x00\x00\x00\x00\x68\x00\x00\x5a\x46
2025-03-19 20:05:27.639967500 Traceback (most recent call last):
2025-03-19 20:05:27.639976500   File "/data/apps/dbus-serialbattery/dbushelper.py", line 855, in publish_battery
2025-03-19 20:05:27.639980500     self.battery.manage_charge_voltage()
2025-03-19 20:05:27.639982500   File "/data/apps/dbus-serialbattery/battery.py", line 568, in manage_charge_voltage
2025-03-19 20:05:27.639986500     self.manage_charge_voltage_limit()
2025-03-19 20:05:27.639988500   File "/data/apps/dbus-serialbattery/battery.py", line 870, in manage_charge_voltage_limit
2025-03-19 20:05:27.640214500     + (f" • penalty_sum: {penalty_sum:.3f} V" if utils.CVL_CONTROLLER_MODE == 1 else "")
2025-03-19 20:05:27.640219500 UnboundLocalError: local variable 'penalty_sum' referenced before assignment
```

### Please confirm that you have completed the following steps before submitting your support request

- [x] I have carefully read all the instructions above including the descriptions.
- [x] I have tried the `nightly` build first, if I was not already using it. This means the logs I provide are from the `nightly` build.
- [x] I have rebooted at least once after I saw the issue.
- [x] I have read the complete [changelog](https://github.com/mr-manuel/dbus-serialbattery/blob/master/CHANGELOG.md).
- [x] I have provided the logfiles in the correct format with a HUMAN READABLE timestamp by piping it through `tai64nlocal`.
- [ ] I have clicked all checkboxes because I did not read the instructions.

mr-manuel, 2025-03-20:
The variable is only assigned, if charge mode is bulk/absorption and cvl controler mode is 1. Probably your driver started with float. I will fix that.

https://github.com/mr-manuel/venus-os_dbus-serialbattery/blob/f769aebeac3eb7a51b5f9c4e26ac3dd422d42eff/dbus-serialbattery/battery.py#L728-L737

mr-manuel, 2025-03-20:
Thanks for reporting!

commit 11db8d87 (2025-03-20): Fix #219
closed 2025-03-20 by mr-manuel