mr-manuel/venus-os_dbus-serialbattery #222: Daly new 100 percent BMS (blue) - any news?

kr0815, 2025-03-23:
I try to get one of these new Daly 100 percent BMS running under dbus-sb  - connected on RS485

i fpund some unsolved topics in the closed issues,like #163  but there is one comment from transistorgit

> Hi,
> 
> I have 2 100balance BMS and connected them successfully with serial battery on a raspi. I used a newer driver version, so maybe you want to give it another try.
>  
> Make sure to check the following items:
> * Wiring (reverse A/B for test)
> * termination resistor (configurable by BMS App)
> * Address in BMS App configured to "1"
> * protocol type in BMS configured to None/RS485
> * is the new daly-supplied RS485 converter compatible to cerbo? check syslog and lsusb if it is recognised.
> * newest driver version
> 
> best regards 

 _Originally posted by @transistorgit in [#163](https://github.com/mr-manuel/venus-os_dbus-serialbattery/issues/163#issuecomment-2648166653)_

are there others that had succes connecting these new BMS?  
What is meant with newest driver?  I did´t find anything in dbus-serialbattery addressing this problem

I could, like others, control the BMS with the PC Software, also the USB-RS485 interface works under Venus - tested with an older Daly BMS

kr0815, 2025-03-24:
I investigated firther in the problem, read a lot of different pages:

There musst be a difference between older and newer models:

https://github.com/softwarecrash   - there is a version specially for the 100balance series

thjey also speak about differences:  https://github.com/syssi/esphome-daly-bms

Could it be possible that these never series speak modbus over RS485?

Here is the protocol:  https://github.com/syssi/esphome-daly-bms/tree/main/docs

mr-manuel, 2025-03-24:
This is a discussion or BMS request and no bug, support request or enhancement :-)

Why do you open an issue?

transistorgit, 2025-03-24:
Can you post your exact setup?
I have 2 of the new 100balance devices and use them successfully with the original RS485 USB converters and a raspberry pi. We can compare device numbers/types and firmware versions to find differences. 

2 open questions still unanswered:
1. have all users with these problems a cerbo device or is there also someone with a pi? Maybe there is a compatibility issue between cerbo and the new RS485 dongle

2. In the bms settings, did you set something for RS485 protocol? The setting should probably be None.
[image] 

I think that it is possible that the dalys talk 2 different protocols on the same port. Otherwise the protocol setting (see above) would break the pc setting software that uses the same port. 

@mr-manuel is it possible to move this issue to the disscussion section? If not, we should close it and copy the text over. I think we need some more user feedback to gather more data.