mr-manuel/venus-os_dbus-serialbattery #227: JK-BMS  ERROR: No reply - returning [len:273/65024]

cracker0dks, 2025-03-27:
### Describe the problem

Hey, I don't get serial connection to my JK-BMS. Error:

My BMS Version:

[image]

What I tried:
- SerialBattery Version 1.6 / Latest and Nightly
- Swap RX / TX (But then I get nothing in return at all)
- I tried 3 different UART TTL USB Converters with no succes but this one should work afaik: https://www.duppa.net/shop/isolated-4-port-usb-to-ttl-uart-ftdi/?v=5f02f0889301 

Is this BMS Verison just not working? Im thankfull for any suggestion :)

### Installation method

None

### Driver version of the currently installed driver

v2.0.20250324dev

### Venus OS device type

Cerbo GX

### Venus OS version

v3.54

### BMS type

JKBMS (Heltec BMS)

### Cell count

16

### Battery count

1

### Connection type

Serial USB adapter to TTL

### Config file

```ini
LOGGING = INFO
BMS_TYPE = Jkbms
```

### Relevant log output

```shell
2025-03-27 11:30:01.460938500 INFO:SerialBattery:Starting dbus-serialbattery
2025-03-27 11:30:01.464404500 INFO:SerialBattery:Venus OS v3.54 running on Cerbo GX
2025-03-27 11:30:01.465140500 INFO:SerialBattery:dbus-serialbattery v2.0.20250324dev
2025-03-27 11:30:17.477203500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2025-03-27 11:30:17.478068500 INFO:SerialBattery:Testing Jkbms
2025-03-27 11:30:33.572209500 ERROR:SerialBattery:>>> ERROR: No reply - returning [len:273/65024]
2025-03-27 11:30:34.081936500 INFO:SerialBattery:-- Testing BMS: 2 of 3 rounds
2025-03-27 11:30:34.082665500 INFO:SerialBattery:Testing Jkbms
2025-03-27 11:30:50.171175500 ERROR:SerialBattery:>>> ERROR: No reply - returning [len:277/65024]
2025-03-27 11:30:50.683467500 INFO:SerialBattery:-- Testing BMS: 3 of 3 rounds
2025-03-27 11:30:50.683478500 INFO:SerialBattery:Testing Jkbms
2025-03-27 11:31:06.786454500 ERROR:SerialBattery:>>> ERROR: No reply - returning [len:271/65024]
2025-03-27 11:31:07.299690500 ERROR:SerialBattery:ERROR >>> No battery connection at /dev/ttyUSB3
2025-03-27 11:31:07.299700500 INFO:SerialBattery:Exit signal received, exiting gracefully...
2025-03-27 11:31:07.299703500 INFO:SerialBattery:Stopped dbus-serialbattery with exit code 1
```

### Please confirm that you have completed the following steps before submitting your support request

- [x] I have carefully read all the instructions above including the descriptions.
- [x] I have tried the `nightly` build first, if I was not already using it. This means the logs I provide are from the `nightly` build.
- [x] I have rebooted at least once after I saw the issue.
- [x] I have read the complete [changelog](https://github.com/mr-manuel/dbus-serialbattery/blob/master/CHANGELOG.md).
- [x] I have provided the logfiles in the correct format with a HUMAN READABLE timestamp by piping it through `tai64nlocal`.
- [ ] I have clicked all checkboxes because I did not read the instructions.

mr-manuel, 2025-04-05:
If the serial connection is not working there are only a few options:

1. Cabeling is wrong
2. JKBMS RS485 adapter is the wrong one (sometimes the sellers ship the wrong one). There are two different, one for 8S and one for the others. You could try to connect without the JKBMS 485 adapter. Make sure to check the instructions in the docs to not damage anything
3. Your BMS is faulty

Please post some pictures where we can see how you connected your cable/adapter to the BMS TTL port, then how you connected the adapter or cable to the USB to serial adapter. We need to be able to understand and see how you connected which cable to which pin, cable or anything else.

cracker0dks, 2025-04-16:
I used a direct connection for my tests so far (without JKBMS 485 adapter)  but now also bought the adapter and tried to connect the adapter to my UART TTL USB Converters and got a "Checksum Error" then I tried it this way:

[image]

But I dont get a new ttyUSB I still only see ttyUSB0 and ttyUSB3 in the log: ``tail -F -n 100 /data/log/dbus-serialbattery.ttyUSB0/current | tai64nlocal`` and they are not seeing the JK BMS. Is my BMS GPS Port broken? Thanks for your help

closed 2025-04-23 by cracker0dks