mr-manuel/venus-os_dbus-serialbattery #234: 0% SOC in VRM when check for firmware updates

Newti04, 2025-03-31:
### Is your feature request related to a problem? Please describe.

Hi,
when I select in VRM 'Geräteliste' - 'Firmware-Aktualisierung' and the system checks for new firmware, the serial-connections disconnects for a short time. 
I think in this time dbus-serialbattery sends a 0% SOC to VRM and there is always in the VRM graphic the range for the battery SOC from 0 to the real max SOC.
Is it possible, to suppress this temporary 0% SOC?
Best regards
Claus

### Describe the solution you'd like

add a break in set the SOC if the serial connection is stopped.

mr-manuel, 2025-04-05:
Can you provide a screen record where you show this? I was not able to reproduce it with different systems.

Newti04, 2025-04-06:
[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

When I make the pictures, I saw, that bus-serialbattery shows 'Nicht angeschlossen' during Firmware-scan and 'Battery Aggregator' shows 0%
So I go to dbus-spy and found out, that the entries for bus-serialbattery were gone, during the scan was running. And in the scan-time Battery-Aggregator shows 0%.
Is it right, that bus-serialbattery de-registers in dbus, when the serial connection is stopped?

mr-manuel, 2025-04-12:
I was now able to reproduce it. It’s a Venus OS problem/design as the VE.Direct devices get disconnected when checking the firmware. Probably the serial service/connections get stopped so that it can check for firmware of the devices without interfering communication.

Ask in the Victron Community about that, as also Victron VE.Direct devices as BlueSolar MPPT and SmartShunt disconnect on firmware check.

closed 2025-04-12 by mr-manuel