mr-manuel/venus-os_dbus-serialbattery #235: unsupported format string passed to NoneType.__format__ with GUI_PARAMETERS_SHOW_ADDITIONAL_INFO = True option

marcelrv, 2025-03-31:
### Describe the bug

I'm getting `TypeError('unsupported format string passed to NoneType.__format__')` errors when uning the GUI_PARAMETERS_SHOW_ADDITIONAL_INFO = True option.

When an update fails for the seplos battery (checksum error), no values are passed to the battery.py. 
Which causes the error to be triggered as it tries to format a non value.

Don't know if there is a better way for the seplos3 driver to tell no update has been done, or that the printing
https://github.com/mr-manuel/venus-os_dbus-serialbattery/blob/master/dbus-serialbattery/battery.py#L857-L921 needs to be surrounded by an additional try/catch and ignored

It seems in those cases also to trigger a ` #8 Internal calculation error` which is much more annoying  as I need to click it away in the portal

```
@4000000067e686e22464cb9c INFO:SerialBattery:Error getting data Checksum error in rtu mode: b'\xec\xec' instead of b'\x8d\xe2' . The response is: b'\x00\x04\xd4\x00\x04\x00\x10\x15\x18\x15\xe0\x15\x18\x16\                                                       x80\x12\xc0\x12 \x12\xc0\x10\xe0\r\xc8-T\x0c\x1c\n\x8c\x07\xd0\x03\xe8\x01\xf4\x00\xcb\x00\xcd\x00\xd2\x00d\x01,\x01,\xff5\xff3\xff.\x00d\xfe\xa2\x01,\xfe\xd4\x00j\x02X\x00\x05\x0b\xb8\x00\xcd\x00\n\r\xac\                                                       rH\x05\xdc\x03\xe8\x03 \x00\xc8\x00\x1e\x0c\x81\x0c\x9f\x0c\x9f\x0c\xd1\n\xdd\n\xbf\n\xab\nG\x0c\x9f\x0c\xd1\x0c\xd1\r\x03\n\xc9\nG\n\xab\n\x15\x0c\x81\x0c\x9f\x0c\xd1\r\x03\n\xc9\n\xab\n\xab\nG\r\xfd\x0ea                                                       \r\xfd\x0e\xf7\x0b\x0f\n\xab\x0c\x9f\n\xab\x00\n\rH\x002\x00\x1e\x03\xc0\x00\x96\x00d\x00F\x002qHv\xf6,\xd9\x000\x00<\x00\xf0\x00\n\x00\x07\x00\x00\x00\r\x00\x00\x01\xf4\x01,\x16\x1c\x00\xb4\xffL\x00\x05\x                                                       00\x08\xec\xec' (plain response: b'\x00\x04\xd4\x00\x04\x00\x10\x15\x18\x15\xe0\x15\x18\x16\x80\x12\xc0\x12 \x12\xc0\x10\xe0\r\xc8-T\x0c\x1c\n\x8c\x07\xd0\x03\xe8\x01\xf4\x00\xcb\x00\xcd\x00\xd2\x00d\x01,\                                                       x01,\xff5\xff3\xff.\x00d\xfe\xa2\x01,\xfe\xd4\x00j\x02X\x00\x05\x0b\xb8\x00\xcd\x00\n\r\xac\rH\x05\xdc\x03\xe8\x03 \x00\xc8\x00\x1e\x0c\x81\x0c\x9f\x0c\x9f\x0c\xd1\n\xdd\n\xbf\n\xab\nG\x0c\x9f\x0c\xd1\x0c\                                                       xd1\r\x03\n\xc9\nG\n\xab\n\x15\x0c\x81\x0c\x9f\x0c\xd1\r\x03\n\xc9\n\xab\n\xab\nG\r\xfd\x0ea\r\xfd\x0e\xf7\x0b\x0f\n\xab\x0c\x9f\n\xab\x00\n\rH\x002\x00\x1e\x03\xc0\x00\x96\x00d\x00F\x002qHv\xf6,\xd9\x000\                                                       x00<\x00\xf0\x00\n\x00\x07\x00\x00\x00\r\x00\x00\x01\xf4\x01,\x16\x1c\x00\xb4\xffL\x00\x05\x00\x08\xec\xec')
@4000000067e686e2246ceda4 INFO:SerialBattery:Updating Seplos v3 BMS16S200A-SP05B SP05B2311260061 failed: [False, False, False, False, False]
@4000000067e686e2248645ec ERROR:SerialBattery:Non blocking exception occurred: TypeError('unsupported format string passed to NoneType.__format__') of type <class 'TypeError'> in /data/apps/dbus-serialbatt                                                       ery/battery.py line #868
```

### How to reproduce

- Seplosv3 driver
- communication error (seems to happen occasionally)
- formatting error &  #8 Internal calculation error

### Expected behavior

Expected:
1) Checksum error in the log
2) Gracefully handle the non-update.
3) no further errors displayed: no formatting error, no ' internal calculation error' 

### Installation method

SFTP upload

### Driver version of the currently installed driver

v2.0.20250324dev

### Driver version of the last known working driver

v2.0.20250324dev

### Venus OS device type

Raspberry Pi Zero 2W

### Venus OS version

3.60~48

### BMS type

Seplos v3

### Cell count

16

### Battery count

1

### Connection type

Serial USB adapter to RS485

### USB hub

Yes, with external power supply
[... 85 lines cut ...]
@4000000067e686af1d83792c INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 150.0 A | MAX BATTERY DISCHARGE CURRENT: 160.0 A
@4000000067e686af1d8baad4 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 180 A | MAX BATTERY DISCHARGE CURRENT: 180 A (read from BMS)
@4000000067e686af1d9382a4 INFO:SerialBattery:> CVCM:       True
@4000000067e686af1d9b950c INFO:SerialBattery:> CCCM CV:    True  | DCCM CV:       True
@4000000067e686af1da50ed4 INFO:SerialBattery:> CCCM T:     True  | DCCM T:        True
@4000000067e686af1dad1d54 INFO:SerialBattery:> CCCM T MOS: True  | DCCM T MOS:    True
@4000000067e686af1db654b4 INFO:SerialBattery:> CCCM SOC:   False | DCCM SOC:      False
@4000000067e686af1dc187e4 INFO:SerialBattery:> CHARGE FET: True  | DISCHARGE FET: True | BALANCE FET: None
@4000000067e686af1dcaee24 INFO:SerialBattery:Serial Number/Unique Identifier: SP05B2311260061

after error:

@4000000067e686e524ab13cc INFO:SerialBattery:Error getting data Checksum error in rtu mode: b'\xec\xec' instead of b'Z\xad' . The response is: b'\x00\x04\xd4\x00\x04\x00\x10\x15\x18\x15\xe0\x15\x18\x16\x80\x12\xc0\x12 \x12\xc0\x10\xe0\rH\r\xac\rH\x0eB\x0c\x1c\x0bT\x0c\x1c\n\x8c\x07\xd0\x03\xe8\x01\xf4\x00\xcb\x00\xcd\x00\xd2\x00d\x01,\x01,\xff5\xff3\xff.\x00d\xfe\xa2\x01,\xfe\xd4\x00j\x02X\x00\x05\x0b\xb8\x00\xcd\x00\n\r\xac\rH\x05\xdc\x03\xe8\x03 \x00\xc8\x00\x1e\x0c\x81\x0c\x9f\x0c\x9f\x0c\xd1\n\xdd\n\xbf\n\xab\nG\x0c\x9f\x0c\xd1\x0c\xd1\r\x03\n\xc9\nG\n\xab\n\x15\x0c\x81\x0c\x9f\x0c\xd1\r\x03\n\xc9\n\xab\n\xab\nG\r\xfd\x0ea\r\xfd\x0e\xf7\x0b\x0f\n\xab\x0c\x9f\n\xab\x00\n\rH\x002\x00\x1e\x03\xc0\x00\x96\x00d\x00\xfe\x070\x00<\x00\xf0\x00\n\x00\x07\x00\x00\x00\r\x00\x00\x01\xf4\x01,\x16\x1c\x00\xb4\xffL\x00\x05\x00\x08\xec\xec' (plain response: b'\x00\x04\xd4\x00\x04\x00\x10\x15\x18\x15\xe0\x15\x18\x16\x80\x12\xc0\x12 \x12\xc0\x10\xe0\rH\r\xac\rH\x0eB\x0c\x1c\x0bT\x0c\x1c\n\x8c\x07\xd0\x03\xe8\x01\xf4\x00\xcb\x00\xcd\x00\xd2\x00d\x01,\x01,\xff5\xff3\xff.\x00d\xfe\xa2\x01,\xfe\xd4\x00j\x02X\x00\x05\x0b\xb8\x00\xcd\x00\n\r\xac\rH\x05\xdc\x03\xe8\x03 \x00\xc8\x00\x1e\x0c\x81\x0c\x9f\x0c\x9f\x0c\xd1\n\xdd\n\xbf\n\xab\nG\x0c\x9f\x0c\xd1\x0c\xd1\r\x03\n\xc9\nG\n\xab\n\x15\x0c\x81\x0c\x9f\x0c\xd1\r\x03\n\xc9\n\xab\n\xab\nG\r\xfd\x0ea\r\xfd\x0e\xf7\x0b\x0f\n\xab\x0c\x9f\n\xab\x00\n\rH\x002\x00\x1e\x03\xc0\x00\x96\x00d\x00\xfe\x070\x00<\x00\xf0\x00\n\x00\x07\x00\x00\x00\r\x00\x00\x01\xf4\x01,\x16\x1c\x00\xb4\xffL\x00\x05\x00\x08\xec\xec')
@4000000067e686e524b2f36c INFO:SerialBattery:Updating Seplos v3 BMS16S200A-SP05B SP05B2311260061 failed: [False, False, False, False, False]
@4000000067e686e524cf550c ERROR:SerialBattery:Non blocking exception occurred: TypeError('unsupported format string passed to NoneType.__format__') of type <class 'TypeError'> in /data/apps/dbus-serialbattery/battery.py line #868
```

### Please confirm that you have completed the following steps before submitting your support request

- [x] I have carefully read all the instructions above including the descriptions.
- [x] I have tried the `nightly` build first, if I was not already using it. This means the logs I provide are from the `nightly` build.
- [x] I have rebooted at least once after I saw the issue.
- [x] I have read the complete [changelog](https://github.com/mr-manuel/dbus-serialbattery/blob/master/CHANGELOG.md).
- [x] I have provided the logfiles in the correct format with a HUMAN READABLE timestamp by piping it through `tai64nlocal`.
- [ ] I have clicked all checkboxes because I did not read the instructions.

marcelrv, 2025-03-31:
note that the seplos driver is returning ' false' for the refresh I think, which should avoid getthing further errors I was expecting
https://github.com/mr-manuel/venus-os_dbus-serialbattery/blob/master/dbus-serialbattery/bms/seplosv3.py#L291-L292

mr-manuel, 2025-04-05:
Please fix the not followed instructions. I unchecked the one you missed.

closed 2025-04-27 by mr-manuel