mr-manuel/venus-os_dbus-serialbattery #237: Temp shown in Display not on WebUI

DIVERSDIVERS, 2025-04-01:
### Describe the bug

After Update Temp not longer visible over Web Interface only on local Display

### How to reproduce

- update to 1.6.20250131
- restart

[image]

[image]

### Expected behavior

should look like this

[image]

### Installation method

Install script from the documentation

### Driver version of the currently installed driver

v1.6.20250131

### Driver version of the last known working driver

v1.5.20241215

### Venus OS device type

Cerbo GX

### Venus OS version

v3.54

### BMS type

Daly BMS

### Cell count

16

### Battery count

5

### Connection type

Serial USB adapter to TTL

### USB hub

Yes, no external power supply

### Config file

```ini
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):

LOGGING = ERROR

MAX_BATTERY_CHARGE_CURRENT = 500.0
MAX_BATTERY_DISCHARGE_CURRENT = 500.0

MIN_CELL_VOLTAGE   = 2.750
MAX_CELL_VOLTAGE   = 3.575
FLOAT_CELL_VOLTAGE = 3.405

SOC_RESET_VOLTAGE  = 3.575

SOC_RESET_AFTER_DAYS = 5

BLOCK_ON_DISCONNECT = False

LINEAR_LIMITATION_ENABLE = True

LINEAR_RECALCULATION_EVERY = 60

LINEAR_RECALCULATION_ON_PERC_CHANGE = 5

CVCM_ENABLE = True

CELL_VOLTAGE_DIFF_KEEP_MAX_VOLTAGE_UNTIL = 0.025

CELL_VOLTAGE_DIFF_KEEP_MAX_VOLTAGE_TIME_RESTART = 0.035

CELL_VOLTAGE_DIFF_TO_RESET_VOLTAGE_LIMIT = 0.075

MAX_VOLTAGE_TIME_SEC = 300

SOC_LEVEL_TO_RESET_VOLTAGE_LIMIT = 95

CVL_ICONTROLLER_MODE   = True
CVL_ICONTROLLER_FACTOR = 0.2

CCCM_CV_ENABLE = True

DCCM_CV_ENABLE = True

CELL_VOLTAGES_WHILE_CHARGING   = 3.60, 3.585, 3.580, 3.565, 3.555, 3.545, 3.535, 3.505
MAX_CHARGE_CURRENT_CV_FRACTION =    0,  0.01, 0.015, 0.025,  0.05,   0.1,  0.15,     1

CELL_VOLTAGES_WHILE_DISCHARGING   = 2.75, 2.85, 2.90, 2.95, 3.00
MAX_DISCHARGE_CURRENT_CV_FRACTION =    0, 0.02, 0.25,  0.5,    1

CCCM_T_ENABLE = True

DCCM_T_ENABLE = True

TEMPERATURES_WHILE_CHARGING      = 0,   3,   7,  10,  15,  40,  45,  50,  55, 60
MAX_CHARGE_CURRENT_T_FRACTION    = 0, 0.2, 0.5, 0.8,   1,   1, 0.8, 0.5, 0.2,  0

TEMPERATURES_WHILE_DISCHARGING   = -35,  -25, -10,   0,  5,  40,  50,  55, 60
MAX_DISCHARGE_CURRENT_T_FRACTION =   0, 0.4,  0.6, 0.8,  1,   1, 0.5, 0.2,  0

CCCM_SOC_ENABLE = True

DCCM_SOC_ENABLE = True

SOC_WHILE_CHARGING                 =  100,   99,   98,   97,   96
MAX_CHARGE_CURRENT_SOC_FRACTION    = 0.02, 0.10, 0.15, 0.25, 1.00

SOC_WHILE_DISCHARGING              =    5,   10,   12,   15
MAX_DISCHARGE_CURRENT_SOC_FRACTION = 0.01, 0.10, 0.20, 1.00

TIME_TO_GO_ENABLE = True

TIME_TO_SOC_VALUE_TYPE = 30

BMS_TYPE = Daly

BATTERY_CELL_DATA_FORMAT = 1

AUTO_RESET_SOC = True

TELEMETRY = False

SOC_LOW_WARNING = 15
SOC_LOW_ALARM   = 10

BATTERY_CAPACITY = 302

INVERT_CURRENT_MEASUREMENT = -1
```

### Relevant log output

```shell
root@einstein:~# tail -F -n 100 /data/log/dbus-serialbattery.ttyUSB0/current | tai64nlocal
2025-04-01 12:18:05.324463500 *** CCGX booted (0) ***
2025-04-01 12:53:32.394457500 *** CCGX booted (0) ***
```

### Please confirm that you have completed the following steps before submitting your support request

- [x] I have carefully read all the instructions above including the descriptions.
- [x] I have tried the `nightly` build first, if I was not already using it. This means the logs I provide are from the `nightly` build.
- [x] I have rebooted at least once after I saw the issue.
- [x] I have read the complete [changelog](https://github.com/mr-manuel/dbus-serialbattery/blob/master/CHANGELOG.md).
- [x] I have provided the logfiles in the correct format with a HUMAN READABLE timestamp by piping it through `tai64nlocal`.
- [x] I have clicked all checkboxes because I did not read the instructions.

mr-manuel, 2025-04-01:
Filed the issue to Victron https://github.com/victronenergy/gui-v2/issues/2013

mr-manuel, 2025-04-04:
The upstream fix was merged now.

closed 2025-04-04 by mr-manuel