mr-manuel/venus-os_dbus-serialbattery #238: Differing temperature units

pkeberly, 2025-04-01:
### Describe the bug

Temperature values in History are appear to be in °C units but are displayed in °F
Temperature values in Parameters are display in °C.  but GUI units are in °F

[image]
[image]

   

### How to reproduce

View battery history and parameters

### Expected behavior

Use GUI units for values.

### Installation method

Install script from the documentation

### Driver version of the currently installed driver

2.0.20250324dev

### Venus OS device type

Cerbo GX

### Venus OS version

v3.60~48 - v3.60~53

### BMS type

Jiabaida JDB BMS (LLT Power, Overkill Solar, Smart BMS)

### Cell count

8

### Battery count

1

### Connection type

Serial USB adapter to TTL

### USB hub

Yes, no external power supply

### Config file

```ini
[DEFAULT]

;------------ Setup for Luna's 8S energy battery ---------
;  12/16/2024 PWK  first connections, 300aH test
;  12/17/2024 PWK  driver update, bump top voltage to 3.52, tune charge ramp to get better balancing.
;  12/22/2024 PWK  lowered CCL/DCL Recovery Threshold to 0.8% (1.2A)
;  01/09/2025 PWK  Updated driver, changed config.ini to match
;  03/26/2025 PWK  Swapped to Cerbo-S after USB failure. changed config.ini to match latest drvr.

; --------- Set logging level ---------
; ERROR: Only errors are logged
; WARNING: Errors and warnings are logged
; INFO: Errors, warnings, and info messages are logged
[... 477 lines cut ...]
2025-04-01 13:43:06.385052500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 200.0 A | MAX BATTERY DISCHARGE CURRENT: 300.0 A (read from BMS)
2025-04-01 13:43:06.385718500 INFO:SerialBattery:> CVCM:       True
2025-04-01 13:43:06.386413500 INFO:SerialBattery:> CCCM CV:    True  | DCCM CV:       True
2025-04-01 13:43:06.387289500 INFO:SerialBattery:> CCCM T:     True  | DCCM T:        True
2025-04-01 13:43:06.388192500 INFO:SerialBattery:> CCCM T MOS: True  | DCCM T MOS:    True
2025-04-01 13:43:06.389136500 INFO:SerialBattery:> CCCM SOC:   False | DCCM SOC:      False
2025-04-01 13:43:06.390145500 INFO:SerialBattery:> CHARGE FET: True  | DISCHARGE FET: True | BALANCE FET: True
2025-04-01 13:43:06.391328500 INFO:SerialBattery:Serial Number/Unique Identifier: 39_304.0Ah
2025-04-01 13:43:53.110038500 ERROR:SerialBattery:>>> ERROR: No reply - returning [len:40/41]
2025-04-01 13:43:53.110046500 INFO:SerialBattery:cell_voltages_good: False - min: 3.242 > 3.25 - max: 3.246 < 3.35
2025-04-01 13:43:53.359915500 ERROR:SerialBattery:>>> ERROR: BMS send insufficient data. Received 50 expected 45
2025-04-01 13:43:55.416723500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2025-04-01 13:43:55.418389500 INFO:SerialBattery:cell_voltages_good: False - min: 3.243 > 3.25 - max: 3.246 < 3.35
2025-04-01 13:44:14.226297500 ERROR:SerialBattery:>>> ERROR: No reply - returning [len:39/44]
2025-04-01 13:44:14.234226500 INFO:SerialBattery:cell_voltages_good: False - min: 3.242 > 3.25 - max: 3.246 < 3.35
```

### Please confirm that you have completed the following steps before submitting your support request

- [x] I have carefully read all the instructions above including the descriptions.
- [x] I have tried the `nightly` build first, if I was not already using it. This means the logs I provide are from the `nightly` build.
- [x] I have rebooted at least once after I saw the issue.
- [x] I have read the complete [changelog](https://github.com/mr-manuel/dbus-serialbattery/blob/master/CHANGELOG.md).
- [x] I have provided the logfiles in the correct format with a HUMAN READABLE timestamp by piping it through `tai64nlocal`.
- [ ] I have clicked all checkboxes because I did not read the instructions.

mr-manuel, 2025-04-12:
The issue in the parameters page was fixed. The issue on the history page has to be fixed in the upstream code.

For reference only: https://github.com/victronenergy/gui-v2/issues/2040

Please do not comment on the linked issue. Once it's fixed in the upstream, I will merge it in the mod and then update this issue here.

pkeberly, 2025-04-21:
Parameter values are correct.
What do the red cell voltages indicate?

[image]

mr-manuel, 2025-04-29:
The issue was fixed upstream and merged into the mod. You can install the new GUIv2 with
```bash
/data/apps/dbus-serialbattery/custom-gui-install.sh
```

closed 2025-04-29 by mr-manuel

mr-manuel, 2025-05-04:
Is this via browser or on the local display? If in the browser clear your cache. On all my four test system it works.