mr-manuel/venus-os_dbus-serialbattery #239: Unable to connect to JK BMS over Rs485

StartledP, 2025-04-01:
### Describe the problem

Im unable to connect to my JK BMS over RS485.  Im using the official adapter (not the 8s version) with a CH340 RS485 to USB converter.  The USB adapter is seen in Venus OS, but the JKBMS has no connection.  Everything is at default, the RS485 adapter is connected to a hub on usb1 of my pi2 zerow.  

Running tail -F -n 100 /data/log/serial-starter/current | grep dbus-serialbattery | tai64nlocal gives no output at all, it just hangs.

Any help is much appreciated.

### Installation method

Install script from the documentation

### Driver version of the currently installed driver

v1.6.20250131

### Venus OS device type

Raspberry Pi Zero 2W

### Venus OS version

v3.52

### BMS type

JKBMS (Heltec BMS)

### Cell count

16

### Battery count

1

### Connection type

Serial USB adapter to RS485

### USB hub

Yes, no external power supply

### Config file

```ini
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you>
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting>
; MAX_BATTERY_CHARGE_CURRENT = 50.0
; MAX_BATTERY_DISCHARGE_CURRENT = 60.0
```

### Relevant log output

```shell
root@raspberrypi2:/data/etc/dbus-serialbattery# tail -F -n 100 /data/log/dbus-serialbattery.ttyUSB0/current | tai64nlocal
2025-04-01 19:14:43.561523500 INFO:SerialBattery:|- on slave address 0
2025-04-01 19:14:44.368840500 INFO:SerialBattery:|- on slave address 1
2025-04-01 19:14:45.184819500 INFO:SerialBattery:|- on slave address 2
2025-04-01 19:14:46.000791500 INFO:SerialBattery:|- on slave address 3
2025-04-01 19:14:46.816255500 INFO:SerialBattery:|- on slave address 4
2025-04-01 19:14:47.632186500 INFO:SerialBattery:|- on slave address 5
2025-04-01 19:14:48.448046500 INFO:SerialBattery:|- on slave address 6
2025-04-01 19:14:49.263158500 INFO:SerialBattery:|- on slave address 7
2025-04-01 19:14:50.078906500 INFO:SerialBattery:|- on slave address 8
[... 135 lines cut ...]
root@raspberrypi2:/data/etc/dbus-serialbattery# lsusb
Bus 001 Device 005: ID 0403:6015 Future Technology Devices International, Ltd Bridge(I2C/SPI/UART/FIFO)
Bus 001 Device 004: ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC
Bus 001 Device 003: ID 1a86:7523 QinHeng Electronics HL-340 USB-Serial adapter
Bus 001 Device 002: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

tail -F -n 100 /data/log/serial-starter/current | grep dbus-serialbattery | tai64nlocal
Hangs - no output
```

### Any other information that may be helpful

(https://www.amazon.nl/-/en/USB-Hub-3-0-Portable-Indicator/dp/B0BLRZWTGM/ref=sr_1_1?crid=3KWSHKC2KNVAR&dib=eyJ2IjoiMSJ9.AMLLw-s0T7Uhz3jGJl02vpDCCqjUkTrgOPvoNVl2YJ1Ui4r82ww9GsKTfRpMsuu_syrhiiVJhryxQINBEIgtX1_JhmPvFmQR9wEjpRKiSGGgFnKj_-XTAuLnLy1FaIgERr8ASeJCPsc-9xK7d1OKFODV6_EcjfHbYfD3ojtyWPteiC1GrPB3m_P21loQFiI9V1FT4VFE1zbzyv8Q1sadpcs6fEtGJcniIblBQ1ke2eAlefnOwGNJsFiTlNR_88bV9TX2smjxnkFYspaCLFFoH9WvCGpLh9wUtlwKj8LUl-TJXxgCht1C3rcT4cO-Tynj_izCs8dDucU2QHDTpNDQSrj2SNkuH-RCHIm-7OoRYQ1Y9Bq0hQHOAuNdcXN3BDoqJYUBM5LVQGK5tgAw63sSK7WKKnaNf7DqYp5IkU08MEqfKkWDlFFkqYO8Fdbcxn_J.rThQ55PKA-ppKjNuN9nwoi3Pl-fc7na3WP1yaj_0__U&dib_tag=se&keywords=USB+Hub+3.0+-+iDsonix+4-Port+USB&qid=1743535933&sprefix=usb+hub+3.0+-+idsonix+4-port+usb%2Caps%2C73&sr=8-1)

I have a USB to TTL adapter connected to a ET112 on the same hub as the USB RS485 to JK BMS RS485  and that works perfectly.

### Please confirm that you have completed the following steps before submitting your support request

- [x] I have carefully read all the instructions above including the descriptions.
- [x] I have tried the `nightly` build first, if I was not already using it. This means the logs I provide are from the `nightly` build.
- [x] I have rebooted at least once after I saw the issue.
- [x] I have read the complete [changelog](https://github.com/mr-manuel/dbus-serialbattery/blob/master/CHANGELOG.md).
- [x] I have provided the logfiles in the correct format with a HUMAN READABLE timestamp by piping it through `tai64nlocal`.
- [ ] I have clicked all checkboxes because I did not read the instructions.

Albarge, 2025-04-01:
Have you tried swapping A + B connections around? Is it a long run / unshielded next to power cables?

StartledP, 2025-04-01:
> Have you tried swapping A + B connections around? Is it a long run / unshielded next to power cables?

Thanks for the quick reply. 

Yes but to no avail unfortunately.  Its a very short run as its using the JK BMS adapter, not running near any power cables beyond the battery cables.  Ill try a different RS485 to USB adapter.

StartledP, 2025-04-01:
Same issue with a different USB adapter, removing the hub also doesnt help.

Albarge, 2025-04-01:
I don't know what the exceptions in the log file refer to, maybe something else happened. Or does the Heltec bluetooth app have an option to turn on or off RS485 or select the TTL protocol like the JK app does, don't know if it's on by default or not can't remember with mine?

StartledP, 2025-04-01:
I had made one mistake, I was using the 1234 password instead of 123456 which is the default for my BMS.  no change in connectivity though.

The BMS is set to use RS485 in the JK app instead of CAN.

StartledP, 2025-04-01:
Im pretty sure its looking on the wrong USB port, as its trying to use usb0 instead of 1.  Is there a way to change this?

Tomashaglund1, 2025-04-02:
Here's instructions https://mr-manuel.github.io/venus-os_dbus-serialbattery_docs/next/troubleshoot/ for force another USB

StartledP, 2025-04-03:
Thanks for the reply, 

I ran 

rm /data/var/lib/serial-starter/* and rebooted

and after that did work I ran:

echo "sbattery" > /data/var/lib/serial-starter/ttyUSB1 and rebooted

however its still not connecting.  The password is default.

lsusb:

Bus 001 Device 005: ID 1a86:7523 QinHeng Electronics HL-340 USB-Serial adapter

No connection on USB1:

2025-04-01 21:45:51.756584500 ERROR:SerialBattery:ERROR >>> No battery connection at /dev/ttyUSB1

Running  head /data/var/lib/serial-starter/* shows the sbattery on two ports

==> /data/var/lib/serial-starter/ttyUSB0 <==
cgwacs

==> /data/var/lib/serial-starter/ttyUSB1 <==
sbattery

==> /data/var/lib/serial-starter/ttyUSB2 <==
sbattery

I have heard the BMS beep which I assume means it was attempting to connect.  The RX and TX lights flash on the USB adapter.

StartledP, 2025-04-03:
root@raspberrypi2:~# svstat /service/dbus-serialbattery.tty*
/service/dbus-serialbattery.ttyUSB2: up (pid 8189) 13 seconds

I have set it to USB1 multiple times, but its not taking hold.

The driver is clearly crashing and fixed to run on the wrong port.  Im unable to uninstall the driver as the directory for the ininstall.sh script is not present.

StartledP, 2025-04-03:
Tried to look into the config.ini file but found the directory is missing, listing the /etc directory shows just the ssl and timestamp folders visible.  Not sure what that uninstall script did but it looks like Ill be rebuilding the pi.

mr-manuel, 2025-04-05:
The PIN has nothing to do with the serial connection. If the serial connection is not working there are only a few options:

1. Cabeling is wrong
2. JKBMS RS485 adapter is the wrong one (sometimes the sellers ship the wrong one). There are two different, one for 8S and one for the others. You could try to connect without the JKBMS 485 adapter. Make sure to check the instructions in the docs to not damage anything
3. Your BMS is faulty

Please post some pictures where we can see how you connected your cable/adapter to the BMS TTL port, then how you connected the adapter or cable to the USB to serial adapter. We need to be able to understand and see how you connected which cable to which pin, cable or anything else.

StartledP, 2025-04-15:
Thanks for the help, got it working briefly after a full reinstall and finally getting the driver assigned to a single USB port (it most definitely wasnt just a cabling issue), but unfortunately after 5 minutes the BMS died.  Tried connecting a second BMS but that died as well. I suspect a problem with the RS485 or the RS232 adapter.  Both appeared to be wired correctly and it did work for a while but its so tricky with clamping such tiny wires (although they were firmly clamped).

Bought a third new BMS and a Smartshunt.  This was a very expensive and time consuming exercise, I would advise others to consider very carefully if this a path they want to go down before attempting to set this solution up.

mr-manuel, 2025-04-27:
Just follow carefully the instructions. If have no clue what you are doing, then it's very likely you break something.

commit 5a93013b (2023-04-20): fix #239
by @TimGFoley
commit 520eedc9 (2023-04-20): fix #239
by @TimGFoley
commit d8a5d643 (2023-04-26): Some fixes

* Changed: Fix System Switch in IO page
* Changed: Fix #239 by @TimGFoley
* Changed: Fix #311
* Changed: Run code analyse only on code change
* Changed: Separate Time-To-Go and Time-To-SoC activation
* Changed: fix UnicodeDecodeError https://github.com/Louisvdw/dbus-serialbattery/discussions/485#discussioncomment-5694781
closed 2025-04-27 by mr-manuel