mr-manuel/venus-os_dbus-serialbattery #242: No Floating only bulk and INFO:SerialBattery:cell_voltages_good: False

Jautze82, 2025-04-04:
### Describe the problem

Hi,

I have 2 or three questions/problems.

I have a Battery with 16cells with a daly BMS and a standalone Daly Balancer. For me it looks like the Balancer did not work. 
Yesterday I added these lines to see the parameters:
GUI_PARAMETERS_SHOW_ADDITIONAL_INFO = True
CVCM_ENABLE = True

From there I saw that maybe the problems is/was that my cells have a big diff and I only loaded with bulk and not with float.
So I added this lines:
CELL_VOLTAGE_DIFF_TO_RESET_VOLTAGE_LIMIT        = 0.010
CELL_VOLTAGE_DIFF_KEEP_MAX_VOLTAGE_UNTIL        = 0.285
CELL_VOLTAGE_DIFF_KEEP_MAX_VOLTAGE_TIME_RESTART = 0.290

[image]
Here I see now to go to float it needs max_Battery_Voltage 55.2 <=53.71

but how could that be when my MAX_CELL_VOLTAGE is set to 3.45V?

And another thing:
When I had a look in the logs I found this:
2025-04-04 09:10:44.889225500 INFO:SerialBattery:cell_voltages_good: False - min: 3.351 > 3.25 - max: 3.371 < 3.35
where comes the 3.25 from and where the 3.35?

Thanks and best regards
Marcel

### Installation method

Install script from the documentation

### Driver version of the currently installed driver

v1.6.20250131

### Driver version of the last known working driver

v1.6.20250131

### Venus OS device type

Raspberry Pi 3

### Venus OS version

v3.54

### BMS type

Daly BMS

### Cell count

16

### Battery count

1

### Connection type

Serial USB adapter to RS485

### USB hub

Yes, no external power supply

### Config file

```ini
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):
BMS_TYPE = Daly
BATTERY_CAPACITY = 280
INVERT_CURRENT_MEASUREMENT = -1
MAX_BATTERY_CHARGE_CURRENT = 90.0
MAX_BATTERY_DISCHARGE_CURRENT = 110.0
MAX_CELL_VOLTAGE=3.45
;VOLTAGE_DROP = 0.24
TIME_TO_SOC_POINTS = 100, 50, 25, 0
SOC_LOW_WARNING = 5
SOC_LOW_ALARM   = 2
;CELL_VOLTAGES_WHILE_CHARGING      =  3.65, 3.60, 3.50, 3.45, 3.40, 3.30
;MAX_CHARGE_CURRENT_CV_FRACTION    =     0, 0.01, 0.05,  0.2,  0.5,    1
CELL_VOLTAGES_WHILE_CHARGING      = 3.45, 3.42, 3.39, 3.30
MAX_CHARGE_CURRENT_CV_FRACTION    =    0, 0.05,  0.5,    1
;CELL_VOLTAGES_WHILE_CHARGING      = 3.55, 3.50, 3.45, 3.30
;MAX_CHARGE_CURRENT_CV_FRACTION    =    0, 0.05,  0.5,    1

;von Michael:
;MAX_VOLTAGE_TIME_SEC = 900
;FLOAT_CELL_VOLTAGE = 3.35
SOC_LEVEL_TO_RESET_VOLTAGE_LIMIT = 90

;Test:
CELL_VOLTAGE_DIFF_TO_RESET_VOLTAGE_LIMIT        = 0.010
SOC_RESET_VOLTAGE = 3.650
CELL_VOLTAGE_DIFF_KEEP_MAX_VOLTAGE_UNTIL        = 0.285
CELL_VOLTAGE_DIFF_KEEP_MAX_VOLTAGE_TIME_RESTART = 0.290
GUI_PARAMETERS_SHOW_ADDITIONAL_INFO = True
CVCM_ENABLE = True
;LINEAR_LIMITATION_ENABLE = False
```

### Relevant log output

```shell
2025-04-04 09:08:59.451501500 INFO:SerialBattery:cell_voltages_good: False - min: 3.351 > 3.25 - max: 3.361 < 3.35
2025-04-04 09:09:01.433625500 INFO:SerialBattery:refresh_data: result: False. If you don't see this warning very often, you can ignore it.
2025-04-04 09:09:01.434306500 INFO:SerialBattery:cell_voltages_good: False - min: 3.346 > 3.25 - max: 3.367 < 3.35
2025-04-04 09:09:04.436290500 INFO:SerialBattery:refresh_data: result: False. If you don't see this warning very often, you can ignore it.
2025-04-04 09:09:04.436991500 INFO:SerialBattery:cell_voltages_good: False - min: 3.35 > 3.25 - max: 3.366 < 3.35
2025-04-04 09:09:06.442948500 INFO:SerialBattery:refresh_data: result: False. If you don't see this warning very often, you can ignore it.
2025-04-04 09:09:06.443660500 INFO:SerialBattery:cell_voltages_good: False - min: 3.346 > 3.25 - max: 3.37 < 3.35
2025-04-04 09:09:21.396734500 INFO:SerialBattery:refresh_data: result: False. If you don't see this warning very often, you can ignore it.
2025-04-04 09:09:21.397458500 INFO:SerialBattery:cell_voltages_good: False - min: 3.347 > 3.25 - max: 3.369 < 3.35
2025-04-04 09:09:24.458929500 INFO:SerialBattery:refresh_data: result: False. If you don't see this warning very often, you can ignore it.
2025-04-04 09:09:24.459665500 INFO:SerialBattery:cell_voltages_good: False - min: 3.346 > 3.25 - max: 3.376 < 3.35
2025-04-04 09:09:25.442808500 INFO:SerialBattery:refresh_data: result: False. If you don't see this warning very often, you can ignore it.
2025-04-04 09:09:33.446808500 INFO:SerialBattery:refresh_data: result: False. If you don't see this warning very often, you can ignore it.
2025-04-04 09:09:33.447507500 INFO:SerialBattery:cell_voltages_good: False - min: 3.353 > 3.25 - max: 3.363 < 3.35
2025-04-04 09:10:37.016579500 INFO:SerialBattery:refresh_data: result: False. If you don't see this warning very often, you can ignore it.
2025-04-04 09:10:37.017248500 INFO:SerialBattery:cell_voltages_good: False - min: 3.352 > 3.25 - max: 3.365 < 3.35
2025-04-04 09:10:44.888572500 INFO:SerialBattery:refresh_data: result: False. If you don't see this warning very often, you can ignore it.
2025-04-04 09:10:44.889225500 INFO:SerialBattery:cell_voltages_good: False - min: 3.351 > 3.25 - max: 3.371 < 3.35
```

### Any other information that may be helpful

I haven't tried the nightly build, because this is more a generell question I think.

### Please confirm that you have completed the following steps before submitting your support request

- [x] I have carefully read all the instructions above including the descriptions.
- [x] I have tried the `nightly` build first, if I was not already using it. This means the logs I provide are from the `nightly` build.
- [x] I have rebooted at least once after I saw the issue.
- [x] I have read the complete [changelog](https://github.com/mr-manuel/dbus-serialbattery/blob/master/CHANGELOG.md).
- [x] I have provided the logfiles in the correct format with a HUMAN READABLE timestamp by piping it through `tai64nlocal`.
- [ ] I have clicked all checkboxes because I did not read the instructions.

onemorename, 2025-04-04:
Same here, driver doesn't switch to float on a new setup i built. Consists of a MPPT which charges the cells up to 3,52V and should switch to a 3,45V float after 30 minutes. It stays however at the target voltage and Idle's then with 100%. I have to enable debug logging there to get some further insights

mr-manuel, 2025-04-05:
> Here I see now to go to float it needs max_Battery_Voltage 55.2 <=53.71
>
> but how could that be when my MAX_CELL_VOLTAGE is set to 3.45V?

That will now work, as you saw. If you accept higher imbalance you also have to accept higher cell voltages. Your settings

> ```
> CELL_VOLTAGES_WHILE_CHARGING      = 3.45, 3.42, 3.39, 3.30
> MAX_CHARGE_CURRENT_CV_FRACTION    =    0, 0.05,  0.5,    1
> ```

force the charging to stop, if the cell voltage of 3.45 V of one cell is reached and that prevents the cell from having a higher voltage. Due to your high cell imbalance you will never reach 3.45 * 16 = 55.2 V. If you accept the high cell imbalance you have to accept also a higher max cell charging voltage. Depending on how many cells reach the 3.45 V you have to increase that value to 3.50 V or even higher.

Worst case ( 1 cell reaching `MAX_CELL_VOLTAGE `, 15 cells are on `MAX_CELL_VOLTAGE ` - `CELL_VOLTAGE_DIFF_KEEP_MAX_VOLTAGE_UNTIL`): 3.45 + (3.45 - 0.285) * 15 = 50.925 V, which is far away from 55.2 V
Even with a `MAX_CELL_VOLTAGE` you would not reach 55.2 V in that case.

Optimal solution would be to top balance the cells again. If your BMS is too weak to do that add a external balancer like the NEEY. If for whatever reason you still like to accept the high cell imbalance you have to set the `MAX_CELL_VOLTAGE` to a voltage where the cells are not drifting that much apart. Maybe 3.40 V or even lower.

> 2025-04-04 09:10:44.889225500 INFO:SerialBattery:cell_voltages_good: False - min: 3.351 > 3.25 - max: 3.371 < 3.35
where comes the 3.25 from and where the 3.35?

This message indicates, that your BMS connection is lost. The driver then checks if the cell voltages are in a safe range between the values `BLOCK_ON_DISCONNECT_VOLTAGE_MIN` and `BLOCK_ON_DISCONNECT_VOLTAGE_MAX`. If yes it's safe to proceed for some time, even if the BMS connection is lost. If not, then it's not safe and the driver exits.

Therefore this is no bug, but a misconfiguration.

@onemorename is you do not have exactly the same issue, open a new support issue for your case.

closed 2025-04-27 by mr-manuel