mr-manuel/venus-os_dbus-serialbattery #247: Seplos V3 Parallel Mode

mycroft2k, 2025-04-12:
### Is your feature request related to a problem? Please describe.

Unfortunately, it is currently only possible to read out one Seplos V3 BMS module.
At the moment, I have two units connected via RS485, but the RS485 interface only outputs a shared broadcast – from all connected battery packs combined.

The messages look as follows:

```
2025.04.06 00:17:27:008:Total Len:266
02 04 10 00 00 12 74 F4 02 04 24 14 75 FD 9F 33 DF 4C 2C 04 0E 02 A9 03 E1 00 44 0C C8 0B 57 0C CE 0C C6 0B 5C 0B 54 00 00 00 82 00 82 03 E8 70 A5 02 04 11 00 00 1A 74 CE 02 04 34 0C C8 0C C9 0C C9 0C C7 0C C7 0C CA 0C C8 0C C9 0C C6 0C CA 0C C8 0C CB 0C C7 0C C9 0C CB 0C CE 0B 5C 0B 54 0B 56 0B 56 0A AB 0A AB 0A AB 0A AB 0B 51 0B 53 17 36 02 01 12 00 00 90 39 2D 02 01 12 00 00 00 00 00 00 00 00 01 00 00 00 00 00 00 03 00 00 19 96 01 04 24 14 73 FD E5 31 AA 4C 2C 02 34 02 8C 03 E4 00 25 0C C7 0B 5C 0C CC 0C C5 0B 64 0B 56 00 00 00 82 00 82 03 E8 E5 0E 01 04 34 0C C9 0C CA 0C C9 0C C8 0C C5 0C C7 0C C7 0C C6 0C C7 0C C7 0C C7 0C C7 0C C7 0C C7 0C C9 0C CC 0B 64 0B 59 0B 56 0B 5E 0A AB 0A AB 0A AB 0A AB 0B 5B 0B 54 12 35 01 01 12 00 00 00 00 00 00 00 00 01 00 00 00 00 00 00 03 00 00 2A A5 

2025.04.06 00:17:27:218:Total Len:266
02 04 10 00 00 12 74 F4 02 04 24 14 75 FD 99 33 DF 4C 2C 04 0E 02 A9 03 E1 00 44 0C C9 0B 57 0C CF 0C C6 0B 5C 0B 54 00 00 00 82 00 82 03 E8 E4 58 02 04 11 00 00 1A 74 CE 02 04 34 0C C8 0C C9 0C C8 0C C7 0C C7 0C CB 0C C8 0C C9 0C C6 0C CA 0C C8 0C CB 0C C7 0C C9 0C CA 0C CF 0B 5C 0B 54 0B 56 0B 56 0A AB 0A AB 0A AB 0A AB 0B 51 0B 53 A0 6A 02 01 12 00 00 90 39 2D 02 01 12 00 00 00 00 00 00 00 00 01 00 00 00 00 00 00 03 00 00 19 96 01 04 24 14 73 FD E4 31 AA 4C 2C 02 34 02 8C 03 E4 00 25 0C C7 0B 5C 0C CC 0C C4 0B 64 0B 56 00 00 00 82 00 82 03 E8 8A 4F 01 04 34 0C C9 0C CA 0C C9 0C C8 0C C4 0C C7 0C C7 0C C6 0C C7 0C C7 0C C7 0C C7 0C C7 0C C7 0C C9 0C CC 0B 64 0B 59 0B 56 0B 5E 0A AB 0A AB 0A AB 0A AB 0B 5B 0B 54 D0 74 01 01 12 00 00 00 00 00 00 00 00 01 00 00 00 00 00 00 03 00 00 2A A5 

2025.04.06 00:17:27:409:Total Len:266
02 04 10 00 00 12 74 F4 02 04 24 14 75 FD 96 33 DF 4C 2C 04 0E 02 A9 03 E1 00 44 0C C8 0B 57 0C CF 0C C6 0B 5C 0B 54 00 00 00 82 00 82 03 E8 BD A6 02 04 11 00 00 1A 74 CE 02 04 34 0C C7 0C C9 0C C8 0C C7 0C C7 0C CB 0C C8 0C CA 0C C6 0C CA 0C C8 0C CB 0C C7 0C CA 0C CA 0C CF 0B 5C 0B 54 0B 56 0B 56 0A AB 0A AB 0A AB 0A AB 0B 51 0B 53 86 9E 02 01 12 00 00 90 39 2D 02 01 12 00 00 00 00 00 00 00 00 01 00 00 00 00 00 00 03 00 00 19 96 01 04 24 14 73 FD E6 31 AA 4C 2C 02 34 02 8C 03 E4 00 25 0C C7 0B 5C 0C CC 0C C4 0B 64 0B 56 00 00 00 82 00 82 03 E8 53 8F 01 04 34 0C C9 0C CA 0C C9 0C C8 0C C4 0C C7 0C C7 0C C6 0C C7 0C C7 0C C7 0C C7 0C C7 0C C7 0C C9 0C CC 0B 64 0B 59 0B 56 0B 5E 0A AB 0A AB 0A AB 0A AB 0B 5B 0B 54 D0 74 01 01 12 00 00 00 00 00 00 00 00 01 00 00 00 00 00 00 03 00 00 2A A5 

2025.04.06 00:17:27:611:Total Len:266
02 04 10 00 00 12 74 F4 02 04 24 14 75 FD 95 33 DF 4C 2C 04 0E 02 A9 03 E1 00 44 0C C9 0B 57 0C CF 0C C6 0B 5C 0B 54 00 00 00 82 00 82 03 E8 34 5A 02 04 11 00 00 1A 74 CE 02 04 34 0C C7 0C C9 0C C8 0C C7 0C C7 0C CB 0C C8 0C CA 0C C6 0C CA 0C C8 0C CB 0C C7 0C CA 0C CA 0C CF 0B 5C 0B 54 0B 56 0B 56 0A AB 0A AB 0A AB 0A AB 0B 51 0B 53 86 9E 02 01 12 00 00 90 39 2D 02 01 12 00 00 00 00 00 00 00 00 01 00 00 00 00 00 00 03 00 00 19 96 01 04 24 14 73 FD E3 31 AA 4C 2C 02 34 02 8C 03 E4 00 25 0C C7 0B 5C 0C CC 0C C4 0B 64 0B 56 00 00 00 82 00 82 03 E8 8F 8E 01 04 34 0C C9 0C CA 0C C9 0C C8 0C C4 0C C7 0C C7 0C C6 0C C7 0C C7 0C C7 0C C7 0C C7 0C C7 0C C9 0C CC 0B 64 0B 59 0B 56 0B 5E 0A AB 0A AB 0A AB 0A AB 0B 5B 0B 54 D0 74 01 01 12 00 00 00 00 00 00 00 00 01 00 00 00 00 00 00 03 00 00 2A A5 

2025.04.06 00:17:27:814:Total Len:266
02 04 10 00 00 12 74 F4 02 04 24 14 75 FD 92 33 DF 4C 2C 04 0E 02 A9 03 E1 00 44 0C C9 0B 57 0C CF 0C C6 0B 5C 0B 54 00 00 00 82 00 82 03 E8 31 9B 02 04 11 00 00 1A 74 CE 02 04 34 0C C7 0C C9 0C C8 0C C7 0C C7 0C CB 0C C8 0C CA 0C C6 0C CA 0C C8 0C CB 0C C7 0C CA 0C CA 0C CF 0B 5C 0B 54 0B 56 0B 56 0A AB 0A AB 0A AB 0A AB 0B 51 0B 53 86 9E 02 01 12 00 00 90 39 2D 02 01 12 00 00 00 00 00 00 00 00 01 00 00 00 00 00 00 03 00 00 19 96 01 04 24 14 73 FD E0 31 AA 4C 2C 02 34 02 8C 03 E4 00 25 0C C7 0B 5C 0C CC 0C C4 0B 64 0B 56 00 00 00 82 00 82 03 E8 3B 8E 01 04 34 0C C9 0C CA 0C C9 0C C8 0C C4 0C C7 0C C7 0C C6 0C C7 0C C7 0C C7 0C C7 0C C7 0C C7 0C C9 0C CC 0B 64 0B 59 0B 56 0B 5E 0A AB 0A AB 0A AB 0A AB 0B 5B 0B 54 D0 74 01 01 12 00 00 00 00 00 00 00 00 01 00 00 00 00 00 00 03 00 00 2A A5 

2025.04.06 00:17:28:015:Total Len:266
02 04 10 00 00 12 74 F4 02 04 24 14 75 FD 8F 33 DF 4C 2C 04 0E 02 A9 03 E1 00 44 0C C9 0B 57 0C CF 0C C6 0B 5C 0B 54 00 00 00 82 00 82 03 E8 4D 9F 02 04 11 00 00 1A 74 CE 02 04 34 0C C7 0C C9 0C C8 0C C7 0C C7 0C CB 0C C8 0C CA 0C C6 0C CA 0C C8 0C CB 0C C7 0C C9 0C CA 0C CF 0B 5C 0B 54 0B 56 0B 56 0A AB 0A AB 0A AB 0A AB 0B 51 0B 53 79 D5 02 01 12 00 00 90 39 2D 02 01 12 00 00 00 00 00 00 00 00 01 00 00 00 00 00 00 03 00 00 19 96 01 04 24 14 73 FD DF 31 AA 4C 2C 02 34 02 8C 03 E4 00 25 0C C7 0B 5C 0C CC 0C C5 0B 64 0B 56 00 00 00 82 00 82 03 E8 1D 06 01 04 34 0C C9 0C CA 0C C9 0C C8 0C C5 0C C7 0C C7 0C C6 0C C7 0C C6 0C C7 0C C7 0C C7 0C C7 0C C9 0C CC 0B 64 0B 59 0B 56 0B 5E 0A AB 0A AB 0A AB 0A AB 0B 5B 0B 54 7F F5 01 01 12 00 00 00 00 00 00 00 00 01 00 00 00 00 00 00 03 00 00 2A A5

```

[image]

[image]

Is it possible to integrate the parallel mode?

mr-manuel, 2025-04-13:
@marcelrv could you take a look?

marcelrv, 2025-04-13:
Yes, let me do so.. some patience is required as I'm traveling the next 3 weeks, so won't be a quicky.

I remember I looked at this once before, but recall the used modbus library doesn't support receiving broadcast messages.
Not sure, so would need to test this again.

marcelrv, 2025-04-13:
@mycroft2k tried to decode the modbus message you shared, but it seems not valid.
Do you know if the program you used to extract this takes out some of the data ( e.g. header and/or CRC)

https://npulse.net/en/online-modbus
https://rapidscada.net/modbus/

marcelrv, 2025-04-13:
@mycroft2k 
Maybe you can try what happens if you comment out these lines which disables the broadcast mode
https://github.com/mr-manuel/venus-os_dbus-serialbattery/blob/master/dbus-serialbattery/bms/seplosv3.py#L48-L51

```
    def get_modbus(self, slaveaddress=0) -> minimalmodbus.Instrument:
        # hack to allow communication to the Seplos BMS using minimodbus which uses slaveaddress 0 as broadcast
        # Make sure we re-set these values whenever we want to access the modbus. Just in case of a
        # multi-device setup with different addresses and subsequent tries on a different address modified it.
 #       if slaveaddress == 0:
 #           minimalmodbus._SLAVEADDRESS_BROADCAST = 0xF0
  #      else:
#            minimalmodbus._SLAVEADDRESS_BROADCAST = 0
```

mycroft2k, 2025-04-13:
> [@mycroft2k](https://github.com/mycroft2k) tried to decode the modbus message you shared, but it seems not valid. Do you know if the program you used to extract this takes out some of the data ( e.g. header and/or CRC)
> 
> https://npulse.net/en/online-modbus https://rapidscada.net/modbus/

the data is from seplos BMS-Studio Message
is there a better way to record the data?

[image]

mycroft2k, 2025-04-13:
> [@mycroft2k](https://github.com/mycroft2k) Maybe you can try what happens if you comment out these lines which disables the broadcast mode [...]

The changes made had no impact on the Venus operating system, serialbattery is not displayed

mycroft2k, 2025-04-14:
I found a connector for ioBroker that works ( https://github.com/DpunktS/ioBroker.seplos-v3-sniffer ) – is it possible to integrate it into serialbattery?

marcelrv, 2025-04-14:
@mycroft2k As I can see it is different structure than the existing seplos v3 format.

Some elements are the same e.g. https://github.com/DpunktS/ioBroker.seplos-v3-sniffer/blob/main/main.js#L358 can most likely be processed with https://github.com/mr-manuel/venus-os_dbus-serialbattery/blob/master/dbus-serialbattery/bms/seplosv3.py#L189
But the other blocks I don't directly see the resemblence

Hence it will require a new BMS driver .py to process this format, unlikely that it can be processed with minor changes with the existing seplosv3.py driver.