mr-manuel/venus-os_dbus-serialbattery #248: gui v1 not working with venus.os 3.55

gerd-git, 2025-04-18:
### Describe the bug

After venus.os upgrade to v3.55 white screen with GUI v1, both on the web and on the connected display. GUI v2 is working. No problem with venus.os v3.54. 
nightly build "v2.0.20250324dev"

### How to reproduce

upgrade to v3.55 + watch GUI v1

### Expected behavior

GUI v1 working

### Installation method

Install script from the documentation

### Driver version of the currently installed driver

v2.0.20250324dev

### Venus OS device type

Raspberry Pi 4

### Venus OS version

v3.55

### BMS type

Pace BMS, JKBMS Inverter (JKBMS PB Model)

### Cell count

16

### Battery count

2 (JK via CAN, Pace via RS232)

### Connection type

CAN

### USB hub

No, the Serial USB adapter is connected directly to the Venus OS device

### Config file

```ini
root@raspberrypi2:~# cat /data/apps/dbus-serialbattery/config.ini
[DEFAULT]

; --------- Set logging level ---------
; ERROR: Only errors are logged
; WARNING: Errors and warnings are logged
; INFO: Errors, warnings, and info messages are logged
; DEBUG: Errors, warnings, info, and debug messages are logged
LOGGING = DEBUG

; --------- Battery Current Limits ---------
; +++ Limits apply to each individual battery/BMS. +++
; +++ If you have multiple batteries, you need a battery aggregator. +++
MAX_BATTERY_CHARGE_CURRENT    = 100.0
MAX_BATTERY_DISCHARGE_CURRENT = 100.0

; --------- Cell Voltages ---------
; Description:
[... 545 lines cut ...]
@40000000680278031ddc860c INFO: Create daemontools service dbus-serialbattery.ttyUSB0
@400000006802780522a0e804 INFO: Create daemontools service vedirect-interface.ttyUSB1
@400000006802780713fed3ec INFO: Create daemontools service mk2-dbus.ttyUSB2
@40000000680278092189f5dc INFO: Create daemontools service vedirect-interface.ttyUSB3
@400000006802780a02ae3204 INFO: Start service dbus-serialbattery.ttyUSB0 once
@400000006802780b0dc4d184 INFO: Create daemontools service vedirect-interface.ttyUSB4
@400000006802780b34829aa4 INFO: Start service vedirect-interface.ttyUSB1
@400000006802780d23e24d5c INFO: Start service mk2-dbus.ttyUSB2
@400000006802780f344635b4 INFO: Start service vedirect-interface.ttyUSB3 once
@400000006802781115c1cd24 INFO: Start service vedirect-interface.ttyUSB4
@400000006802784310270c44 INFO: Create daemontools service gps-dbus.ttyUSB0
@40000000680278492de922bc INFO: Start service gps-dbus.ttyUSB0 once
@40000000680278532c1dbd44 INFO: Create daemontools service vedirect-interface.ttyUSB0
@40000000680278592fcfd094 INFO: Start service vedirect-interface.ttyUSB0 once
@400000006802785f0d4b9a44 INFO: Start service dbus-serialbattery.ttyUSB0 once
```

### Please confirm that you have completed the following steps before submitting your support request

- [x] I have carefully read all the instructions above including the descriptions.
- [x] I have tried the `nightly` build first, if I was not already using it. This means the logs I provide are from the `nightly` build.
- [x] I have rebooted at least once after I saw the issue.
- [x] I have read the complete [changelog](https://github.com/mr-manuel/dbus-serialbattery/blob/master/CHANGELOG.md).
- [x] I have provided the logfiles in the correct format with a HUMAN READABLE timestamp by piping it through `tai64nlocal`.
- [ ] I have clicked all checkboxes because I did not read the instructions.

mr-manuel, 2025-04-18:
Check https://github.com/mr-manuel/venus-os_dbus-serialbattery/issues/197

gerd-git, 2025-04-18:
Thanks Manuel, I have read #197. It is not exactly my problem. Because for me the problem disappears when I go back to version 3.54.

Do I understand you correctly [here](https://github.com/mr-manuel/venus-os_dbus-serialbattery/issues/197#issuecomment-2688561475), that the problem is known and you solve it with v3.60?

mr-manuel, 2025-04-18:
As you can see in the log you updated it's a GuiMods related problem.

closed 2025-04-18 by mr-manuel