mr-manuel/venus-os_dbus-serialbattery #249: Daly ActiveBalance BMS running but  Internal calculation error

Freensman, 2025-04-19:
### Describe the problem

New blue Daly ActiveBalance BMS system running (with help of transistorgit & mr-manuel) thanks a lot. 
Two Daly BMS dasy chained over CAN. 

All working fine and running but still get two Errors: Internal calculation error on every BMS.
Just visible on VRM not on Cerbo:

[image]

I tried to get the log files but under option "/data/log/dbus-serialbattery.ttyUSB*/current"  no folder "dbus-serialbattery". 

Second option "tail -F -n 100 /data/log/serial-starter/current | grep dbus-serialbattery | tai64nlocal" 
no output, nothing happens. 
If I go to have a look in the "current" file over SFTP I get this see below:

I run the actual nightly build "v2.0.20250324dev"

Any Idea why the log files are not there? And what could cause those errors?

Thanks for help.  Freensman

### Installation method

Install script from the documentation

### Driver version of the currently installed driver

v2.0.20250324dev

### Driver version of the last known working driver

v2.0.20250324dev

### Venus OS device type

Cerbo GX

### Venus OS version

v2.42 / 3.55

### BMS type

Daly BMS

### Cell count

8

### Battery count

2

### Connection type

CAN

### USB hub

No, the Serial USB adapter is connected directly to the Venus OS device

### Config file

```ini
[DEFAULT]

; --------- Set logging level ---------
; ERROR: Only errors are logged
; WARNING: Errors and warnings are logged
[... 480 lines cut ...]
03/10/146138514338 00:42:01 INFO: loading config file /etc/venus/serial-starter.conf
03/10/146138514338 00:42:02 INFO: loading config file /data/conf/serial-starter.d/dbus-serialbattery.conf
03/10/146138514338 00:42:03 INFO: Create daemontools service mk2-dbus.ttyS4
03/10/146138514338 00:42:05 INFO: Create daemontools service vedirect-interface.ttyS5
03/10/146138514338 00:42:07 INFO: Create daemontools service vedirect-interface.ttyS6
03/10/146138514338 00:42:08 INFO: Create daemontools service vedirect-interface.ttyS7
03/10/146138514338 00:42:10 INFO: Start service mk2-dbus.ttyS4
03/10/146138514338 00:42:11 INFO: Start service vedirect-interface.ttyS5
03/10/146138514338 00:42:13 INFO: Start service vedirect-interface.ttyS6
03/10/146138514338 00:42:14 INFO: Start service vedirect-interface.ttyS7
```

### Any other information that may be helpful

Issue was there wit Venus OS 3.42 - I updated with "white screen issues" on remote console to 3.55.
Now running 3.55

### Please confirm that you have completed the following steps before submitting your support request

- [x] I have carefully read all the instructions above including the descriptions.
- [x] I have tried the `nightly` build first, if I was not already using it. This means the logs I provide are from the `nightly` build.
- [x] I have rebooted at least once after I saw the issue.
- [x] I have read the complete [changelog](https://github.com/mr-manuel/dbus-serialbattery/blob/master/CHANGELOG.md).
- [x] I have provided the logfiles in the correct format with a HUMAN READABLE timestamp by piping it through `tai64nlocal`.
- [ ] I have clicked all checkboxes because I did not read the instructions.

mr-manuel, 2025-04-19:
Just read the documentation: https://mr-manuel.github.io/venus-os_dbus-serialbattery_docs/troubleshoot/#driver-log-files

Freensman, 2025-04-19:
root@einstein:~# tail -F -n 100 /data/log/dbus-canbattery.can0/current | tai64nlocal

2025-04-19 13:14:29.554744500 DEBUG:SerialBattery:[can0] Received: ID=0x18984001, Daten=bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00')
2025-04-19 13:14:29.574261500 DEBUG:SerialBattery:Data check: 2
2025-04-19 13:14:29.624979500 DEBUG:SerialBattery:logged to dbus [75.8]
2025-04-19 13:14:29.628487500 DEBUG:SerialBattery:Cells:[1]3.366V [2]3.365V [3]3.364V [4]3.365V [5]3.365V [6]3.372V [7]3.365V [8]3.365V 
2025-04-19 13:14:29.686882500 DEBUG:SerialBattery:[can0] Received: ID=0x18904002, Daten=bytearray(b'\x01\r\x00\x00uf\x03+')
2025-04-19 13:14:29.686891500 DEBUG:SerialBattery:[can0] Received: ID=0x18914002, Daten=bytearray(b'\r%\x03\r$\x01\xff\xff')
2025-04-19 13:14:29.686895500 DEBUG:SerialBattery:[can0] Received: ID=0x18924002, Daten=bytearray(b'6\x025\x01\xff\xff\xff\xff')
2025-04-19 13:14:29.687899500 DEBUG:SerialBattery:[can0] Received: ID=0x18934002, Daten=bytearray(b'\x01\x01\x01&\x00\x02z<')
2025-04-19 13:14:29.692663500 DEBUG:SerialBattery:[can0] Received: ID=0x18944002, Daten=bytearray(b'\x08\x02\x00\x00\x00\x00\x016')
2025-04-19 13:14:29.694348500 DEBUG:SerialBattery:[can0] Received: ID=0x18a64002, Daten=bytearray(b'\x01\r$\r$\r%\x00')
2025-04-19 13:14:29.695963500 DEBUG:SerialBattery:[can0] Received: ID=0x18a74002, Daten=bytearray(b'\x02\r%\r%\r%\x00')
2025-04-19 13:14:29.697591500 DEBUG:SerialBattery:[can0] Received: ID=0x18a84002, Daten=bytearray(b'\x03\r$\r%\r%\x00')
2025-04-19 13:14:29.699500500 DEBUG:SerialBattery:[can0] Received: ID=0x18974002, Daten=bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00')
2025-04-19 13:14:29.701150500 DEBUG:SerialBattery:[can0] Received: ID=0x18984002, Daten=bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00')
2025-04-19 13:14:29.727355500 DEBUG:SerialBattery:[can0] Received: ID=0x305, Daten=bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00')
2025-04-19 13:14:29.729156500 DEBUG:SerialBattery:[can0] Received: ID=0x307, Daten=bytearray(b'\x124VxVIC\x00')
2025-04-19 13:14:29.756873500 DEBUG:SerialBattery:Data check: 2
2025-04-19 13:14:29.774647500 DEBUG:SerialBattery:[can0] Received: ID=0x351, Daten=bytearray(b'\x1e\x01\xc4\t\xc4\t\xd4\x00')
2025-04-19 13:14:29.775732500 DEBUG:SerialBattery:[can0] Received: ID=0x355, Daten=bytearray(b'K\x00d\x00\x9c\x1d\x00\x00')
2025-04-19 13:14:29.786684500 DEBUG:SerialBattery:[can0] Received: ID=0x356, Daten=bytearray(b'\x82\n+\x00\x82\x00\x00\x00')
2025-04-19 13:14:29.803256500 DEBUG:SerialBattery:[can0] Received: ID=0x35a, Daten=bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00')
2025-04-19 13:14:29.816509500 DEBUG:SerialBattery:[can0] Received: ID=0x35e, Daten=bytearray(b'LVaiiey\x00')
2025-04-19 13:14:29.832908500 DEBUG:SerialBattery:logged to dbus [81.1]
2025-04-19 13:14:29.834136500 DEBUG:SerialBattery:Cells:[1]3.364V [2]3.364V [3]3.365V [4]3.365V [5]3.365V [6]3.365V [7]3.364V [8]3.365V 
2025-04-19 13:14:29.835227500 DEBUG:SerialBattery:Polling data took 0.368 seconds
2025-04-19 13:14:29.851456500 DEBUG:SerialBattery:[can0] Received: ID=0x35f, Daten=bytearray(b'\x00\x00\x01\x01\x97\x00\x00\x00')
2025-04-19 13:14:29.853523500 DEBUG:SerialBattery:[can0] Received: ID=0x370, Daten=bytearray(b'Lithium\x00')
2025-04-19 13:14:29.856571500 DEBUG:SerialBattery:[can0] Received: ID=0x371, Daten=bytearray(b'Valley\x00\x00')
2025-04-19 13:14:29.858756500 DEBUG:SerialBattery:[can0] Received: ID=0x373, Daten=bytearray(b'%\r,\r\x1e\x01\x1f\x01')
2025-04-19 13:14:29.864547500 DEBUG:SerialBattery:[can0] Received: ID=0x374, Daten=bytearray(b'\x02\x00\x06\x00\x02\x00\x01\x00')
2025-04-19 13:14:29.870850500 DEBUG:SerialBattery:[can0] Received: ID=0x375, Daten=bytearray(b'\x06\x00\x00\x00\x00\x00\x00\x00')
2025-04-19 13:14:29.891626500 DEBUG:SerialBattery:[can0] Received: ID=0x376, Daten=bytearray(b'\x02\x00\x00\x00\x00\x00\x00\x00')
2025-04-19 13:14:29.893858500 DEBUG:SerialBattery:[can0] Received: ID=0x377, Daten=bytearray(b'\x01\x00\x00\x00\x00\x00\x00\x00')
2025-04-19 13:14:29.895905500 DEBUG:SerialBattery:[can0] Received: ID=0x379, Daten=bytearray(b'\x97\x00\x00\x00\x00\x00\x00\x00')
2025-04-19 13:14:30.207352500 DEBUG:SerialBattery:[can0] Received: ID=0x351, Daten=bytearray(b' \x01\xc4\t\xc4\t\xe4\x00')
2025-04-19 13:14:30.239116500 DEBUG:SerialBattery:[can0] Received: ID=0x355, Daten=bytearray(b'Q\x00d\x00\xae\x1f\x00\x00')
2025-04-19 13:14:30.240650500 DEBUG:SerialBattery:[can0] Received: ID=0x356, Daten=bytearray(b'\x82\n6\x00\x82\x00\x00\x00')
2025-04-19 13:14:30.242018500 DEBUG:SerialBattery:[can0] Received: ID=0x35a, Daten=bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00')
2025-04-19 13:14:30.243348500 DEBUG:SerialBattery:[can0] Received: ID=0x35e, Daten=bytearray(b'LVaiiey\x00')
2025-04-19 13:14:30.244632500 DEBUG:SerialBattery:[can0] Received: ID=0x35f, Daten=bytearray(b'\x00\x00\x01\x01\xa2\x00\x00\x00')
2025-04-19 13:14:30.245925500 DEBUG:SerialBattery:[can0] Received: ID=0x370, Daten=bytearray(b'Lithium\x00')
2025-04-19 13:14:30.247217500 DEBUG:SerialBattery:[can0] Received: ID=0x371, Daten=bytearray(b'Valley\x00\x00')
2025-04-19 13:14:30.248665500 DEBUG:SerialBattery:[can0] Received: ID=0x373, Daten=bytearray(b'$\r%\r\x1e\x01\x1f\x01')
2025-04-19 13:14:30.250041500 DEBUG:SerialBattery:[can0] Received: ID=0x374, Daten=bytearray(b'\x01\x00\x03\x00\x01\x00\x02\x00')
2025-04-19 13:14:30.251435500 DEBUG:SerialBattery:[can0] Received: ID=0x375, Daten=bytearray(b'\x03\x00\x00\x00\x00\x00\x00\x00')
[... cut ...]
2025-04-19 13:14:51.095649500 INFO:SerialBattery:> CCCM CV:    True  | DCCM CV:       True
2025-04-19 13:14:51.096801500 INFO:SerialBattery:> CCCM T:     True  | DCCM T:        True
2025-04-19 13:14:51.097823500 INFO:SerialBattery:> CCCM T MOS: True  | DCCM T MOS:    True
2025-04-19 13:14:51.098987500 INFO:SerialBattery:> CCCM SOC:   False | DCCM SOC:      False
2025-04-19 13:14:51.100517500 INFO:SerialBattery:> CHARGE FET: True  | DISCHARGE FET: True | BALANCE FET: None
2025-04-19 13:14:51.102087500 INFO:SerialBattery:Serial Number/Unique Identifier: can0__0x02

mr-manuel, 2025-04-21:
The log is not reflecting the error you get. You have to wait until you get the error and then check and post the logs.

Freensman, 2025-04-23:
Thanks for analyzing. The Error shows up right from the beginning when I restarted the driver. 

I restart again and had a look in the log, nothing special. Than I checked the Alarm in VRM, it was dated way back. 
So it looks the Alarm was triggered but never "untriggered" so as soon as I deleted the Alarm in VRM its gone.
Its running since a couple of hours now.

Thanks, I'm now getting more and more into the "drivers" control options. 

Will send you some coffee dineros.

Cheers Silvio

closed 2025-04-23 by Freensman

mr-manuel, 2025-04-23:
Thank you very much!