mr-manuel/venus-os_dbus-serialbattery #251: Switch to Float problem

pkeberly, 2025-04-21:
### Describe the problem

I seem to have an issue getting my battery to switch to float.  It gets right to the point where it should start timing for the switch, but it never does switch.    

Also I see a lot of long polling data, is that BMS related or the Cerbo running slow?

### Installation method

Install script from the documentation

### Driver version of the currently installed driver

v2.0.20250324dev

### Venus OS device type

Cerbo GX

### Venus OS version

v3.60~60

### BMS type

Jiabaida JDB BMS (LLT Power, Overkill Solar, Smart BMS)

### Cell count

8

### Battery count

1

### Connection type

Serial USB adapter to TTL

### USB hub

Yes, no external power supply

### Config file

```ini
[DEFAULT]

;------------ Setup for Luna's 8S energy battery ---------
;  12/16/2024 PWK  first connections, 300aH test
;  12/17/2024 PWK  driver update, bump top voltage to 3.52, tune charge ramp to get better balancing.
;  12/22/2024 PWK  lowered CCL/DCL Recovery Threshold to 0.8% (1.2A)
;  01/09/2025 PWK  Updated driver, changed config.ini to match
;  03/26/2025 PWK  Swapped to Cerbo-S after USB failure. changed config.ini to match latest drvr.
;  04/01/2025 PWK  Solar test, Alternator test, generator test.  Bump charge current 120A->150A
;  04/13/2025 PWK  Lowered cell voltages.  retesting switch to float
;  04/19/2025 PWK  Changed current recovery threshold from 1.5% to 0.5% (2.25A to 0.75A) set balance turn-on to 3.35V

LOGGING = INFO

; --------- Battery Current Limits ---------
; +++ Limits apply to each individual battery/BMS. +++
; +++ If you have multiple batteries, you need a battery aggregator. +++
MAX_BATTERY_CHARGE_CURRENT    = 150.0
MAX_BATTERY_DISCHARGE_CURRENT = 200.0

; --------- Cell Voltages ---------
; Description:
;     Calculates the minimum and maximum battery voltage based on the number of cells and their voltage.
;     ATTENTION: This does not prevent overcharging or overdischarging of individual cells.
;     To prevent overcharging and overdischarging, use these features (enabled by default):
[... 394 lines cut ...]
2025-04-21 14:17:08.580060500 INFO:SerialBattery:> CHARGE FET: True  | DISCHARGE FET: True | BALANCE FET: True
2025-04-21 14:17:08.582963500 INFO:SerialBattery:Serial Number/Unique Identifier: 39_304.0Ah
2025-04-21 14:19:03.270004500 WARNING:SerialBattery:Polling data took 1.095 seconds. Automatically increase interval in 3 cycles.
2025-04-21 14:19:07.703315500 WARNING:SerialBattery:Polling data took 1.219 seconds. Automatically increase interval in 3 cycles.
2025-04-21 14:19:43.946214500 WARNING:SerialBattery:Polling data took 1.164 seconds. Automatically increase interval in 3 cycles.
2025-04-21 14:20:57.443110500 WARNING:SerialBattery:Polling data took 1.011 seconds. Automatically increase interval in 3 cycles.
2025-04-21 14:22:22.696018500 WARNING:SerialBattery:Polling data took 1.026 seconds. Automatically increase interval in 3 cycles.
2025-04-21 14:22:33.425490500 WARNING:SerialBattery:Polling data took 1.236 seconds. Automatically increase interval in 3 cycles.
2025-04-21 14:22:59.845505500 WARNING:SerialBattery:Polling data took 1.057 seconds. Automatically increase interval in 3 cycles.
2025-04-21 14:23:03.434672500 WARNING:SerialBattery:Polling data took 1.225 seconds. Automatically increase interval in 3 cycles.
2025-04-21 14:23:37.498871500 WARNING:SerialBattery:Polling data took 1.061 seconds. Automatically increase interval in 3 cycles.
2025-04-21 14:23:43.029448500 WARNING:SerialBattery:Polling data took 1.218 seconds. Automatically increase interval in 3 cycles.
2025-04-21 14:23:52.989070500 WARNING:SerialBattery:Polling data took 1.024 seconds. Automatically increase interval in 3 cycles.
2025-04-21 14:24:22.921636500 WARNING:SerialBattery:Polling data took 1.069 seconds. Automatically increase interval in 3 cycles.
2025-04-21 14:24:24.217962500 WARNING:SerialBattery:Polling data took 1.072 seconds. Automatically increase interval in 2 cycles.
```

### Please confirm that you have completed the following steps before submitting your support request

- [x] I have carefully read all the instructions above including the descriptions.
- [x] I have tried the `nightly` build first, if I was not already using it. This means the logs I provide are from the `nightly` build.
- [x] I have rebooted at least once after I saw the issue.
- [x] I have read the complete [changelog](https://github.com/mr-manuel/dbus-serialbattery/blob/master/CHANGELOG.md).
- [x] I have provided the logfiles in the correct format with a HUMAN READABLE timestamp by piping it through `tai64nlocal`.
- [ ] I have clicked all checkboxes because I did not read the instructions.

mr-manuel, 2025-04-21:
Just scroll down on the parameters page, you were already there...

pkeberly, 2025-04-21:
Yes I know the values that the driver requires to switch, but it never gets to that point.  

The pic I added shows a CCL of 1.2A yet the driver is calculating 0A.  I've adjusted the charger taper with CV fraction, and reduced the recovery threshold, and widened the cell voltage differential.  I'm uncertain what to change next to get the last bit of charge required to switch to float.  What am I doing wrong?

mr-manuel, 2025-04-21:
If you know the values, then please post them.

pkeberly, 2025-04-21:
[image]

mr-manuel, 2025-04-21:
As you see, your battery is not reaching `max_battery_voltage`. Try to enable `SVS - Shared voltage sense` or set a voltage drop value in the config of the driver.

pkeberly, 2025-04-21:
I feel the cabling isn't an issue, that is very robust.  And this problem is always at the end of charge when the currents are so low that effects from voltage drop should be minimal.

The problem as I see it, as the pack approaches max_battery_voltage, the charge current tapers off to zero and never returns to push the voltage >= max_battery_voltage.    It charges perfectly until that last little bit.

In the first picture you can see the CCL is 1.2A, yet current and current_calc are both 0.0.  This is the state that my charge fails to continue on.   I did watch the charger current vs. battery current, they track nicely

[image]

[image]

mr-manuel, 2025-04-21:
It does not have to be the cable. The main reason is because the BMS is measuring one voltage and the charger is measuring one voltage. The votlages are off by 0.04 V, therefore it will never change to float. This is normal, since the measurements from the different devices are not that accurate.

I updated also the docs: https://mr-manuel.github.io/venus-os_dbus-serialbattery_docs/faq/#why-is-my-battery-not-switching-to-floatbulk

Just follow them and your problem will be solved.

closed 2025-04-21 by mr-manuel

pkeberly, 2025-04-21:
Thank you, now I know here to look.

Data download shows at 0A the battery reads 30mV lower than the charger and at 48A the battery is ~170mV lower, 150mV due to wire resistance, 30mV due to measurement error. 

If the inverter reads higher than the BMS, and the BMS is the lead in requesting voltage, wouldn't that increase the current at lower voltages? 

Is voltage drop just to compensate for voltage reading mismatch at low current and I need to tune out just the 30mV diff?