mr-manuel/venus-os_dbus-serialbattery #256: Cerbo display mode issue

pkeberly, 2025-05-09:
### Describe the bug

The latest Venus update has triggered an odd issue.  The display mode of both of my Cerbo GX devices have changed from Dark display mode to Light display mode, ignoring the display mode selection.  

Switching firmware does not fix the issue.  I did a factory reset of the Cerbo which fixed the issue, only to return after a reboot which also coincided with the serial driver install. 

Executing: bash custom-gui-uninstall.sh, reboot, display mode functionality returns.
Executing: bash custom-gui-install.sh, reboot, display mode switches to Light display mode again.

[image]

### How to reproduce

set display mode to dark.

**When serial driver was already installed:**
update firmware to 3.60~71
reboot or downgrade firmware
reload 3.60~71. 

**After factory reset**
reboot
install serial driver
reboot

During startup, the splash screen & GUI will switch light mode, and ignore the mode selection.

### Expected behavior

Display Mode should follow the selection option.

### Installation method

Install script from the documentation

### Driver version of the currently installed driver

2.0.20250324rc

### Venus OS device type

Cerbo GX

### Venus OS version

v3.60~71

### BMS type

Jiabaida JDB BMS (LLT Power, Overkill Solar, Smart BMS)

### Cell count

8

### Battery count

1

### Connection type

Serial USB adapter to TTL

### USB hub

Yes, no external power supply

### Config file

```ini
[... 433 lines cut ...]
2025-05-09 11:57:32.656741500 INFO:SerialBattery:Battery LLT/JBD connected to dbus from /dev/ttyUSB2
2025-05-09 11:57:32.657855500 INFO:SerialBattery:========== Settings ==========
2025-05-09 11:57:32.658942500 INFO:SerialBattery:> Connection voltage: 25.96 V | Current: 6.18 A | SoC: 41%
2025-05-09 11:57:32.659791500 INFO:SerialBattery:> Cell count: 8 | Cells populated: 8
2025-05-09 11:57:32.660484500 INFO:SerialBattery:> CHARGE MODE: Linear
2025-05-09 11:57:32.661204500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.900 V | MAX CELL VOLTAGE: 3.520 V| FLOAT CELL VOLTAGE: 3.375 V
2025-05-09 11:57:32.661880500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 120.0 A | MAX BATTERY DISCHARGE CURRENT: 200.0 A
2025-05-09 11:57:32.662817500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 200.0 A | MAX BATTERY DISCHARGE CURRENT: 300.0 A (read from BMS)
2025-05-09 11:57:32.663719500 INFO:SerialBattery:> CVCM:       True
2025-05-09 11:57:32.666009500 INFO:SerialBattery:> CCCM CV:    True  | DCCM CV:       True
2025-05-09 11:57:32.673578500 INFO:SerialBattery:> CCCM T:     True  | DCCM T:        True
2025-05-09 11:57:32.681924500 INFO:SerialBattery:> CCCM T MOS: True  | DCCM T MOS:    True
2025-05-09 11:57:32.681933500 INFO:SerialBattery:> CCCM SOC:   False | DCCM SOC:      False
2025-05-09 11:57:32.681936500 INFO:SerialBattery:> CHARGE FET: True  | DISCHARGE FET: True | BALANCE FET: True
2025-05-09 11:57:32.681940500 INFO:SerialBattery:Serial Number/Unique Identifier: 39_304.0Ah
```

### Please confirm that you have completed the following steps before submitting your support request

- [x] I have carefully read all the instructions above including the descriptions.
- [x] I have tried the `nightly` build first, if I was not already using it. This means the logs I provide are from the `nightly` build.
- [x] I have rebooted at least once after I saw the issue.
- [x] I have read the complete [changelog](https://github.com/mr-manuel/dbus-serialbattery/blob/master/CHANGELOG.md).
- [x] I have provided the logfiles in the correct format with a HUMAN READABLE timestamp by piping it through `tai64nlocal`.
- [ ] I have clicked all checkboxes because I did not read the instructions.

mr-manuel, 2025-05-09:
This is not a bug, but a feature. If you open the GUIv2 in the browser if follows the system color scheme. Since the needed path to select this behavior is missing, the default applies which is following the client device system theme. Therefore the entry `Remote console color mode` below `Display mode` is `Unkown`.

If you want to change that, execute this command once and reboot your device after:

```bash
dbus -y com.victronenergy.settings /Settings AddSetting Gui RemoteConsoleColorMode 1 i 0 1
```

This will automatically work with the todays nightly version of the driver.

pkeberly, 2025-05-09:
Well that was easy enough, thank you.   
I didn't see that feature in the change log?

mr-manuel, 2025-05-09:
In which changelog? It's not officially merged, I only merged it right away to have it, since it's very useful.

pkeberly, 2025-05-09:
The Remote console color mode selection, I don't recall seeing that option.  
Is that a Venus feature or serial driver feature?

mr-manuel, 2025-05-09:
It's a Venus OS GUIv2 feature I developed. It will be merged in a few weeks.

mr-manuel, 2025-05-11:
I reverted the setting for now, since it seems too confusing for most of the people.

closed 2025-05-11 by mr-manuel