mr-manuel/venus-os_dbus-serialbattery #257: CerboGX Bootloop

Jacke87, 2025-05-09:
### Describe the problem

Hello, my CerboGX keeps restarting. I use dbusserial Battery and dbusmqqtPV as well as the Kostal Piko plugin. I first tried to start without USB devices, unfortunately without success. Then I disconnected the Kostal and the Growatt inverter from the LAN, also without success. I then removed the LAN connection from the CerboGX and it booted. Then I did not start the Kostal software anymore. I removed the files from rc.local and emulated the Kostal via TCP mode using Node Red as EM24. Now serialBattery and mqttPV are still running. But the Cerbo was in a boot loop again, after some time it starts but then goes off in between. I executed this command ‘cat /var/log/start-gui/* | tai64nlocal’ and got the following result.

Hallo mein CerboGX startet immer wieder neu. Ich benutze dbusserial Battery und dbusmqqtPV sowie das Kostal Piko Plugin. Ich habe erstmal versucht ohne USB Geräte zu starten leider ohne Erfolg. Dann Habe ich den Kostal und den Growatt Wechselrichter vom Lan geklemmt, ebenfalls ohne Erfolg. Daraufhin habe ich die LAN Verbindung des CerboGX entfernt und er bootet. Dann habe ich die Kostal Software nicht mehr gestartet. habe die Dateien aus der rc.local entfernt und den Kostal über Modus TCP per Node Red als EM24 emuliert. Nun laufen noch serialBattery und mqttPV. Der Cerbo befand sich aber wieder in einem BootLoop nach einiger Zeit startet er dann geht aber zwischendurch aus. Ich habe diesen Befehl ausgeführt "cat /var/log/start-gui/* | tai64nlocal" daraufhin bekam ich.

I hope I can get some advice here

### Installation method

Install script from the documentation

### Driver version of the currently installed driver

v1.6.20250131

### Driver version of the last known working driver

v1.6.20250131

### Venus OS device type

Cerbo GX

### Venus OS version

3.55

### BMS type

JKBMS (Heltec BMS)

### Cell count

16

### Battery count

1

### Connection type

Serial USB adapter to RS485

### USB hub

No, the Serial USB adapter is connected directly to the Venus OS device

### Config file

```ini
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):
 MAX_BATTERY_CHARGE_CURRENT = 70.0
 MAX_BATTERY_DISCHARGE_CURRENT = 60.0
 ;MAX_CELL_VOLTAGE   = 3.450
 ;FLOAT_CELL_VOLTAGE = 3.375

; -- CVL reset based on cell voltage diff (linear mode)
; Specify cell voltage diff where CVL limit is kept until diff is equal or lower
CELL_VOLTAGE_DIFF_KEEP_MAX_VOLTAGE_UNTIL = 0.015
; Specify cell voltage diff where MAX_VOLTAGE_TIME_SEC restarts if diff is bigger
CELL_VOLTAGE_DIFF_KEEP_MAX_VOLTAGE_TIME_RESTART = 0.020

; Specify cell voltage diff where CVL limit is reset to max voltage, if value get above
; the cells are considered as imbalanced, if the cell diff exceeds 5% of the nominal cell voltage
[... 268 lines cut ...]
@40000000681e040c27ed7d2c INFO:SerialBattery:Polling interval: 1.000 s
@40000000681e040c28141fcc INFO:SerialBattery:Battery JKBMS connected to dbus from /dev/ttyUSB0
@40000000681e040c2823332c INFO:SerialBattery:========== Settings ==========
@40000000681e040c2835e06c INFO:SerialBattery:> Connection voltage: 54.42 V | Current: -0.0 A | SoC: 98% | SoC calc: 100%
@40000000681e040c284b9704 INFO:SerialBattery:> Cell count: 16 | Cells populated: 16
@40000000681e040c285ad55c INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
@40000000681e040c286b5404 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.900 V | MAX CELL VOLTAGE: 3.450 V| FLOAT CELL VOLTAGE: 3.375 V
@40000000681e040c287959dc INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 70.0 A | MAX BATTERY DISCHARGE CURRENT: 60.0 A
@40000000681e040c28870dac INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 100.0 A | MAX BATTERY DISCHARGE CURRENT: 120.0 A (read from BMS)
@40000000681e040c2896a9c4 INFO:SerialBattery:> CVCM:     True
@40000000681e040c28bf156c INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
@40000000681e040c28d145ac INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
@40000000681e040c28e2f8ec INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
@40000000681e040c29016214 INFO:SerialBattery:> CHARGE FET: True | DISCHARGE FET: True | BALANCE FET: True
@40000000681e040c29110214 INFO:SerialBattery:Serial Number/Unique Identifier: EVE_lf280k_#
```

### Please confirm that you have completed the following steps before submitting your support request

- [x] I have carefully read all the instructions above including the descriptions.
- [x] I have tried the `nightly` build first, if I was not already using it. This means the logs I provide are from the `nightly` build.
- [x] I have rebooted at least once after I saw the issue.
- [x] I have read the complete [changelog](https://github.com/mr-manuel/dbus-serialbattery/blob/master/CHANGELOG.md).
- [x] I have provided the logfiles in the correct format with a HUMAN READABLE timestamp by piping it through `tai64nlocal`.
- [ ] I have clicked all checkboxes because I did not read the instructions.

Jacke87, 2025-05-09:
`root@einstein:~# cat /var/log/start-gui/* | tai64nlocal
2025-05-09 12:57:28.058061500 *** starting start-gui ***
2025-05-09 12:57:28.302063500 Error org.freedesktop.DBus.Error.ServiceUnknown: The name com.victronenergy.settings was not provided by any .service files
2025-05-09 12:57:29.461204500 Error org.freedesktop.DBus.Error.ServiceUnknown: The name com.victronenergy.settings was not provided by any .service files
2025-05-09 12:57:30.533090500 Error org.freedesktop.DBus.Error.ServiceUnknown: The name com.victronenergy.settings was not provided by any .service files
2025-05-09 12:57:31.691703500 Error org.freedesktop.DBus.Error.ServiceUnknown: The name com.victronenergy.settings was not provided by any .service files
2025-05-09 12:57:32.819012500 Error org.freedesktop.DBus.Error.ServiceUnknown: The name com.victronenergy.settings was not provided by any .service files
2025-05-09 12:57:33.924823500 Error org.freedesktop.DBus.Error.ServiceUnknown: The name com.victronenergy.settings was not provided by any .service files
2025-05-09 12:57:35.004831500 Error org.freedesktop.DBus.Error.ServiceUnknown: The name com.victronenergy.settings was not provided by any .service files
2025-05-09 12:57:36.127062500 Error org.freedesktop.DBus.Error.ServiceUnknown: The name com.victronenergy.settings was not provided by any .service files
2025-05-09 12:57:37.193054500 Error org.freedesktop.DBus.Error.ServiceUnknown: The name com.victronenergy.settings was not provided by any .service files
@40000000681dfbe22498cc1c *** CCGX booted (0) ***
2025-05-09 12:59:05.614107500 *** CCGX booted (0) ***
2025-05-09 12:59:06.115307500 *** starting start-gui ***
2025-05-09 12:59:06.344576500 Error org.freedesktop.DBus.Error.ServiceUnknown: The name com.victronenergy.settings was not provided by any .service files
2025-05-09 12:59:07.491681500 Error org.freedesktop.DBus.Error.ServiceUnknown: The name com.victronenergy.settings was not provided by any .service files
2025-05-09 12:59:08.642986500 Error org.freedesktop.DBus.Error.ServiceUnknown: The name com.victronenergy.settings was not provided by any .service files
2025-05-09 12:59:09.779520500 Error org.freedesktop.DBus.Error.ServiceUnknown: The name com.victronenergy.settings was not provided by any .service files
2025-05-09 12:59:10.871556500 Error org.freedesktop.DBus.Error.ServiceUnknown: The name com.victronenergy.settings was not provided by any .service files
2025-05-09 12:59:11.998781500 Error org.freedesktop.DBus.Error.ServiceUnknown: The name com.victronenergy.settings was not provided by any .service files
2025-05-09 12:59:13.092017500 Error org.freedesktop.DBus.Error.ServiceUnknown: The name com.victronenergy.settings was not provided by any .service files
2025-05-09 12:59:14.228321500 Error org.freedesktop.DBus.Error.ServiceUnknown: The name com.victronenergy.settings was not provided by any .service files
2025-05-09 12:59:15.320488500 Error org.freedesktop.DBus.Error.ServiceUnknown: The name com.victronenergy.settings was not provided by any .service files
2025-05-09 12:59:16.408864500 Error org.freedesktop.DBus.Error.ServiceUnknown: The name com.victronenergy.settings was not provided by any .service files
2025-05-09 12:59:17.495142500 Error org.freedesktop.DBus.Error.ServiceUnknown: The name com.victronenergy.settings was not provided by any .service files
2025-05-09 12:59:18.575140500 Error org.freedesktop.DBus.Error.ServiceUnknown: The name com.victronenergy.settings was not provided by any .service files
2025-05-09 12:59:20.533405500 method return time=1746795560.533192 sender=:1.57 -> destination=:1.65 serial=15 reply_serial=2
2025-05-09 12:59:20.533415500    int32 0
2025-05-09 12:59:20.877589500 *** headless device=1
2025-05-09 12:59:20.939618500 *** Waiting for localsettings...
2025-05-09 12:59:21.300537500 *** Localsettings is up, continuing...
2025-05-09 12:59:21.596050500 method return time=1746795561.595144 sender=:1.57 -> destination=:1.70 serial=154 reply_serial=2
2025-05-09 12:59:21.596060500    int32 0
2025-05-09 12:59:21.670409500 *** Starting gui, with VNC enabled)
2025-05-09 12:59:22.359286500 QVNCServer created on port 5900
2025-05-09 12:59:22.492062500 Reloading input devices: "LinuxInput:/dev/input/event0 LinuxInput:/dev/input/event1" 
2025-05-09 12:59:22.507048500 "using /etc/venus for runtime features" 
2025-05-09 12:59:22.527118500 "running on einstein" 
2025-05-09 12:59:22.616013500 Connecting to deprecated signal QDBusConnectionInterface::serviceOwnerChanged(QString,QString,QString)
2025-05-09 12:59:22.826742500 Creating settings 
2025-05-09 12:59:23.663458500 [VeQItemExportedDbusServices] Registered service "debug.victronenergy.gui" 
2025-05-09 12:59:24.180594500 language is "de" 
2025-05-09 12:59:24.236599500 accessLevelChanged:  2 
2025-05-09 12:59:24.247579500 setAccessLevel:  2 
2025-05-09 12:59:24.430404500 LocalThemeDaemonClient: Looking for assets in "/opt/victronenergy/themes/ccgx" 
2025-05-09 12:59:24.431376500 Screen: 480x272 169x96mm 72x72dpi
2025-05-09 12:59:25.050349500 TimeZone::valueChanged:  "Europe/Berlin" 
2025-05-09 12:59:25.069620500 ScreenSaver::dbusDisplayOffTimeChanged:  600 
2025-05-09 12:59:25.075995500 ScreenSaver::enable 
2025-05-09 12:59:25.077595500 ScreenSaver::screenOn 
2025-05-09 12:59:47.195299500 ScreenSaver::disable 
2025-05-09 12:59:47.195686500 ScreenSaver::screenOn 
2025-05-09 12:59:53.495429500 *** starting start-gui ***
[... cut ...]
2025-05-09 13:30:01.918039500 QDBusConnection: name 'com.victronenergy.system' had owner '' but we thought it was ':1.42'
2025-05-09 13:30:14.715218500 file:///opt/victronenergy/gui/qml/OverviewTiles.qml:158: Unable to assign [undefined] to QString text
2025-05-09 13:32:28.056863500 QVNCServer new incoming connection from "127.0.0.1" 
2025-05-09 13:32:28.096503500 QVNCServer Client protocol version RFB 003.003
2025-05-09 13:32:28.096512500 
2025-05-09 13:32:28.096728500 QVNCServer No password configured, accepting connection
root@einstein:~# `

mr-manuel, 2025-05-09:
Check your CPU load: https://mr-manuel.github.io/venus-os_dbus-serialbattery_docs/faq/#how-to-troubleshoot-high-cpu-load

That's the only reason I have in mind why the system should restart. There is a watchdog which triggers the reboot, but at the moment I can't find the correct log for it.

closed 2025-06-08 by mr-manuel