mr-manuel/venus-os_dbus-serialbattery #260: Regular System Reboots

Albarge, 2025-05-12:
### Describe the problem

Ever since around Venus v3.5X / Gui v2 I've had an unstable Cerbo and now Ekrano. It keeps rebooting every few hours. I have tried v3.55 and the last few beta's to 3.60~72 with v1.5, v1.6.20250131, v2.0rc and v2.20250511dev.

I have just done a Victron factory reset, although I'm not sure if it's normal for that to go into a boot loop until pulling the power and the SD card with venus image out, then it boots with a fresh OS.

I have only installed  SB, and SetupHelper to get BatteryAggregator, no other Mods, on a fresh v3.55 with the SB rc, which was unstable so I tried the dev, which was the same so I tried the Venus beta, and still getting random reboots, large and normal both the same.

Often after a reboot I get a temporary Error #67 BMS Connection Lost on an MPPT before they are found again, this happens multiple times during the day, but the Ekrano will also randomly reboot at night.

I don't know if this is a sbattery issue or BatteryAggregator or Venus, but it's been ongoing for a while now and as I have 3 batteries I need both mods, It would be really nice to have a stable system to then try and test NodeRed for hot water diversion control, but I'm leaving that until it's stable.

### Installation method

Install script from the documentation

### Driver version of the currently installed driver

v2.20250511dev

### Venus OS device type

Ekrano GX

### Venus OS version

v3.60~72

### Venus OS image type

Large

### BMS type

JKBMS (Heltec BMS)

### Cell count

8

### Battery count

3

### Connection type

Serial USB adapter to RS485

### USB hub

Yes, with external power supply

### Config file

```ini
MAX_BATTERY_CHARGE_CURRENT = 170.0
MAX_BATTERY_DISCHARGE_CURRENT = 200.0

LOGGING = DEBUG

MIN_CELL_VOLTAGE   = 2.900

MAX_CELL_VOLTAGE   = 3.500

FLOAT_CELL_VOLTAGE = 3.375

SOC_RESET_CELL_VOLTAGE = 3.501

SWITCH_TO_FLOAT_WAIT_FOR_SEC = 600

[... 155 lines cut ...]

==> /data/var/lib/serial-starter/ttyUSB2 <==
mkx

==> /data/var/lib/serial-starter/ttyUSB3 <==
sbattery

==> /data/var/lib/serial-starter/ttyUSB4 <==
sbattery
```

### Any other information that may be helpful

The root password thing was strange as I used a number but it looks like it registered an apostrophe.

Sbattery is starting fine on my USB1, USB3 and USB4

### Please confirm that you have completed the following steps before submitting your support request

- [x] I have carefully read all the instructions above including the descriptions.
- [x] I have tried the `nightly` build first, if I was not already using it. This means the logs I provide are from the `nightly` build.
- [x] I have rebooted at least once after I saw the issue.
- [x] I have read the complete [changelog](https://github.com/mr-manuel/dbus-serialbattery/blob/master/CHANGELOG.md).
- [x] I have provided the logfiles in the correct format with a HUMAN READABLE timestamp by piping it through `tai64nlocal`.
- [ ] I have clicked all checkboxes because I did not read the instructions.

mr-manuel, 2025-05-12:
Check your system load, the logs under `/var/log/messages*` and the dbus roundtrip time graphs on VRM. For the last you need to go to VRM portal -> Your installation -> Advanced -> Widgets (icon on the top right) -> Custom Widget -> Create custom widget and then: 

[image]

Albarge, 2025-05-12:
```
==> /var/log/messages.5 <==
May 11 23:36:37 ekrano kern.info kernel: [    0.000000] OF: reserved mem: Reserved memory: No reserved-memory node in the DT
May 11 23:36:37 ekrano kern.info kernel: [    0.000000] psci: probing for conduit method from DT.
May 11 23:36:37 ekrano kern.info kernel: [    0.000000] psci: Using PSCI v0.1 Function IDs from DT
May 11 23:36:37 ekrano kern.info kernel: [    0.000000] percpu: Embedded 14 pages/cpu s25996 r8192 d23156 u57344
May 11 23:36:37 ekrano kern.debug kernel: [    0.000000] pcpu-alloc: s25996 r8192 d23156 u57344 alloc=14*4096
May 11 23:36:37 ekrano kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
May 11 23:36:37 ekrano kern.notice kernel: [    0.000000] Kernel command line: console=ttyS0,115200 root=/dev/mmcblk1p2 rootwait ro rootfstype=ext4 5 vt.global_cursor_default=0
May 11 23:36:37 ekrano kern.notice kernel: [    0.000000] Unknown kernel command line parameters "5", will be passed to user space.
May 11 23:36:37 ekrano kern.info kernel: [    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
May 11 23:36:37 ekrano kern.info kernel: [    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
May 11 23:36:37 ekrano kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 262144
May 11 23:36:37 ekrano kern.info kernel: [    0.000000] mem auto-init: stack:all(zero), heap alloc:off, heap free:off
May 11 23:36:37 ekrano kern.info kernel: [    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
May 11 23:36:37 ekrano kern.info kernel: [    0.000000] rcu: Hierarchical RCU implementation.
May 11 23:36:37 ekrano kern.info kernel: [    0.000000] 	Tracing variant of Tasks RCU enabled.
May 11 23:36:37 ekrano kern.info kernel: [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
May 11 23:36:37 ekrano kern.info kernel: [    0.000000] RCU Tasks Trace: Setting shift to 2 and lim to 1 rcu_task_cb_adjust=1 rcu_task_cpu_ids=4.
May 11 23:36:37 ekrano kern.info kernel: [    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
May 11 23:36:37 ekrano kern.info kernel: [    0.000000] GIC: Using split EOI/Deactivate mode
May 11 23:36:37 ekrano kern.info kernel: [    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
May 11 23:36:37 ekrano kern.info kernel: [    0.000000] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 79635851949 ns
May 11 23:36:37 ekrano kern.info kernel: [    0.000000] arch_timer: cp15 timer(s) running at 24.00MHz (phys).
May 11 23:36:37 ekrano kern.info kernel: [    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
May 11 23:36:37 ekrano kern.info kernel: [    0.000002] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
May 11 23:36:37 ekrano kern.info kernel: [    0.000017] Switching to timer-based delay loop, resolution 41ns
May 11 23:36:37 ekrano kern.info kernel: [    0.000298] Console: colour dummy device 80x30
May 11 23:36:37 ekrano kern.info kernel: [    0.000386] Calibrating delay loop (skipped), value calculated using timer frequency.. 48.00 BogoMIPS (lpj=240000)
May 11 23:36:37 ekrano kern.info kernel: [    0.000406] CPU: Testing write buffer coherency: ok
May 11 23:36:37 ekrano kern.info kernel: [    0.000470] pid_max: default: 32768 minimum: 301
May 11 23:36:37 ekrano kern.info kernel: [    0.000634] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
May 11 23:36:37 ekrano kern.info kernel: [    0.000652] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
May 11 23:36:37 ekrano kern.err kernel: [    0.001745] /cpus/cpu@0 missing clock-frequency property
May 11 23:36:37 ekrano kern.err kernel: [    0.001784] /cpus/cpu@1 missing clock-frequency property
May 11 23:36:37 ekrano kern.err kernel: [    0.001802] /cpus/cpu@2 missing clock-frequency property
May 11 23:36:37 ekrano kern.err kernel: [    0.001820] /cpus/cpu@3 missing clock-frequency property
May 11 23:36:37 ekrano kern.info kernel: [    0.001832] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
May 11 23:36:37 ekrano kern.info kernel: [    0.040206] Setting up static identity map for 0x40100000 - 0x40100060
May 11 23:36:37 ekrano kern.info kernel: [    0.050096] rcu: Hierarchical SRCU implementation.
May 11 23:36:37 ekrano kern.info kernel: [    0.050104] rcu: 	Max phase no-delay instances is 1000.
May 11 23:36:37 ekrano kern.info kernel: [    0.050483] Timer migration: 1 hierarchy levels; 8 children per group; 1 crossnode level
May 11 23:36:37 ekrano kern.info kernel: [    0.060737] smp: Bringing up secondary CPUs ...
May 11 23:36:37 ekrano kern.info kernel: [    0.100493] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
May 11 23:36:37 ekrano kern.info kernel: [    0.130705] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
May 11 23:36:37 ekrano kern.info kernel: [    0.160927] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
May 11 23:36:37 ekrano kern.info kernel: [    0.161042] smp: Brought up 1 node, 4 CPUs
May 11 23:36:37 ekrano kern.info kernel: [    0.161055] SMP: Total of 4 processors activated (192.00 BogoMIPS).
May 11 23:36:37 ekrano kern.info kernel: [    0.161065] CPU: All CPU(s) started in HYP mode.
May 11 23:36:37 ekrano kern.info kernel: [    0.161069] CPU: Virtualization extensions available.
[... cut ...]
May 11 23:36:37 ekrano kern.info kernel: [    0.216115] io scheduler bfq registered
May 11 23:36:37 ekrano kern.info kernel: [    0.226382] Serial: 8250/16550 driver, 8 ports, IRQ sharing disabled
May 11 23:36:37 ekrano kern.info kernel: [    0.230494] sun6i-rtc 1f00000.rtc: registered as rtc1
May 11 23:36:37 ekrano kern.info kernel: [    0.230866] i2c_dev: i2c /dev entries driver
May 11 23:36:37 ekrano kern.info kernel: [    0.231961] sunxi-wdt 1c20ca0.watchdog: Watchdog enabled (timeout=16 sec, nowayout=1)

```

Albarge, 2025-05-12:
I had issues with the Cerbo, which was why I got an Ekrano hoping it could handle 3 batteries, the DBus Round trip is definitely much less with the Ekrano but maybe not enough!?

[image]

mr-manuel, 2025-05-12:
What does the system load say? Did you check the FAQ with high CPU usage?

Albarge, 2025-05-12:
Having said the above it's a little worse right now:

`
Mem: 544916K used, 484164K free, 17868K shrd, 11600K buff, 158588K cached
CPU:  51% usr  15% sys   0% nic  28% idle   0% io   2% irq   2% sirq
Load average: 6.12 7.75 5.45 9/358 8034
  PID  PPID USER     STAT   VSZ %VSZ %CPU COMMAND
  904   900 root     S     235m  23%  23% /opt/victronenergy/gui-v2/venus-gui-v2
  619   618 messageb S     3840   0%   6% dbus-daemon --system --nofork
  895   889 root     S    44200   4%   6% {vrmlogger.py} /usr/bin/python3 -u /opt/victronenergy/vrmlogger/vrmlogger.py
  959   956 root     R    31056   3%   4% {dbus_systemcalc} /usr/bin/python3 -u /opt/victronenergy/dbus-systemcalc-py/dbus_systemcalc.py
 1851  1845 root     S    48144   5%   3% /usr/bin/flashmq
 2739  1006 root     R    29356   3%   3% python /data/BatteryAggregator/battery_service.py
 1341   972 root     S    28392   3%   3% {dbus_generator.} /usr/bin/python3 -u /opt/victronenergy/dbus-generator/dbus_generator.py
  901   891 root     S    27916   3%   2% /opt/victronenergy/venus-platform/venus-platform
  995   979 root     S    22404   2%   2% /opt/victronenergy/dbus-fronius/dbus-fronius
  951   944 root     S    20636   2%   2% /opt/victronenergy/hub4control/hub4control
 1565   929 www-data S     8096   1%   2% nginx: worker process
 2840  2834 root     R    45876   4%   2% python /data/apps/dbus-serialbattery/dbus-serialbattery.py /dev/ttyUSB1
 1928  1921 root     S    27592   3%   2% {vesmart_server.} /usr/bin/python3 -u /opt/victronenergy/vesmart-server/vesmart_server.py -i hci0
 2371  2352 root     S    45876   4%   2% python /data/apps/dbus-serialbattery/dbus-serialbattery.py /dev/ttyUSB3
 2360  2349 root     S    45876   4%   1% python /data/apps/dbus-serialbattery/dbus-serialbattery.py /dev/ttyUSB4
 1743  1691 root     S     3256   0%   1% /opt/victronenergy/mk2-dbus/mk2-dbus --log-before 25 --log-after 25 --banner -w -s /dev/ttyS2 -i -t mk3 --settings /data/va
   63     2 root     IW       0   0%   1% [kworker/u16:5-e]
   64     2 root     IW       0   0%   1% [kworker/u16:6-d]
 1642     2 root     SW       0   0%   0% [irq/81-spi1.0]
 6507     2 root     IW       0   0%   0% [kworker/u16:0-p]
 1669  1665 root     S     8048   1%   0% /opt/victronenergy/vecan-dbus/vecan-dbus -c socketcan:vecan0 --banner --log-before 25 --log-after 25 -vv
 1633  1611 root     S     7932   1%   0% /opt/victronenergy/vedirect-interface/vedirect-dbus -v --log-before 25 --log-after 25 -t 0 --banner -s /dev/ttyS1
   43     2 root     IW       0   0%   0% [kworker/u16:1-g]
  950   935 root     S    32968   3%   0% {localsettings.p} /usr/bin/python3 -u /opt/victronenergy/localsettings/localsettings.py --path=/data/conf
 1834  1809 root     S     7932   1%   0% /opt/victronenergy/vedirect-interface/vedirect-dbus -v --log-before 25 --log-after 25 -t 0 --banner -s /dev/ttyS3
 1551  1548 root     S     7904   1%   0% /opt/victronenergy/vecan-dbus/vecan-dbus -c socketcan:vecan1 --banner --log-before 25 --log-after 25 -vv
 1010  1003 root     S    72532   7%   0% python /data/SetupHelper/PackageManager.py
 1808   986 root     S     2992   0%   0% /opt/victronenergy/dbus-ble-sensors/dbus-ble-sensors --banner
 7831  4446 root     R     2740   0%   0% top
 1994  1987 root     S     7932   1%   0% /opt/victronenergy/vedirect-interface/vedirect-dbus -v --log-before 25 --log-after 25 -t 0 --banner -s /dev/ttyUSB0
 2307  2267 root     S     3140   0%   0% /opt/victronenergy/mk2-dbus/mk2-dbus --log-before 25 --log-after 25 --banner -w -s /dev/ttyUSB2 -t mk3 --settings /data/var
 1005   991 root     S     2952   0%   0% /opt/victronenergy/dbus-adc/dbus-adc --banner
  989   970 root     S    52988   5%   0% {dbus-modbus-cli} /usr/bin/python3 -u /opt/victronenergy/dbus-modbus-client/dbus-modbus-client.py
  952   946 root     S    26032   3%   0% {dbus_vebus_to_p} /usr/bin/python3 -u /opt/victronenergy/dbus-vebus-to-pvinverter/dbus_vebus_to_pvinverter.py
  931   913 root     S     7784   1%   0% {serial-starter.} /bin/bash /opt/victronenergy/serial-starter/serial-starter.sh
  906   903 simple-u S    10040   1%   0% /bin/simple-upnpd --xml /var/run/simple-upnpd.xml -d
  918   909 root     S     7780   1%   0% {navico-announce} /bin/sh /opt/victronenergy/service-advertiser/navico-announcement.sh 80 /app
 2343  2336 root     S     1912   0%   0% multilog t s500000 n4 /var/log/dbus-serialbattery.ttyUSB3
 2345  2334 root     S     1912   0%   0% multilog t s500000 n4 /var/log/dbus-serialbattery.ttyUSB4
^C982  1668 root     S     1912   0%   0% multilog t s500000 n4 /var/log/dbus-serialbattery.ttyUSB1
`

Albarge, 2025-05-12:
I changed polling to 2 seconds and now at 3 seconds, looks a little better:

[image]

Albarge, 2025-05-12:
Just installed htop again this is with 3 second polling:

[image]

mr-manuel, 2025-05-12:
Like you see in htop somehow the GUIv2 seems to be the problem here. Check, if this is also the case without dbus-serialbattery GUI mods. Run custom-gui-uninstall.sh and reboot

Albarge, 2025-05-12:
I just selected Gui v1 on the Ekrano and have got the dreaded white screen again :( This happened a few weeks ago on an earlier v3.6x beta

Albarge, 2025-05-12:
I got out of the white screen by booting to v3.55

I uninstalled battery aggregator and package manager, then ran the setuphelper script to uninstall it.

This is what Gui v2 looks like with just sbattery running on v3.55:

[image]

Albarge, 2025-05-12:
I uninstalled the driver with: bash /data/apps/dbus-serialbattery/uninstall.sh 

Then tried: custom-gui-uninstall.sh but it said command not found I guess the bash script already did it.

I remembered to go to all four MPPT's and reset the network connection before the Quattro shuts down, but forgot to also redect and restart the VE.Bus System in the Quattro settings, so it powered off anyway - I wish on every reboot venus would automatically redetect - that's annoying to manually do it or get a power off.

Anyway, no mods and no Sbattery. Gui v2, Venus v3.55:

[image]

Still the odd spike, but overall lower load but I don't know what it should be, if that's the issue?

mr-manuel, 2025-05-12:
In that case it's better you open a topic in the Victron Community.

Albarge, 2025-05-13:
I posted on Victron Community: [https://community.victronenergy.com/t/ekrano-with-mods-rebooting-what-should-load-averages-be/32049](url)

Out of interest, how does the polling change? If it's set for 3 seconds, do all BMS's only get polled once every 3 seconds and at the same moment, or in my case with 3 BMS's is it possible each one could get polled sequentially, so all 3 are polled staggered over the course of 3 seconds?

mr-manuel, 2025-05-14:
Over 80% only for GUIv2 seems high, yes.

It depends on how you connected the batteries. In parallel they are polled probably at the same time, in serial they are polled one after another.

Albarge, 2025-05-14:
I have a feeling my issues are related to installing Sbattery or htop? - I always have to execute the htop commands twice? and how it modifies Gui v2, possibly beta v3.6X related.

All was fine (low Dbus RTT, Low load average <2 in htop) running v3.55 with only Sbattery no other mods Gui v1, so I tried Gui v2, all was initially pretty much the same for an hour or so with only one instance in htop of gui-v2 @ around 10% CPU, so last night I downloaded v3.60~72 which also initially looked good, but after 9+ hours this morning, I had a few instances of the (/opt/victronenergy/gui-v2/venus-gui-v2) thread hog @ 70% + CPU again. So something was maybe leaking in the Gui-v2?

I switched to Gui v1 and got the blank white screen again, followed the FAQ steps for white screen, had to prevent auto install and then force update. Re-installed Sbattery and Gui-v2 process was using around 30%. Just switched back to v1 and no white screen, htop looks better:

[image]

I didn't get any responses on the Victron community, AFAIU you're waiting until v3.60 is released officially for a stable Sbattery v2? So I guess it's probably best to stick with Gui v1 or v3.55 for now. 

I could test overnight v3.55 gui v2 to see if I get the same process hogs by the morning to narrow it down, unless there's some logs that might be good to check?

Any thoughts or tips appreciated.

Thanks  Al

mr-manuel, 2025-05-14:
Yes, I'm waiting for Venus OS v3.60 released for releasing dbus-serialbattery v2.

I will install Telegraf an monitor all processes on my test systems.

Albarge, 2025-05-15:
I think.. It may be resolved. After four factory resets over the last few days, I finally tried as a last resort a full Venus OS reinstall. - I wish I had tried that earlier, in hindsight I don't think there's much point doing a reset if theres some rogue modified files that wont get removed, but the full OS reinstall seemed to clear it. 

The system has been up 20 hours without reboot, I am on stock Venus v3.55 Gui-v1 with v2.0.20250324rc and BatteryAggregator v3.33. Dbus RTT seems to now be 2ms with average of 7ms spikes and some up to 17ms, way better than before. Load averages are generally below 2 often below 1 and average idle of 80%.

Ideally over the next few days I want to check Gui-v2 and the Venus beta's again, I'm just a bit wary of the amount of time it takes to sort out when the system goes wrong and I have no idea what's causing it.

[image]

[image]

closed 2025-06-08 by mr-manuel