mr-manuel/venus-os_dbus-serialbattery #261: Bulk & SoC Reset / Absorbtion Wrong Value

DIVERSDIVERS, 2025-05-15:
### Describe the bug

After Reboot wrong Valus is picked. Normal it should be 3.560 * 16 = 56.96V but it takes the value from SOC_RESET_VOLTAGE = 3.585 *16 = 57,36.

[image]

### How to reproduce

only rebootet didn´t change anything

### Expected behavior

should be 3.560 * 16 = 56.96V

### Installation method

Install script from the documentation

### Driver version of the currently installed driver

v1.6.20250131

### Driver version of the last known working driver

v1.6.20250131

### Venus OS device type

Cerbo GX

### Venus OS version

v3.55

### Venus OS image type

Normal

### BMS type

Daly BMS

### Cell count

16

### Battery count

5

### Connection type

Serial USB adapter to TTL

### USB hub

Yes, no external power supply

### Config file

```ini
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):

LOGGING = ERROR

[... 181 lines cut ...]
2025-05-03 18:19:22.314486500 *** CCGX booted (0) ***
2025-05-03 21:37:36.330370500 *** CCGX booted (0) ***
2025-05-03 21:49:49.324339500 *** CCGX booted (0) ***
2025-05-04 20:34:23.347087500 *** CCGX booted (0) ***
2025-05-04 21:43:38.302478500 *** CCGX booted (0) ***
2025-05-05 20:28:54.311649500 *** CCGX booted (0) ***
2025-05-05 21:18:23.316300500 *** CCGX booted (0) ***
2025-05-06 08:40:57.352789500 *** CCGX booted (0) ***
2025-05-06 15:18:35.312525500 *** CCGX booted (0) ***
2025-05-07 15:09:14.297882500 *** CCGX booted (0) ***
2025-05-11 01:12:29.327408500 *** CCGX booted (0) ***
2025-05-12 14:16:58.311217500 *** CCGX booted (0) ***
2025-05-15 09:25:38.342069500 *** CCGX booted (0) ***
2025-05-15 09:31:36.289613500 *** CCGX booted (0) ***
2025-05-15 09:37:48.313323500 *** CCGX booted (0) ***
```

### Please confirm that you have completed the following steps before submitting your support request

- [x] I have carefully read all the instructions above including the descriptions.
- [x] I have tried the `nightly` build first, if I was not already using it. This means the logs I provide are from the `nightly` build.
- [x] I have rebooted at least once after I saw the issue.
- [x] I have read the complete [changelog](https://github.com/mr-manuel/dbus-serialbattery/blob/master/CHANGELOG.md).
- [x] I have provided the logfiles in the correct format with a HUMAN READABLE timestamp by piping it through `tai64nlocal`.
- [ ] I have clicked all checkboxes because I did not read the instructions.

mr-manuel, 2025-05-15:
You checked, that you tryed the nightly version, but in the version you specified the stable. Did you try with the nightly or nor?

See also https://mr-manuel.github.io/venus-os_dbus-serialbattery_docs/faq/#why-is-my-battery-not-switching-to-floatbulk to enable the debugging info in the GUI. It's now in the `dbus-serialbattery - General` page.

Post a screenshot of the three debugging fields.

DIVERSDIVERS, 2025-05-15:
Hello,

Yes, you're right. I didn't do that because I have two systems, and on the other one (see pictures), it's correct. Hence my question: why one is like this and the other is like that, even though they have the same versions and the same settings.

[image]
[image]
[image]

ElCyborg, 2025-05-15:
Just ran into this myself, well the question of "why the "SOC_RESET_VOLTAGE" and not "MAX_CELL_VOLTAGE""

In my case, it had been more than SOC_RESET_AFTER_DAYS days (in truth I just enabled it, it had never been set). Therefore, it charged until SOC_RESET_VOLTAGE had been reached. After that, the state has moved to "float transition" with the correct voltage of MAX_CELL_VOLTAGE.

A screenshot of the debug fields would determine if that was the case.

DIVERSDIVERS, 2025-05-15:
Yes, that was it. I took it out and everything was fine. Thank you!