mr-manuel/venus-os_dbus-serialbattery #262: Daren485 variant: different baud rate

kopierschnitte, 2025-05-18:
### Describe the bug

As already stated by the contributor, some revisions or variants of older PaPool-branded batteries don't work with the Daren485 implementation.

I've found out that the cause might be the baud rate. My battery also refuses to connect and then I've changed the settings to '9600' instead of the default '19200' and it works like a charm.

Don't know the ideal way to deal with this for other users but it might be an idea to try the connection with both baud rates during discovery process. 

### How to reproduce

- Connect a Daren485 based BMS using an RS485->USB adaptor.
- Start Serialbattery
- Log indicated no discovered BMS with default settings

Then edited dbus-serialbattery.py:
`    {"bms": Daren485, "baud": 9600, "address": b"\x01"},`
... works!

### Expected behavior

- Serialbattery should try baud rate 19200 and 9600

### Installation method

Install script from the documentation

### Driver version of the currently installed driver

v2.0.20250513dev

### Venus OS device type

Cerbo GX

### Venus OS version

v3.55

### Venus OS image type

Large

### BMS type

Daren 485

### Cell count

16

### Battery count

1

### Connection type

Serial USB adapter to RS485

### USB hub

Yes, no external power supply

### Config file

```ini
(empty)
```

### Relevant log output

```shell
INFO:SerialBattery:Testing Daren485 at address "\x01"
ERROR:SerialBattery:>>> ERROR: No reply - returning
```

### Please confirm that you have completed the following steps before submitting your support request

- [x] I have carefully read all the instructions above including the descriptions.
- [x] I have tried the `nightly` build first, if I was not already using it. This means the logs I provide are from the `nightly` build.
- [x] I have rebooted at least once after I saw the issue.
- [x] I have read the complete [changelog](https://github.com/mr-manuel/dbus-serialbattery/blob/master/CHANGELOG.md).
- [x] I have provided the logfiles in the correct format with a HUMAN READABLE timestamp by piping it through `tai64nlocal`.
- [ ] I have clicked all checkboxes because I did not read the instructions.

mr-manuel, 2025-05-18:
Fixed with the latest nightly

closed 2025-05-18 by mr-manuel

kopierschnitte, 2025-05-18:
Works fine. Thanks!