mr-manuel/venus-os_dbus-serialbattery #267: I repeatedly get connection lost

RolandRd, 2025-06-02:
### Describe the problem

The battery driver hangs often. So I have to check and restart the system by myself.
[image]
It cannot restart the driver.
It seems that the reason for the issue is:
Error while connecting to BLE client: BleakDBusError('org.bluez.Error.Failed', 'Software caused connection abort') of type <class 'bleak.exc.BleakDBusError'> in /data/apps/dbus-serialbattery/bms/jkbms_brn.py line #457
In the older driver (v1.6.20250131 and also my original from louisvw v1.2.20240408) is was:
error while connecting to bt: BleakDBusError('org.bluez.Error.Failed', 'Software caused connection abort') of type <class 'bleak.exc.BleakDBusError'> in /opt/victronenergy/dbus-serialbattery/bms/jkbms_brn.py line #443
As I could find comes the failure from the non thread save method. But I'm not familar with the software.
Please help!
Thank you so much
Roland Roderer

### Installation method

Install script from the documentation

### Driver version of the currently installed driver

v2.0.20250518dev

### Venus OS device type

Raspberry Pi 4

### Venus OS version

v3.54

### Venus OS image type

Large

### BMS type

JKBMS (Heltec BMS)

### Cell count

16

### Battery count

1

### Connection type

Bluetooth

### USB hub

No, the Serial USB adapter is connected directly to the Venus OS device

### Config file

```ini
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):
MAX_BATTERY_CHARGE_CURRENT = 100.0
MAX_BATTERY_DISCHARGE_CURRENT = 120.0

BMS_TYPE = Jkbms

; BLUETOOTH_BMS = Jkbms_Ble C8:47:8C:00:00:00
BLUETOOTH_BMS = Jkbms_Ble C8:47:8C:E9:13:DC
[... 477 lines cut ...]
2025-06-02 03:24:53.024421500 INFO:SerialBattery:
2025-06-02 03:24:53.025724500 ERROR:SerialBattery:>>> ERROR: Battery does not respond, init/reset values <<<
2025-06-02 03:24:53.026328500 ERROR:SerialBattery:    |- Cell voltages are in a safe threshold to proceed with charging/discharging without communication to the battery.
2025-06-02 03:24:53.026755500 ERROR:SerialBattery:    |- Min cell voltage: 3.320 > Min Threshold: 3.250 --> OK
2025-06-02 03:24:53.027127500 ERROR:SerialBattery:    |- Max cell voltage: 3.323 < Max threshold: 3.350 --> OK
2025-06-02 03:24:53.027423500 ERROR:SerialBattery:    |- Trying further for 1200 s, then restart the driver
2025-06-02 03:24:53.049217500 INFO:SerialBattery:>>> Battery reconnected <<<
2025-06-02 03:24:58.055775500 INFO:SerialBattery:Jkbms_Ble: Bluetooth connection interrupted. Got no fresh data since 1215 s.
2025-06-02 03:25:23.123621500 INFO:SerialBattery:
2025-06-02 03:25:23.124865500 ERROR:SerialBattery:>>> ERROR: Battery does not respond, init/reset values <<<
2025-06-02 03:25:23.125415500 ERROR:SerialBattery:    |- Cell voltages are in a safe threshold to proceed with charging/discharging without communication to the battery.
2025-06-02 03:25:23.125852500 ERROR:SerialBattery:    |- Min cell voltage: 3.320 > Min Threshold: 3.250 --> OK
2025-06-02 03:25:23.126228500 ERROR:SerialBattery:    |- Max cell voltage: 3.323 < Max threshold: 3.350 --> OK
2025-06-02 03:25:23.126517500 ERROR:SerialBattery:    |- Trying further for 1200 s, then restart the driver
2025-06-02 03:25:23.126892500 ERROR:SerialBattery:>>> Battery did not recover in 1200 s. Restarting driver...
```

### Please confirm that you have completed the following steps before submitting your support request

- [x] I have carefully read all the instructions above including the descriptions.
- [x] I have tried the `nightly` build first, if I was not already using it. This means the logs I provide are from the `nightly` build.
- [x] I have rebooted at least once after I saw the issue.
- [x] I have read the complete [changelog](https://github.com/mr-manuel/dbus-serialbattery/blob/master/CHANGELOG.md).
- [x] I have provided the logfiles in the correct format with a HUMAN READABLE timestamp by piping it through `tai64nlocal`.
- [ ] I have clicked all checkboxes because I did not read the instructions.

RolandRd, 2025-06-15:
The driver worked well for nearly one week. But now it hung again. I tryed to restart the driver - but no success. Then I restarted the Raspi. Here is the log:
2025-06-13 10:57:02.782836500 INFO:SerialBattery:Processing frame with settings info
2025-06-13 16:23:11.644009500 INFO:SerialBattery:JK BMS SOC reset finished.
2025-06-14 02:53:26.736113500 INFO:SerialBattery:Jkbms_Ble: Bluetooth connection interrupted. Got no fresh data since 15 s.
2025-06-14 02:53:51.785312500 INFO:SerialBattery:
2025-06-14 02:53:56.796820500 INFO:SerialBattery:Jkbms_Ble: Bluetooth connection interrupted. Got no fresh data since 45 s.
2025-06-14 02:54:21.847735500 INFO:SerialBattery:
2025-06-14 02:54:21.848314500 ERROR:SerialBattery:>>> ERROR: Battery does not respond, init/reset values <<<
2025-06-14 02:54:21.848822500 ERROR:SerialBattery:    |- Cell voltages are in a safe threshold to proceed with charging/discharging without communication to the battery.
2025-06-14 02:54:21.849524500 ERROR:SerialBattery:    |- Min cell voltage: 3.324 > Min Threshold: 3.250 --> OK
2025-06-14 02:54:21.849529500 ERROR:SerialBattery:    |- Max cell voltage: 3.327 < Max threshold: 3.350 --> OK
2025-06-14 02:54:21.849963500 ERROR:SerialBattery:    |- Trying further for 1200 s, then restart the driver
2025-06-14 02:54:26.908528500 INFO:SerialBattery:Jkbms_Ble: Bluetooth connection interrupted. Got no fresh data since 75 s.
2025-06-14 02:54:51.964449500 INFO:SerialBattery:

...

2025-06-14 03:14:06.674219500 ERROR:SerialBattery:>>> ERROR: Battery does not respond, init/reset values <<<
2025-06-14 03:14:06.674585500 ERROR:SerialBattery:    |- Cell voltages are in a safe threshold to proceed with charging/discharging without communication to the battery.
2025-06-14 03:14:06.674963500 ERROR:SerialBattery:    |- Min cell voltage: 3.324 > Min Threshold: 3.250 --> OK
2025-06-14 03:14:06.675327500 ERROR:SerialBattery:    |- Max cell voltage: 3.327 < Max threshold: 3.350 --> OK
2025-06-14 03:14:06.675624500 ERROR:SerialBattery:    |- Trying further for 1200 s, then restart the driver
2025-06-14 03:14:06.675960500 ERROR:SerialBattery:>>> Battery did not recover in 1215 s. Restarting driver...
2025-06-14 19:40:41.608864500 INFO:SerialBattery:Exit signal received, exiting gracefully...
2025-06-14 19:40:41.644879500 INFO:SerialBattery:Exit signal received, exiting gracefully...
2025-06-14 19:40:51.659868500 INFO:SerialBattery:Stopped dbus-serialbattery with exit code 0
2025-06-14 19:40:51.660062500 Exception ignored in: <module 'threading' from '/usr/lib/python3.12/threading.py'>
2025-06-14 19:40:51.660066500 Traceback (most recent call last):
2025-06-14 19:40:51.660126500   File "/usr/lib/python3.12/threading.py", line 1624, in _shutdown
2025-06-14 19:40:51.663454500     lock.acquire()
2025-06-14 19:40:51.663659500   File "/data/apps/dbus-serialbattery/dbus-serialbattery.py", line 125, in exit_driver
2025-06-14 19:40:51.664294500     battery[0].disconnect()
2025-06-14 19:40:51.664494500   File "/data/apps/dbus-serialbattery/bms/jkbms_ble.py", line 284, in disconnect
2025-06-14 19:40:51.665414500     self.jk.stop_scraping()
2025-06-14 19:40:51.665595500   File "/data/apps/dbus-serialbattery/bms/jkbms_brn.py", line 547, in stop_scraping
2025-06-14 19:40:51.667002500     sleep(0.1)
2025-06-14 19:40:51.667179500   File "/data/apps/dbus-serialbattery/dbus-serialbattery.py", line 139, in exit_driver
2025-06-14 19:40:51.667778500     sys.exit(code)
2025-06-14 19:40:51.667970500 SystemExit: 0
2025-06-14 19:41:00.549249500 INFO:SerialBattery:
2025-06-14 19:41:00.549594500 INFO:SerialBattery:Starting dbus-serialbattery
2025-06-14 19:41:00.551032500 INFO:SerialBattery:Venus OS v3.60~87 (large) running on Raspberry Pi 4 Model B Rev 1.5 
2025-06-14 19:41:00.551261500 INFO:SerialBattery:dbus-serialbattery v2.0.20250518dev
2025-06-14 19:41:00.734387500 INFO:SerialBattery:Init of Jkbms_Ble at C8:47:8C:E9:13:DC
2025-06-14 19:41:00.912800500 INFO:SerialBattery:|- Try to connect to Jkbms_Ble at C8:47:8C:E9:13:DC
2025-06-14 19:41:00.937671500 INFO:SerialBattery:Error while connecting to BLE client: BleakDBusError('org.freedesktop.DBus.Error.ServiceUnknown', 'The name org.bluez was not provided by any .service files') of type <class 'bleak.exc.BleakDBusError'> in /data/apps/dbus-serialbattery/bms/jkbms_brn.py line #457
2025-06-14 19:41:02.942398500 INFO:SerialBattery:|- Try to connect to Jkbms_Ble at C8:47:8C:E9:13:DC
2025-06-14 19:41:02.955513500 INFO:SerialBattery:Error while connecting to BLE client: BleakDBusError('org.freedesktop.DBus.Error.ServiceUnknown', 'The name org.bluez was not provided by any .service files') of type <class 'bleak.exc.BleakDBusError'> in /data/apps/dbus-serialbattery/bms/jkbms_brn.py line #457
2025-06-14 19:41:04.962601500 INFO:SerialBattery:|- Try to connect to Jkbms_Ble at C8:47:8C:E9:13:DC
2025-06-14 19:41:04.975706500 INFO:SerialBattery:Error while connecting to BLE client: BleakDBusError('org.freedesktop.DBus.Error.ServiceUnknown', 'The name org.bluez was not provided by any .service files') of type <class 'bleak.exc.BleakDBusError'> in /data/apps/dbus-serialbattery/bms/jkbms_brn.py line #457
2025-06-14 19:41:06.979972500 INFO:SerialBattery:|- Try to connect to Jkbms_Ble at C8:47:8C:E9:13:DC
2025-06-14 19:41:06.993077500 INFO:SerialBattery:Error while connecting to BLE client: BleakDBusError('org.freedesktop.DBus.Error.ServiceUnknown', 'The name org.bluez was not provided by any .service files') of type <class 'bleak.exc.BleakDBusError'> i [... cut ...]
2025-06-14 19:43:26.062441500 INFO:SerialBattery:> CCCM T:     True  | DCCM T:        True
2025-06-14 19:43:26.062980500 INFO:SerialBattery:> CCCM T MOS: True  | DCCM T MOS:    True
2025-06-14 19:43:26.063366500 INFO:SerialBattery:> CCCM SOC:   False | DCCM SOC:      False
2025-06-14 19:43:26.063760500 INFO:SerialBattery:> CHARGE FET: True  | DISCHARGE FET: True | BALANCE FET: True
2025-06-14 19:43:26.064140500 INFO:SerialBattery:Serial Number/Unique Identifier: 2092245176

Thank you again for your help!
Roland

mr-manuel, 2025-06-15:
The only thing you can try is to play with this values:

https://github.com/mr-manuel/venus-os_dbus-serialbattery/blob/2c8dfa8ff4b5c1ccb889f6fa425f80bf0108880e/dbus-serialbattery/config.default.ini#L132-L139

mr-manuel, 2025-06-15:
As I wrote, you have to play with this values and try different combinations.

RolandRd, 2025-06-15:
I misunderstood you, sorry. So I'll try it.

mr-manuel, 2025-07-17:
See https://github.com/mr-manuel/venus-os_dbus-serialbattery/issues/86#issuecomment-2710027233

Victron will not update BlueZ. It has to be updated in the Yocto project.