mr-manuel/venus-os_dbus-serialbattery #284: JKBMS can't see cell volatges after update

tazinblack, 2025-07-02:
### Describe the problem

after update to version v1.6.20250131 i cannot see the cell volatges of my two jkbms devices.
I tried to find this in the default config and played around with the settings. Maybe I just overread something.
This is with stable version + v1 and v2 gui and with nightly version with v2 gui

### Installation method

Install script from the documentation

### Driver version of the currently installed driver

v1.6.20250131

### Driver version of the last known working driver

v1.0.20230531

### Venus OS device type

Raspberry Pi 2

### Venus OS version

v3.55

### Venus OS image type

Normal

### BMS type

JKBMS (Heltec BMS)

### Cell count

16

### Battery count

2

### Connection type

Serial USB adapter to RS485

### USB hub

Yes, no external power supply

### Config file

```ini
MAX_BATTERY_CHARGE_CURRENT = 70.0
MAX_BATTERY_DISCHARGE_CURRENT = 70.0

MIN_CELL_VOLTAGE = 3.0
MAX_CELL_VOLTAGE = 3.45
FLOAT_CELL_VOLTAGE = 3.4

BMS_TYPE = Jkbms

SOC_LOW_WARNING = 8
SOC_LOW_ALARM = 3

TIME_TO_SOC_POINTS = 50, 25, 15, 0
BATTERY_CELL_DATA_FORMAT = 2
PUBLISH_CONFIG_VALUES = True
GUI_PARAMETERS_SHOW_ADDITIONAL_INFO = False
```

### Relevant log output

```shell
2025-07-02 10:38:14.398963500 *** CCGX booted (0) ***
2025-07-02 10:39:45.031567500 INFO:SerialBattery:
2025-07-02 10:39:45.034258500 INFO:SerialBattery:Starting dbus-serialbattery
2025-07-02 10:39:45.039101500 INFO:SerialBattery:Venus OS v3.55 running on Raspberry Pi 2 Model B Rev 1.1
2025-07-02 10:39:45.041297500 INFO:SerialBattery:dbus-serialbattery v1.6.20250131
2025-07-02 10:40:01.064865500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2025-07-02 10:40:01.064876500 INFO:SerialBattery:Testing Jkbms
2025-07-02 10:40:01.159438500 INFO:SerialBattery:-- Connection established to Jkbms
2025-07-02 10:40:03.116107500 INFO:SerialBattery:Reconnected to previously identified battery with DeviceInstance: 1
2025-07-02 10:40:04.450074500 INFO:SerialBattery:Use DeviceInstance: 1
2025-07-02 10:40:04.453079500 INFO:SerialBattery:Use dbus ServiceName: com.victronenergy.battery.ttyACM0
2025-07-02 10:40:04.618249500 INFO:SerialBattery:Polling interval: 1.000 s
2025-07-02 10:40:04.621923500 INFO:SerialBattery:Battery JKBMS connected to dbus from /dev/ttyACM0
2025-07-02 10:40:04.645591500 INFO:SerialBattery:========== Settings ==========
2025-07-02 10:40:04.646921500 INFO:SerialBattery:> Connection voltage: 53.56 V | Current: -1.02 A | SoC: 68%
2025-07-02 10:40:04.648015500 INFO:SerialBattery:> Cell count: 16 | Cells populated: 16
2025-07-02 10:40:04.649008500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2025-07-02 10:40:04.650078500 INFO:SerialBattery:> MIN CELL VOLTAGE: 3.000 V | MAX CELL VOLTAGE: 3.450 V| FLOAT CELL VOLTAGE: 3.400 V
2025-07-02 10:40:04.651292500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 70.0 A | MAX BATTERY DISCHARGE CURRENT: 70.0 A
2025-07-02 10:40:04.657770500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 100.0 A | MAX BATTERY DISCHARGE CURRENT: 100.0 A (read from BMS)
2025-07-02 10:40:04.658919500 INFO:SerialBattery:> CVCM:     True
2025-07-02 10:40:04.659779500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2025-07-02 10:40:04.660965500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2025-07-02 10:40:04.662032500 INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
2025-07-02 10:40:04.662867500 INFO:SerialBattery:> CHARGE FET: True | DISCHARGE FET: True | BALANCE FET: True
2025-07-02 10:40:04.663609500 INFO:SerialBattery:Serial Number/Unique Identifier: JK-B2A24S20P

second BMS:

2025-07-02 10:38:14.449708500 *** CCGX booted (0) ***
2025-07-02 10:39:32.059981500 INFO:SerialBattery:
2025-07-02 10:39:32.061451500 INFO:SerialBattery:Starting dbus-serialbattery
2025-07-02 10:39:32.071225500 INFO:SerialBattery:Venus OS v3.55 running on Raspberry Pi 2 Model B Rev 1.1
2025-07-02 10:39:32.072552500 INFO:SerialBattery:dbus-serialbattery v1.6.20250131
2025-07-02 10:39:48.094379500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2025-07-02 10:39:48.094387500 INFO:SerialBattery:Testing Jkbms
2025-07-02 10:39:48.364354500 INFO:SerialBattery:-- Connection established to Jkbms
2025-07-02 10:39:51.443440500 INFO:SerialBattery:Reconnected to previously identified battery with DeviceInstance: 2
2025-07-02 10:39:52.816483500 INFO:SerialBattery:Use DeviceInstance: 2
2025-07-02 10:39:52.816701500 INFO:SerialBattery:Use dbus ServiceName: com.victronenergy.battery.ttyUSB2
2025-07-02 10:39:53.046453500 INFO:SerialBattery:Polling interval: 1.000 s
2025-07-02 10:39:53.061241500 INFO:SerialBattery:Battery JKBMS connected to dbus from /dev/ttyUSB2
2025-07-02 10:39:53.061255500 INFO:SerialBattery:========== Settings ==========
2025-07-02 10:39:53.061260500 INFO:SerialBattery:> Connection voltage: 53.54 V | Current: 0.64 A | SoC: 71%
2025-07-02 10:39:53.061265500 INFO:SerialBattery:> Cell count: 16 | Cells populated: 16
2025-07-02 10:39:53.067734500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2025-07-02 10:39:53.069182500 INFO:SerialBattery:> MIN CELL VOLTAGE: 3.000 V | MAX CELL VOLTAGE: 3.450 V| FLOAT CELL VOLTAGE: 3.400 V
2025-07-02 10:39:53.071710500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 70.0 A | MAX BATTERY DISCHARGE CURRENT: 70.0 A
2025-07-02 10:39:53.071721500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 100.0 A | MAX BATTERY DISCHARGE CURRENT: 100.0 A (read from BMS)
2025-07-02 10:39:53.073879500 INFO:SerialBattery:> CVCM:     True
2025-07-02 10:39:53.081228500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2025-07-02 10:39:53.081239500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2025-07-02 10:39:53.081243500 INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
2025-07-02 10:39:53.081248500 INFO:SerialBattery:> CHARGE FET: True | DISCHARGE FET: True | BALANCE FET: True
2025-07-02 10:39:53.081253500 INFO:SerialBattery:Serial Number/Unique Identifier: Monsterles_GJK-B2A24S20P
```

### Please confirm that you have completed the following steps before submitting your support request

- [x] I have carefully read all the instructions above including the descriptions.
- [x] I have tried the `nightly` build first, if I was not already using it. This means the logs I provide are from the `nightly` build.
- [x] I have rebooted at least once after I saw the issue.
- [x] I have read the complete [changelog](https://github.com/mr-manuel/dbus-serialbattery/blob/master/CHANGELOG.md).
- [x] I have provided the logfiles in the correct format with a HUMAN READABLE timestamp by piping it through `tai64nlocal`.
- [ ] I have clicked all checkboxes because I did not read the instructions.

onemorename, 2025-07-11:
Had the same issue here, but after one more reboot everything was back. Could be that the GUI needs to be updated first with the customizations right?

mr-manuel, 2025-07-11:
Yes, it's installed by running `/data/apps/dbus-serialbattery/custom-gui-install.sh` with driver v2.