mr-manuel/venus-os_dbus-serialbattery #290: GUI-2 installation failed

pkeberly, 2025-07-22:
### Describe the problem

I tried to update the serial driver see if it might solve a display issue I'm having where the Brief center won't show info when there are 4 info rings selected.   I noticed this during install.   

Installing QML files for GUI v2...
|- >>> ERROR: GUIv2 installation for local display FAILED
|- >>>        Your Venus OS version v3.70~15 is NOT SUPPORTED by this version of the driver.
|- >>>        Update to the latest available driver to solve this issue. If you are using
|- >>>        Venus OS beta, please use the latest beta or nightly version of driver.

### Installation method

Install script from the documentation

### Driver version of the currently installed driver

nightly build "v2.0.20250717dev"

### Venus OS device type

Cerbo GX

### Venus OS version

v3.70~15

### Venus OS image type

Large

### BMS type

Jiabaida JDB BMS (LLT Power, Overkill Solar, Smart BMS)

### Cell count

8

### Battery count

1

### Connection type

Serial USB adapter to TTL

### USB hub

No, the Serial USB adapter is connected directly to the Venus OS device

### Config file

```ini
[DEFAULT]

;------------ Setup for Luna's 8S energy battery ---------
;  12/16/2024 PWK  first connections, 300aH test
;  12/17/2024 PWK  driver update, bump top voltage to 3.52, tune charge ramp to get better balancing.
;  12/22/2024 PWK  lowered CCL/DCL Recovery Threshold to 0.8% (1.2A)
;  01/09/2025 PWK  Updated driver, changed config.ini to match
;  03/26/2025 PWK  Swapped to Cerbo-S after USB fail. changed config.ini to match latest
;  05/08/2025 PWK  Firmware failure ~71 req'd reset, reinstall driver & new config from old data
;  05/11/2025 PWK  Adjust MAX_CELL, Update others to latest BMS version

LOGGING = INFO

; --------- Battery Current Limits ---------
; +++ Limits apply to each individual battery/BMS. +++
; +++ If you have multiple batteries, you need a battery aggregator. +++
MAX_BATTERY_CHARGE_CURRENT    = 120.0
[... 473 lines cut ...]
2025-07-22 23:27:35.931462500 ERROR:SerialBattery:    |- Max cell voltage: 3.238 < Max threshold: 3.350 --> OK
2025-07-22 23:27:35.934668500 ERROR:SerialBattery:    |- Trying further for 60 s, then restart the driver
2025-07-22 23:27:37.976765500 ERROR:SerialBattery:>>> ERROR: Invalid response packet. Expected begin packet character 0xDD
2025-07-22 23:27:37.984552500 ERROR:SerialBattery:>>> ERROR: Incorrect Reply. Expected end packet character 0x77
2025-07-22 23:27:37.984563500 ERROR:SerialBattery:>>> ERROR: Battery does not respond, init/reset values <<<
2025-07-22 23:27:37.984566500 ERROR:SerialBattery:    |- Cell voltages are NOT in a safe threshold to proceed with charging/discharging without communication to the battery.
2025-07-22 23:27:38.000377500 ERROR:SerialBattery:    |- Min cell voltage: 3.231 > Min Threshold: 3.250 --> NOT OK
2025-07-22 23:27:38.008780500 ERROR:SerialBattery:    |- Max cell voltage: 3.238 < Max threshold: 3.350 --> OK
2025-07-22 23:27:38.008790500 ERROR:SerialBattery:    |- Trying further for 60 s, then restart the driver
2025-07-22 23:27:57.233674500 ERROR:SerialBattery:>>> No response from battery. Connection lost or battery not recognized. Check cabeling! [len:33/36]
2025-07-22 23:27:57.238431500 ERROR:SerialBattery:>>> ERROR: Battery does not respond, init/reset values <<<
2025-07-22 23:27:57.239243500 ERROR:SerialBattery:    |- Cell voltages are NOT in a safe threshold to proceed with charging/discharging without communication to the battery.
2025-07-22 23:27:57.240025500 ERROR:SerialBattery:    |- Min cell voltage: 3.231 > Min Threshold: 3.250 --> NOT OK
2025-07-22 23:27:57.240661500 ERROR:SerialBattery:    |- Max cell voltage: 3.238 < Max threshold: 3.350 --> OK
2025-07-22 23:27:57.241152500 ERROR:SerialBattery:    |- Trying further for 60 s, then restart the driver
```

### Please confirm that you have completed the following steps before submitting your support request

- [x] I have carefully read all the instructions above including the descriptions.
- [x] I have tried the `nightly` build first, if I was not already using it. This means the logs I provide are from the `nightly` build.
- [x] I have rebooted at least once after I saw the issue.
- [x] I have read the complete [changelog](https://github.com/mr-manuel/dbus-serialbattery/blob/master/CHANGELOG.md).
- [x] I have provided the logfiles in the correct format with a HUMAN READABLE timestamp by piping it through `tai64nlocal`.
- [ ] I have clicked all checkboxes because I did not read the instructions.

mr-manuel, 2025-07-23:
Hi, thanks for reporting!

Can you post the result of this command?

```
ls -l /opt/victronenergy/
```

There should be no entry with `gui-v2`. Did you maybe do some tests to get GUIv2 working on the display?

pkeberly, 2025-07-23:
I do see an entry for gui-v2 from May, I wouldn't know what to do to test the display.   

[image]

I do have an odd looking display after the latest Victron v3.70~19 update.  

[image]

closed 2025-07-23 by mr-manuel

pkeberly, 2025-07-24:
This is on a Cerbo GX-S.  Is it safe to delete with that device?