mr-manuel/venus-os_dbus-serialbattery #291: JK Serial ID

admileno-tech, 2025-07-24:
### Is your feature request related to a problem? Please describe.

hack for serial to show
nano /data/etc/dbus-serialbattery/bms/jkbms_pb.py

#serial_nr = status_data[86:96].decode("utf-8")
serial_nr = status_data[86:106].decode("utf-8").strip()

The new JK soft is sending user data at the end. Limiting the ID to 10 alphadigits will result in same ID

Result after:
INFO:SerialBattery:Serial Number/Unique Identifier: 504185N2A000297JK2
INFO:SerialBattery:Serial Number/Unique Identifier: 504185N2A000294JK1

Also if the BMS is from the same production batch, only the numbers at the end are diff.

### Describe the solution you'd like

...

### Describe alternatives you've considered

...

mr-manuel, 2025-07-24:
Can you please provide the current protocol documentation?

mr-manuel, 2025-09-28:
Can you please also post a screenshot of the serial ID you see in the app? Then I can merge this.