mr-manuel/venus-os_dbus-serialbattery #294: Comm errors

pkeberly, 2025-07-29:
### Describe the problem

I'm seeing some comm errors related to the JDB BMS over several version of Venus and driver.  The driver appears to work fine, do I have a config issue or will this be safe to ignore?

### Installation method

Install script from the documentation

### Driver version of the currently installed driver

v2.0.20250729dev

### Venus OS device type

Cerbo GX

### Venus OS version

v3.64~1

### Venus OS image type

Large

### BMS type

Jiabaida JDB BMS (LLT Power, Overkill Solar, Smart BMS)

### Cell count

8

### Battery count

1

### Connection type

Serial USB adapter to TTL

### USB hub

Yes, no external power supply

### Config file

```ini
[DEFAULT]

;------------ Setup for Luna's 8S energy battery ---------
;  12/16/2024 PWK  first connections, 300aH test
;  12/17/2024 PWK  driver update, bump top voltage to 3.52, tune charge ramp to get better balancing.
;  12/22/2024 PWK  lowered CCL/DCL Recovery Threshold to 0.8% (1.2A)
;  01/09/2025 PWK  Updated driver, changed config.ini to match
;  03/26/2025 PWK  Swapped to Cerbo-S after USB fail. changed config.ini to match latest
;  05/08/2025 PWK  Firmware failure ~71 req'd reset, reinstall driver & new config from old data
;  05/11/2025 PWK  Adjust MAX_CELL, Update others to latest BMS version

LOGGING = INFO

; --------- Battery Current Limits ---------
; +++ Limits apply to each individual battery/BMS. +++
; +++ If you have multiple batteries, you need a battery aggregator. +++
MAX_BATTERY_CHARGE_CURRENT    = 120.0
MAX_BATTERY_DISCHARGE_CURRENT = 200.0

; --------- Cell Voltages ---------
; Description:
;     Calculates the minimum and maximum battery voltage based on the number of cells and their voltage.
;     ATTENTION: This does not prevent overcharging or overdischarging of individual cells.
[... 811 lines cut ...]
2025-07-29 13:45:01.432318500 INFO:SerialBattery:> CCCM T MOS: True  | DCCM T MOS:    True
2025-07-29 13:45:01.433338500 INFO:SerialBattery:> CCCM SOC:   False | DCCM SOC:      False
2025-07-29 13:45:01.436287500 INFO:SerialBattery:> CHARGE FET: True  | DISCHARGE FET: True | BALANCE FET: True
2025-07-29 13:45:01.440655500 INFO:SerialBattery:Serial Number/Unique Identifier: 39_304.0Ah
2025-07-29 13:45:16.918476500 ERROR:SerialBattery:>>> ERROR: BMS send insufficient data. Received 54 expected 45
2025-07-29 13:45:18.899402500 ERROR:SerialBattery:>>> ERROR: Invalid response packet. Expected begin packet character 0xDD
2025-07-29 13:45:18.899413500 ERROR:SerialBattery:>>> ERROR: BMS send insufficient data. Received 24 expected 7
2025-07-29 13:45:21.948261500 ERROR:SerialBattery:device reports readiness to read but returned no data (device disconnected or multiple access on port?)
2025-07-29 13:45:25.292270500 ERROR:SerialBattery:>>> No response from battery. Connection lost or battery not recognized. Check cabeling!
2025-07-29 13:45:42.969681500 ERROR:SerialBattery:>>> No response from battery. Connection lost or battery not recognized. Check cabeling! [len:52/162]
2025-07-29 13:45:42.976610500 ERROR:SerialBattery:>>> ERROR: Battery does not respond, init/reset values <<<
2025-07-29 13:45:42.977527500 ERROR:SerialBattery:    |- Cell voltages are NOT in a safe threshold to proceed with charging/discharging without communication to the battery.
2025-07-29 13:45:42.978285500 ERROR:SerialBattery:    |- Min cell voltage: 3.245 > Min Threshold: 3.250 --> NOT OK
2025-07-29 13:45:42.979108500 ERROR:SerialBattery:    |- Max cell voltage: 3.252 < Max threshold: 3.350 --> OK
2025-07-29 13:45:42.979655500 ERROR:SerialBattery:    |- Trying further for 60 s, then restart the driver
```

### Please confirm that you have completed the following steps before submitting your support request

- [x] I have carefully read all the instructions above including the descriptions.
- [x] I have tried the `nightly` build first, if I was not already using it. This means the logs I provide are from the `nightly` build.
- [x] I have rebooted at least once after I saw the issue.
- [x] I have read the complete [changelog](https://github.com/mr-manuel/dbus-serialbattery/blob/master/CHANGELOG.md).
- [x] I have provided the logfiles in the correct format with a HUMAN READABLE timestamp by piping it through `tai64nlocal`.
- [ ] I have clicked all checkboxes because I did not read the instructions.

mr-manuel, 2025-07-29:
> Also, does the driver adjust the GUI2 at all? I often get strange display changes that I can't explain.

Please explain better what you mean with that.

pkeberly, 2025-07-30:
This is what I normally see. 
[image]

Recently the left 2 tiles change size when inverting and the animations are gone
[image]

I've also seem reverse animations.  Sometime on shore, sometimes while inverting.  I understand it's beta software and I'm not complaining but I'd like to understand how the driver and Venus play together so I know where to start looking when an issue pops up.  What I do understand is the inconsistency.   
https://github.com/user-attachments/assets/796c928f-be81-4e3a-abe2-66b80ed97636

mr-manuel, 2025-07-30:
The driver is using the latest available GUIv2 version when the driver stable version was released. Could be that this was already fixed in upstream. If yes, then it will be in the next driver release.