mr-manuel/venus-os_dbus-serialbattery #297: JK-PB2A16S15PV19 4.3 multi battery not working

jjtoard, 2025-08-06:
### Describe the problem

Today I received my cable for the jk bms JK-PB2A16S15PV19 4.3
It doesn't look to work on the individual RS485.
It doesn't look to work if I plug it into battery with address 1 but doesn't work when connected to battery with address 0.

If the two batteries are connected together via RS485 I do see battery 1 when connected to battery 1 but don't see battery 0. I get a cable error warning at that point. It doesn't work when connected to battery 0
At the end the battery disappears

[image]

Connected to battery 1 changed battery0 to battery2 but that doesn't work. It also brakes the pack config on the BMS

### Installation method

Install script from the documentation

### Driver version of the currently installed driver

/v2.0.20250729

### Venus OS device type

Raspberry Pi 3

### Venus OS version

3.70-24

### Venus OS image type

Normal

### BMS type

JKBMS Inverter (JKBMS PB Model)

### Cell count

16

### Battery count

2

### Connection type

Serial USB adapter to RS485

### USB hub

No, the Serial USB adapter is connected directly to the Venus OS device

### Config file

```ini
It works for one battery but not for 2.  See description
```

### Relevant log output

```shell
is working but only for one battery and only on address 1
```

### Please confirm that you have completed the following steps before submitting your support request

- [x] I have carefully read all the instructions above including the descriptions.
- [x] I have tried the `nightly` build first, if I was not already using it. This means the logs I provide are from the `nightly` build.
- [x] I have rebooted at least once after I saw the issue.
- [x] I have read the complete [changelog](https://github.com/mr-manuel/dbus-serialbattery/blob/master/CHANGELOG.md).
- [x] I have provided the logfiles in the correct format with a HUMAN READABLE timestamp by piping it through `tai64nlocal`.
- [ ] I have clicked all checkboxes because I did not read the instructions.

edwardvanhazendonk, 2025-08-11:
Hi jjtoard,

I tried this myself this evening, for me it is working, which protocol do you set on your JKBMS for the UART mode, as this might influence your outcome?

Which port did you use to connect the BMS to the Venus?

I also see 1 BMS only, but my BMS is address 0 so it looks to work.

Thinking about doing an RS485 bus on the RS485-1 port of the JKBMS to have only 1 connection, but ran out of time today, maybe somewhere this week. (so not a bus on the RS485-2 where all JKBMSes are)

I used the RS485-1 port most left in my case, wiring: UTP 1 is RS495-B, UTP 2 is RS485-A, UTP 3 is GND (I use a waveshare 4 port isolated RS485 to USB)

Cheers, Edward

jjtoard, 2025-08-13:
jk bms has 3 rs485 connectors .  one on the left and 2 at the right aside eachother.
It looks to only work on the right connectors

mr-manuel, 2025-08-15:
In case you missed that: https://mr-manuel.github.io/venus-os_dbus-serialbattery_docs/general/connect#jkbms-pb-model-also-know-as-jk-inverter-bms

closed 2025-08-15 by mr-manuel

jjtoard, 2025-08-15:
Sorry another question.  Beside the two v19 batteries I have two old batteries with an older JK BMS.  I also would like to connect them.  Is that possible ?  

I already added this statement to the config.ini USE_PORT_AS_UNIQUE_ID = True

mr-manuel, 2025-08-15:
You posted the question also as a new issue, which is better since it's another problem. For reference: https://github.com/mr-manuel/venus-os_dbus-serialbattery/issues/301