mr-manuel/venus-os_dbus-serialbattery #304: Driver blocking charge on RS485 JKBMS

spectre338, 2025-08-16:
### Describe the problem

I’ve been struggling ever since my JBD BMSs started clicking and stopped working properly. I replaced them with two JKBMS units, using the BMS_TYPE setting Jkbms_pe. The batteries discharge fine, but they do not charge.

When I check the UI, I see it’s running V1 (since V2 stopped working after the last reflash). The BMS is not blocking charge — I confirmed this by testing with another Victron charger, and it was able to charge the batteries.

Do I need to add something specific to the config.ini file to get charging to work correctly?

[image]

### Installation method

Install script from the documentation

### Driver version of the currently installed driver

v2.0.20250729

### Driver version of the last known working driver

v2.0.20250729

### Venus OS device type

Cerbo GX

### Venus OS version

v3.64 

### Venus OS image type

Large

### BMS type

JKBMS Inverter (JKBMS PB Model)

### Cell count

8s

### Battery count

2

### Connection type

Serial USB adapter to RS485

### USB hub

Yes, with external power supply

### Config file

```ini
root@einstein:~# cat /data/apps/dbus-serialbattery/config.ini

[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.
; NOTICE: Do not copy the whole file, but only the values/settings you want to change.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):
 MAX_BATTERY_CHARGE_CURRENT = 50.0
 MAX_BATTERY_DISCHARGE_CURRENT = 60.0

LOGGING = DEBUG
[... 121 lines cut ...]
2025-08-16 17:18:15.271957500 DEBUG:SerialBattery:Polling data took 0.252 seconds
2025-08-16 17:18:16.346784500 DEBUG:SerialBattery:Setted setting /Settings/Devices/serialbattery_503142E1157/HistoryValues to {"clear": 0.0, "deepest_discharge": -225.43400000000003, "last_discharge": 3.661902211529148, "total_ah_drawn": 3.661902211529149, "full_discharges": 0.0, "minimum_voltage": 25.807, "maximum_voltage": 26.653, "minimum_cell_voltage": 3.191, "maximum_cell_voltage": 3.345, "low_voltage_alarms": 0.0, "high_voltage_alarms": 0.0, "minimum_temperature": 0.0, "maximum_temperature": 27.4, "discharged_energy": 0.09467941077070595, "charged_energy": 0.21103022139648758}
2025-08-16 17:18:16.395423500 DEBUG:SerialBattery:Saved HistoryValues. Before {"clear": 0.0, "deepest_discharge": -225.43400000000003, "last_discharge": 3.661902211529148, "total_ah_drawn": 3.661902211529149, "full_discharges": 0.0, "minimum_voltage": 25.807, "maximum_voltage": 26.647, "minimum_cell_voltage": 3.191, "maximum_cell_voltage": 3.343, "low_voltage_alarms": 0.0, "high_voltage_alarms": 0.0, "minimum_temperature": 0.0, "maximum_temperature": 27.2, "discharged_energy": 0.09467941077070595, "charged_energy": 0.20015593591218597}, after {"clear": 0.0, "deepest_discharge": -225.43400000000003, "last_discharge": 3.661902211529148, "total_ah_drawn": 3.661902211529149, "full_discharges": 0.0, "minimum_voltage": 25.807, "maximum_voltage": 26.653, "minimum_cell_voltage": 3.191, "maximum_cell_voltage": 3.345, "low_voltage_alarms": 0.0, "high_voltage_alarms": 0.0, "minimum_temperature": 0.0, "maximum_temperature": 27.4, "discharged_energy": 0.09467941077070595, "charged_energy": 0.21103022139648758}
2025-08-16 17:18:16.399674500 DEBUG:SerialBattery:logged to dbus [58]
2025-08-16 17:18:16.399683500 DEBUG:SerialBattery:Cells:[1]3.325V [2]3.339V [3]3.345V [4]3.324V [5]3.325V [6]3.334V [7]3.339V [8]3.324V
2025-08-16 17:18:16.399687500 DEBUG:SerialBattery:Polling data took 0.378 seconds
2025-08-16 17:18:17.249944500 DEBUG:SerialBattery:logged to dbus [58]
2025-08-16 17:18:17.249952500 DEBUG:SerialBattery:Cells:[1]3.324V [2]3.339V [3]3.345V [4]3.324V [5]3.325V [6]3.334V [7]3.339V [8]3.324V
2025-08-16 17:18:17.249956500 DEBUG:SerialBattery:Polling data took 0.226 seconds
2025-08-16 17:18:18.212407500 DEBUG:SerialBattery:logged to dbus [58]
2025-08-16 17:18:18.216763500 DEBUG:SerialBattery:Cells:[1]3.324V [2]3.339V [3]3.345V [4]3.324V [5]3.325V [6]3.334V [7]3.339V [8]3.325V
2025-08-16 17:18:18.231581500 DEBUG:SerialBattery:Polling data took 0.200 seconds
2025-08-16 17:18:19.237160500 DEBUG:SerialBattery:logged to dbus [58]
2025-08-16 17:18:19.239241500 DEBUG:SerialBattery:Cells:[1]3.325V [2]3.339V [3]3.345V [4]3.324V [5]3.325V [6]3.334V [7]3.339V [8]3.325V
2025-08-16 17:18:19.240265500 DEBUG:SerialBattery:Polling data took 0.211 seconds
```

### Please confirm that you have completed the following steps before submitting your support request

- [x] I have carefully read all the instructions above including the descriptions.
- [x] I have tried the `nightly` build first, if I was not already using it. This means the logs I provide are from the `nightly` build.
- [x] I have rebooted at least once after I saw the issue.
- [x] I have read the complete [changelog](https://github.com/mr-manuel/dbus-serialbattery/blob/master/CHANGELOG.md).
- [x] I have provided the logfiles in the correct format with a HUMAN READABLE timestamp by piping it through `tai64nlocal`.
- [ ] I have clicked all checkboxes because I did not read the instructions.

spectre338, 2025-08-16:
start of the relog after restart.sh

2025-08-16 17:22:37.737419500 INFO:SerialBattery:  Testing Jkbms_pb at address "\x01" with 115200 baud
2025-08-16 17:22:38.207798500 INFO:SerialBattery:-- Connection established to Jkbms_pb
2025-08-16 17:22:38.978686500 INFO:SerialBattery:Reconnected to previously identified battery with DeviceInstance: 1
2025-08-16 17:22:39.140526500 INFO:SerialBattery:Use DeviceInstance: 1
2025-08-16 17:22:39.144754500 INFO:SerialBattery:Use dbus ServiceName: com.victronenergy.battery.ttyACM3
2025-08-16 17:22:39.245769500 INFO:SerialBattery:Polling interval: 1.000 s
2025-08-16 17:22:39.268895500 INFO:SerialBattery:Battery JKBMS PB Model connected to dbus from /dev/ttyACM3
2025-08-16 17:22:39.277375500 INFO:SerialBattery:========== Settings ==========
2025-08-16 17:22:39.279562500 INFO:SerialBattery:> Connection voltage: 26.688 V | Current: 24.4 A | SoC: 59%
2025-08-16 17:22:39.290356500 INFO:SerialBattery:> Cell count: 8 | Cells populated: 8
2025-08-16 17:22:39.290365500 INFO:SerialBattery:> CHARGE MODE: Linear
2025-08-16 17:22:39.290367500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.900 V | MAX CELL VOLTAGE: 3.450 V| FLOAT CELL VOLTAGE: 3.375 V
2025-08-16 17:22:39.310551500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 50.0 A | MAX BATTERY DISCHARGE CURRENT: 60.0 A
2025-08-16 17:22:39.310561500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 40.0 A | MAX BATTERY DISCHARGE CURRENT: 150.0 A (read from BMS)
2025-08-16 17:22:39.310566500 INFO:SerialBattery:> CVCM:       True
2025-08-16 17:22:39.310568500 INFO:SerialBattery:> CCCM CV:    True  | DCCM CV:       True
2025-08-16 17:22:39.310570500 INFO:SerialBattery:> CCCM T:     True  | DCCM T:        True
2025-08-16 17:22:39.310573500 INFO:SerialBattery:> CCCM T MOS: True  | DCCM T MOS:    True
2025-08-16 17:22:39.310699500 INFO:SerialBattery:> CCCM SOC:   False | DCCM SOC:      False
2025-08-16 17:22:39.310703500 INFO:SerialBattery:> CHARGE FET: 1     | DISCHARGE FET: 1 | BALANCE FET: True
2025-08-16 17:22:39.310706500 INFO:SerialBattery:Serial Number/Unique Identifier: 503142E1157

mr-manuel, 2025-08-16:
https://mr-manuel.github.io/venus-os_dbus-serialbattery_docs/v1.6.20250131/troubleshoot/#driver-behavior-problems

spectre338, 2025-08-16:
My BMS is JK-B2A24S15P

spectre338, 2025-08-16:
[image]

mr-manuel, 2025-08-16:
[Is ](https://mr-manuel.github.io/venus-os_dbus-serialbattery_docs/troubleshoot/#driver-behavior-problems)

mr-manuel, 2025-08-16:
https://mr-manuel.github.io/venus-os_dbus-serialbattery_docs/troubleshoot/#driver-behavior-problems

spectre338, 2025-08-16:
i added the 
GUI_PARAMETERS_SHOW_ADDITIONAL_INFO = True
CVCM_ENABLE = True

to my config and rebooted. then I took the screen shot, i do not have the additional info

mr-manuel, 2025-08-16:
This is another link as the first one.

spectre338, 2025-08-16:
I understand that GUIv2 is what i need to use but this is what i see on GUIv2

[image] and this is what i seet on GUIv1 

[image]

mr-manuel, 2025-08-16:
Since you are watching the wrong page. Here again the right link https://mr-manuel.github.io/venus-os_dbus-serialbattery_docs/troubleshoot/#driver-behavior-problems

mr-manuel, 2025-08-16:
In your GUIv1 screenshot you see the reason, but you have to go up.

mr-manuel, 2025-08-16:
[image]

### Driver v2.x.x
`TEMPERATURE_SOURCE_BATTERY = 1, 2`
https://github.com/mr-manuel/venus-os_dbus-serialbattery/blob/297d048f8a909e9474990e8e10cedb018aaa49da/dbus-serialbattery/config.default.ini#L513-L531

### Driver v1.x.x
`TEMP_BATTERY = 1`
OR
`TEMP_BATTERY = 2`
https://github.com/mr-manuel/venus-os_dbus-serialbattery/blob/01c01b88d785912c3eccf6c5ef96569db57f3285/etc/dbus-serialbattery/config.default.ini#L431-L450

spectre338, 2025-08-16:
thank you I totally missed that! i made the change to my config and I still have the issue

 TEMPERATURE_SOURCE_BATTERY = 1, 2 
  
 ; Temperature sensor 1 name 
 TEMPERATURE_1_NAME = Temp 1 
  
 ; Temperature sensor 2 name 
 TEMPERATURE_2_NAME = Temp 2

mr-manuel, 2025-08-16:
Please add the new screenshot

spectre338, 2025-08-16:
I figured out the issue — it was from copying and pasting on Windows. After running dos2unix, everything works fine now. Thank you!

spectre338, 2025-08-16:
[image]