mr-manuel/venus-os_dbus-serialbattery #309: v2.0.20250729: No response from battery with JK_B2A20S20P

SenH, 2025-08-31:
### Describe the bug

I tried to upgrade today from v1.6.20250131 to v2.0.20250729.
V2.0 is able to connect to the BMS but SerialBattery does not seem to able to communicate?
There is also Python non-blocking exception logged. I downgraded to v1.6.20250131 and all is fine again.

Although I have `BLOCK_ON_DISCONNECT = True` in my `config.ini` the v2 driver kept charging/discharging without communication. Is this correct behaviour? According to the documentation in config.ini it should only do that with `BLOCK_ON_DISCONNECT = False`

**log v2.0.20250729**
```2025-08-31 14:36:54.954712500 INFO:SerialBattery:
2025-08-31 14:36:54.955457500 INFO:SerialBattery:Starting dbus-serialbattery
2025-08-31 14:36:54.959189500 INFO:SerialBattery:Venus OS v3.54 (normal) running on Cerbo-S GX
2025-08-31 14:36:54.959870500 INFO:SerialBattery:dbus-serialbattery v2.0.20250729
2025-08-31 14:37:10.969321500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2025-08-31 14:37:10.970088500 INFO:SerialBattery:  Testing Jkbms with 115200 baud
2025-08-31 14:37:11.046581500 INFO:SerialBattery:-- Connection established to Jkbms
2025-08-31 14:37:11.858741500 INFO:SerialBattery:Reconnected to previously identified battery with DeviceInstance: 1
2025-08-31 14:37:11.924739500 ERROR:root:Failed to add setting /Settings/Devices/serialbattery_JK_B2A20S20P/SocCalc, error -5
2025-08-31 14:37:11.926139500 INFO:SerialBattery:Use DeviceInstance: 1
2025-08-31 14:37:11.928160500 INFO:SerialBattery:Use dbus ServiceName: com.victronenergy.battery.ttyUSB0
2025-08-31 14:37:11.990810500 INFO:SerialBattery:Polling interval: 1.000 s
2025-08-31 14:37:11.993775500 INFO:SerialBattery:Battery JKBMS connected to dbus from /dev/ttyUSB0
2025-08-31 14:37:11.995256500 INFO:SerialBattery:========== Settings ==========
2025-08-31 14:37:11.995260500 INFO:SerialBattery:> Connection voltage: 53.22 V | Current: 4.93 A | SoC: 58% | SoC calc: 42%
2025-08-31 14:37:11.995717500 INFO:SerialBattery:> Cell count: 16 | Cells populated: 16
2025-08-31 14:37:11.996512500 INFO:SerialBattery:> CHARGE MODE: Linear
2025-08-31 14:37:12.029417500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.900 V | MAX CELL VOLTAGE: 3.450 V| FLOAT CELL VOLTAGE: 3.375 V
2025-08-31 14:37:12.032793500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 70.0 A | MAX BATTERY DISCHARGE CURRENT: 70.0 A
2025-08-31 14:37:12.032803500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 120.0 A | MAX BATTERY DISCHARGE CURRENT: 120.0 A (read from BMS)
2025-08-31 14:37:12.032809500 INFO:SerialBattery:> CVCM:       True
2025-08-31 14:37:12.033695500 INFO:SerialBattery:> CCCM CV:    True  | DCCM CV:       True
2025-08-31 14:37:12.035618500 INFO:SerialBattery:> CCCM T:     True  | DCCM T:        True
2025-08-31 14:37:12.035870500 INFO:SerialBattery:> CCCM T MOS: True  | DCCM T MOS:    True
2025-08-31 14:37:12.051676500 INFO:SerialBattery:> CCCM SOC:   False | DCCM SOC:      False
2025-08-31 14:37:12.054061500 INFO:SerialBattery:> CHARGE FET: True  | DISCHARGE FET: True | BALANCE FET: True
2025-08-31 14:37:12.056357500 INFO:SerialBattery:Serial Number/Unique Identifier: JK_B2A20S20P
2025-08-31 14:37:13.017058500 ERROR:SerialBattery:CRC checksum mismatch: Expected 0x0cfe, Got 0x0749
2025-08-31 14:37:13.322564500 ERROR:SerialBattery:Non blocking exception occurred: TypeError("unsupported operand type(s) for -: 'NoneType' and 'NoneType'") of type <class 'TypeError'> in /data/apps/dbus-serialbattery/dbushelper.py line #1082
2025-08-31 14:37:38.982516500 ERROR:SerialBattery:>>> No response from battery. Connection lost or battery not recognized. Check cabeling! [len:274/781]
2025-08-31 14:37:38.983793500 ERROR:SerialBattery:>>> ERROR: Battery does not respond, init/reset values <<<
2025-08-31 14:37:38.986578500 ERROR:SerialBattery:    |- Cell voltages are in a safe threshold to proceed with charging/discharging without communication to the battery.
2025-08-31 14:37:38.986589500 ERROR:SerialBattery:    |- Min cell voltage: 3.324 > Min Threshold: 3.250 --> OK
2025-08-31 14:37:38.987091500 ERROR:SerialBattery:    |- Max cell voltage: 3.329 < Max threshold: 3.350 --> OK
2025-08-31 14:37:38.987828500 ERROR:SerialBattery:    |- Trying further for 1200 s, then restart the driver
2025-08-31 14:37:56.046093500 ERROR:SerialBattery:>>> No response from battery. Connection lost or battery not recognized. Check cabeling! [len:32/14915]
2025-08-31 14:37:56.050104500 ERROR:SerialBattery:>>> ERROR: Battery does not respond, init/reset values <<<
2025-08-31 14:37:56.051198500 ERROR:SerialBattery:    |- Cell voltages are in a safe threshold to proceed with charging/discharging without communication to the battery.
2025-08-31 14:37:56.052223500 ERROR:SerialBattery:    |- Min cell voltage: 3.325 > Min Threshold: 3.250 --> OK
2025-08-31 14:37:56.053176500 ERROR:SerialBattery:    |- Max cell voltage: 3.330 < Max threshold: 3.350 --> OK
2025-08-31 14:37:56.053835500 ERROR:SerialBattery:    |- Trying further for 1200 s, then restart the driver```

**log v1.6.20250123** (after downgrade)
2025-08-31 15:14:19.323317500 *** CCGX booted (0) ***
2025-08-31 15:15:29.355206500 INFO:SerialBattery:
2025-08-31 15:15:29.355849500 INFO:SerialBattery:Starting dbus-serialbattery
2025-08-31 15:15:29.358610500 INFO:SerialBattery:Venus OS v3.54 running on Cerbo-S GX
2025-08-31 15:15:29.359512500 INFO:SerialBattery:dbus-serialbattery v1.6.20250131
2025-08-31 15:15:45.374759500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2025-08-31 15:15:45.375460500 INFO:SerialBattery:Testing Jkbms
2025-08-31 15:15:45.449407500 INFO:SerialBattery:-- Connection established to Jkbms
2025-08-31 15:15:46.140509500 INFO:SerialBattery:Reconnected to previously identified battery with DeviceInstance: 1
2025-08-31 15:15:46.444469500 INFO:SerialBattery:Use DeviceInstance: 1
2025-08-31 15:15:46.446227500 INFO:SerialBattery:Use dbus ServiceName: com.victronenergy.battery.ttyUSB0
2025-08-31 15:15:46.484449500 INFO:SerialBattery:Polling interval: 1.000 s
2025-08-31 15:15:46.486778500 INFO:SerialBattery:Battery JKBMS connected to dbus from /dev/ttyUSB0
2025-08-31 15:15:46.486787500 INFO:SerialBattery:========== Settings ==========
2025-08-31 15:15:46.487930500 INFO:SerialBattery:> Connection voltage: 53.12 V | Current: -0.0 A | SoC: 60% | SoC calc: 44%
2025-08-31 15:15:46.493104500 INFO:SerialBattery:> Cell count: 16 | Cells populated: 16
2025-08-31 15:15:46.493268500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2025-08-31 15:15:46.494434500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.900 V | MAX CELL VOLTAGE: 3.450 V| FLOAT CELL VOLTAGE: 3.375 V
[... 82 lines cut ...]
; Keep charge Current fraction at 1.00 @ 15°C
TEMPERATURES_WHILE_CHARGING      =    0,    2,    5,   10,   15,   20,   35,   40,   55
MAX_CHARGE_CURRENT_T_FRACTION    = 0.00, 0.10, 0.20, 0.40, 1.00, 1.00, 1.00, 0.40, 0.00

TEMPERATURE_1_NAME = T1 C5
TEMPERATURE_2_NAME = T2 C13

BMS_TYPE = Jkbms
BLOCK_ON_DISCONNECT = True
```

### Relevant log output

```shell
See above
```

### Please confirm that you have completed the following steps before submitting your support request

- [x] I have carefully read all the instructions above including the descriptions.
- [x] I have tried the `nightly` build first, if I was not already using it. This means the logs I provide are from the `nightly` build.
- [x] I have rebooted at least once after I saw the issue.
- [x] I have read the complete [changelog](https://github.com/mr-manuel/dbus-serialbattery/blob/master/CHANGELOG.md).
- [x] I have provided the logfiles in the correct format with a HUMAN READABLE timestamp by piping it through `tai64nlocal`.
- [ ] I have clicked all checkboxes because I did not read the instructions.

Smarty73, 2025-09-21:
Is there a solution yet? I just bought the same BMS and want to use it.

One remark about this part:

Although I have BLOCK_ON_DISCONNECT = True in my config.ini the v2 driver kept charging/discharging without communication. Is this correct behaviour? According to the documentation in config.ini it should only do that with BLOCK_ON_DISCONNECT = False

I think it will only disconnect after some time if set to false, depending on the last measured cell voltages, see the explanation in the default config file.

Many greetings,
Smarty

Smarty73, 2025-09-24:
Did you try to download the files again? (See the CRC checksum error)

mr-manuel, 2025-09-28:
Inproper cabeling cannot be a bug. If it would be a bug also a lot of other people would have this issue. What @Smarty73 is doing, is the same procedure for resetting the VE.Bus and therefore also removing the BMS controlling from the MultiPlus, if the BMS is not found after. Not sure how exactly you verify that it‘s working properly. Did you check the logs or it‘s working because there is no error in the GUI? 😅 

I checked the issue with the block on disconnect and was not able to reproduce it. 

The logging in v2 was improved, therefore it‘s easily possible that the issue exists in v1.6.x but is not shown.

That issue should be more spread, if it‘s really a driver issue.

Smarty73, 2025-09-28:
Of course improper cabeling is not a bug :-)  But I thought that this might not have been the reason for this behaving. Maybe at some installations the update will not create a proper "first setup" to the MultiPlus or something like that? In my case everything seemed to work, I got all the informations available by the JKBMS in the Venus surface as well as in the MultiPlus page (e.g. single cells voltages, temperatures and so on.) Nothing was missing. Nevertheless the MultiPlus had a (non resettable) error "BMS communication lost" and therefor there was no charge or discharge done.
This issue can be easily recognized even without logging, as the battery simply doesn't charge or discharge.
But unfortunatelly I didn't create any logfiles when the problem occured, sorry for that.

mr-manuel, 2025-09-28:
The question is, if you now have a clean logfile?

SenH, 2025-09-28:
Not sure why marked as off-topic and not sure why @Smarty73  is polluting the the issue.

I can only conclude that in `v1.6.20250131` I don't have any errors what so ever. 
* So If there are cabling errors why don't they show up as CRC errors in `v1.6.20250131`
* I don't  see missing values in VRM statistics. 
* Data on dbus is coming in every second.

Since the system is my "production" system at home, I don't like to fiddle with it and no issue whatsoever to stay on`v1.6.20250131`

Also the spelling in the source code is wrong: `cabeling` --> `cabling`

Smarty73, 2025-09-28:
> not sure why [@Smarty73](https://github.com/Smarty73) is polluting the the issue.
Sorry, I just wanted to help to solve your problem, as I have the same issue. No reason to get unfriendly...

> The question is, if you now have a clean logfile?
If this question was for me: Well, kind of. Somehow both logfiles have a time stamp from yesterday. Maybe because the logging is set to INFO. Content is the same:

@4000000068d7b8a82d1aa75c INFO:SerialBattery:
@4000000068d7b8a82d2b29ec INFO:SerialBattery:Starting dbus-serialbattery
@4000000068d7b8a82d80eef4 INFO:SerialBattery:Venus OS v3.66 (large) running on MultiPlus-II 
@4000000068d7b8a82d88b724 INFO:SerialBattery:dbus-serialbattery v2.0.20250729
@4000000068d7b8b82d9f775c INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
@4000000068d7b8b82da65144 INFO:SerialBattery:  Testing Jkbms with 115200 baud
@4000000068d7b8b8368757cc INFO:SerialBattery:-- Connection established to Jkbms
@4000000068d7b8b9112f77ac INFO:SerialBattery:Reconnected to previously identified battery with DeviceInstance: 1
@4000000068d7b8b915dfa5c4 INFO:SerialBattery:Use DeviceInstance: 1
@4000000068d7b8b915f8ee6c INFO:SerialBattery:Use dbus ServiceName: com.victronenergy.battery.ttyUSB0
@4000000068d7b8b91b57716c INFO:SerialBattery:Polling interval: 1.000 s
@4000000068d7b8b91ca0933c INFO:SerialBattery:Battery JKBMS connected to dbus from /dev/ttyUSB0
@4000000068d7b8b91ca0ae94 INFO:SerialBattery:========== Settings ==========
@4000000068d7b8b91ca0ba4c INFO:SerialBattery:> Connection voltage: 48.93 V | Current: -0.62 A | SoC: 43%
@4000000068d7b8b91ca0c21c INFO:SerialBattery:> Cell count: 16 | Cells populated: 16
@4000000068d7b8b91ca0cdd4 INFO:SerialBattery:> CHARGE MODE: Linear
@4000000068d7b8b91ca0d5a4 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.900 V | MAX CELL VOLTAGE: 3.500 V| FLOAT CELL VOLTAGE: 3.375 V
@4000000068d7b8b91ca1eafc INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 40.0 A | MAX BATTERY DISCHARGE CURRENT: 60.0 A
@4000000068d7b8b91ca1fa9c INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 45.0 A | MAX BATTERY DISCHARGE CURRENT: 65.0 A (read from BMS)
@4000000068d7b8b91ca20a3c INFO:SerialBattery:> CVCM:       True
@4000000068d7b8b91ca2120c INFO:SerialBattery:> CCCM CV:    True  | DCCM CV:       True
@4000000068d7b8b91ca2779c INFO:SerialBattery:> CCCM T:     True  | DCCM T:        True
@4000000068d7b8b91ca27f6c INFO:SerialBattery:> CCCM T MOS: True  | DCCM T MOS:    True
@4000000068d7b8b91ca28b24 INFO:SerialBattery:> CCCM SOC:   False | DCCM SOC:      False
@4000000068d7b8b91ca296dc INFO:SerialBattery:> CHARGE FET: True  | DISCHARGE FET: True | BALANCE FET: True
@4000000068d7b8b91d5a2edc INFO:SerialBattery:Serial Number/Unique Identifier: 407272D1125