mr-manuel/venus-os_dbus-serialbattery #322: serialbattery is not running after installation nor reboot

tobio-de, 2025-09-25:
### Describe the problem

Hi there!

My Batteries where connected via USB, running on Venus OS 3.1*. I had issues, that every time Battery went under 5%, serialbattery stoppt working. So I updated my Cerbo to latest release and reinstalled serialbattery. First try was with release and an old config file - second try was with release and the new nearly empty config.ini, third try was with the nightly rebuild.

Every time I had the same issue: serialbattery did not stard running atomatically. In /service/ there is no dbus-blebattery.*.  When I do sh */enable.sh the services are generated and serialbattery is running - but after reboot they are gone again.

What do I do wrong?

### Installation method

Install script from the documentation

### Driver version of the currently installed driver

2.1.20250822dev

### Venus OS device type

Cerbo GX

### Venus OS version

3.66

### Venus OS image type

Large

### BMS type

JKBMS (Heltec BMS)

### Cell count

17

### Battery count

2

### Connection type

Bluetooth

### USB hub

⚠️ Choose your option ⚠️

### Config file

```ini
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.
; NOTICE: Do not copy the whole file, but only the values/settings you want to change.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):
MAX_BATTERY_CHARGE_CURRENT = 200.0
MAX_BATTERY_DISCHARGE_CURRENT = 200.0

BLUETOOTH_BMS = Jkbms_Ble C8:47:8C:10:08:92, Jkbms_Ble C8:47:8C:10:31:8D
```

### Relevant log output

```shell
2025-09-24 06:01:04.246133500 *** CCGX booted (0) ***
[... 97 lines cut ...]
2025-09-24 06:33:29.329175500 INFO:SerialBattery:Processing frame with settings info

==> /data/log/dbus-blebattery.0/current <==
2025-09-24 06:34:30.235661500 INFO:SerialBattery:|- Try to connect to Jkbms_Ble at C8:47:8C:10:08:92
2025-09-24 06:34:32.461663500 INFO:SerialBattery:|- Device connected, check if it's really a JKBMS

==> /data/log/dbus-blebattery.1/current <==
2025-09-24 06:34:32.205762500 INFO:SerialBattery:|- Try to connect to Jkbms_Ble at C8:47:8C:10:31:8D

==> /data/log/dbus-blebattery.0/current <==
2025-09-24 06:34:33.721140500 INFO:SerialBattery:Processing frame with settings info

==> /data/log/dbus-blebattery.1/current <==
2025-09-24 06:34:33.763790500 INFO:SerialBattery:|- Device connected, check if it's really a JKBMS
2025-09-24 06:34:35.060180500 INFO:SerialBattery:Processing frame with settings info
```

### Please confirm that you have completed the following steps before submitting your support request

- [x] I have carefully read all the instructions above including the descriptions.
- [x] I have tried the `nightly` build first, if I was not already using it. This means the logs I provide are from the `nightly` build.
- [x] I have rebooted at least once after I saw the issue.
- [x] I have read the complete [changelog](https://github.com/mr-manuel/dbus-serialbattery/blob/master/CHANGELOG.md).
- [x] I have provided the logfiles in the correct format with a HUMAN READABLE timestamp by piping it through `tai64nlocal`.
- [ ] I have clicked all checkboxes because I did not read the instructions.

mr-manuel, 2025-09-28:
Very likely you have other scripts in the startup scripts that fail and prevent dbus-serialbattery from running.

Post the output of

~~~bash
cat /data/rc.local

cat /data/rcS.local

cat /data/apps/dbus-serialbattery/startup.log
~~~

tobio-de, 2025-09-28:
root@einstein:~# ls -l /data
drwxr-xr-x    7 root     root          4096 Sep 24 06:59 apps
drwxr-xr-x    5 root     root          4096 Sep 28 10:17 conf
drwxr-xr-x    2 www-data root          4096 Sep 28 10:17 db
drwxr-xr-x    4 root     root          4096 Aug 23  2023 dbus-kaco_blueplanet
drwxr-xr-x    3 root     root          4096 Sep 17 10:18 etc
drwxr-xr-x    5 root     root          4096 Feb 23  2024 home
drwx------    2 root     root          4096 Sep 17 08:32 keys
drwxr-xr-x   57 root     root          4096 Sep 28 10:16 log
drwx------    2 root     root         16384 Jan  1  1970 lost+found
drwxrwxrwx    2 simple-u 123           4096 May 31  2023 rc
-rw-r--r--    1 root     root           588 Sep 24 07:18 rc.local
-rw-r--r--    1 root     root           292 Feb 22  2024 rc.local~
-rwxr-xr-x    1 root     root            90 Sep 17 10:18 rcS.local
drwxr-xr-x    3 root     root          4096 Jan  1  1970 themes
drwxr-xr-x    3 root     root          4096 Jan  1  1970 var
drwxr-xr-x    2 root     root          4096 Sep 28 10:16 venus
drwxr-xr-x    2 root     root          4096 Sep 24 12:46 vrmfilescache
root@einstein:~#
root@einstein:~# cat /data/rc.local
!#/bin/bash
ln -s /data/dbus-kaco_blueplanet/service/ /opt/victronenergy/service/dbus-kaco_blueplanet
/data/dbus-loxone-smartmeter/install.sh
ln -s /data/dbus-loxone-smartmeter/service /opt/victronenergy/service/dbus-loxone-smartmeter
ln -s /data/dbus-aggregate-batteries/service /opt/victronenergy/service/dbus-aggregate-batteries
bash /data/apps/dbus-serialbattery/enable.sh > /data/apps/dbus-serialbattery/startup.log 2>&1 &
bash /data/dbus-aggregate-batteries/reinstall-local.sh
sh /data/dbus-aggregate-batteries/install.sh
bash /data/apps/dbus-aggregate-batteries/reinstall-local.sh
root@einstein:~#
root@einstein:~# cat /data/rcS.local
#!/bin/bash
bash /data/apps/overlay-fs/enable.sh > /data/apps/overlay-fs/startup.log 2>&1
root@einstein:~#
root@einstein:~# cat /data/apps/dbus-serialbattery/startup.log
cat: can't open '/data/apps/dbus-serialbattery/startup.log': No such file or directory

tobio-de, 2025-09-28:
now rc.local looks like that:

#/bin/bash
bash /data/apps/dbus-serialbattery/enable.sh > /data/apps/dbus-serialbattery/startup.log 2>&1 &
ln -s /data/apps/dbus-aggregate-batteries/service /opt/victronenergy/service/dbus-aggregate-batteries
ln -s /data/dbus-kaco_blueplanet/service/ /opt/victronenergy/service/dbus-kaco_blueplanet
/data/apps/dbus-loxone-smartmeter/install.sh
ln -s /data/apps/dbus-loxone-smartmeter/service /opt/victronenergy/service/dbus-loxone-smartmeter

still not starting...

mr-manuel, 2025-09-28:
What I missed is that your `rc.local` is not executable. Make the file executable with `chmod +x /data/rc.local`. Bst would also be that you delete the irrelevant `rc.local-` (ending with a dash).

mr-manuel, 2025-09-28:
What I missed is that your `rc.local` is not executable. Make the file executable with `chmod +x /data/rc.local`. Best would also be that you delete the irrelevant `rc.local-` (ending with a dash).

mr-manuel, 2025-09-28:
What I missed is that your `rc.local` is not executable. Make the file executable with `chmod +x /data/rc.local`. Best would also be that you delete the irrelevant `rc.local~`.

tobio-de, 2025-09-28:
seams to work
thank you very much!!!

closed 2025-09-28 by mr-manuel