mr-manuel/venus-os_dbus-serialbattery #324: JK BMS with failure #119 settings invalid

MarkusPVHM, 2025-09-30:
### Describe the problem

Hi , I am using the Bluetooth method for JK BMS JK-PB2A16S20P. I can read all the data at my Pi4 Venus system, but about a few hours there occur a failure called "Serialbattery JKBMS BLE #119 Setting invalid.
Any ideas?

### Installation method

Install script from the documentation

### Driver version of the currently installed driver

2.0.20250729

### Venus OS device type

Raspberry Pi 4

### Venus OS version

V3.70-39

### Venus OS image type

Large

### BMS type

JKBMS Inverter (JKBMS PB Model)

### Cell count

16

### Battery count

1

### Connection type

Bluetooth

### USB hub

No, the Serial USB adapter is connected directly to the Venus OS device

### Config file

```ini
GNU nano 7.2                                                                                                     config.ini
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.
; NOTICE: Do not copy the whole file, but only the values/settings you want to change.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):
; MAX_BATTERY_CHARGE_CURRENT = 50.0
; MAX_BATTERY_DISCHARGE_CURRENT = 60.0
MAX_BATTERY_CHARGE_CURRENT    = 100.0
MAX_BATTERY_DISCHARGE_CURRENT = 100.0
MAX_CELL_VOLTAGE   = 3.550
FLOAT_CELL_VOLTAGE = 3.375
BLUETOOTH_BMS = Jkbms_Ble A4:C1:38:00:52:06
```

### Relevant log output

```shell
SERIAL battery connection: The activation is complete. You don't have to do anything more.

BLUETOOTH battery connection: There are a few more steps to complete activation.

    1. Add your Bluetooth BMS to the config file "/data/apps/dbus-serialbattery/config.ini".
       Check the default config file "/data/apps/dbus-serialbattery/config.default.ini" for more informations.
       If your Bluetooth BMS are nearby you can show the MAC address with "bluetoothctl devices".

    2. Make sure to disable Bluetooth in "Settings -> Bluetooth" in the remote console/GUI to prevent reconnects every minute.

    3. Run "/data/apps/dbus-serialbattery/restart.sh", if the Bluetooth BMS were not added to the "config.ini" before.

    ATTENTION!
    If you changed the default connection PIN of your BMS, then you have to pair the BMS first using OS tools like the "bluetoothctl".
    See https://wiki.debian.org/BluetoothUser#Using_bluetoothctl for more details.

CAN battery connection: There are a few more steps to complete activation.

    1. Add your CAN port to the config file "/data/apps/dbus-serialbattery/config.ini".
       Check the default config file "/data/apps/dbus-serialbattery/config.default.ini" for more informations.

    2. In the remote console/GUI, go to 'Settings -> Services -> VE.Can port -> CAN-bus profile' and select a profile with a bitrate that
       matches your BMS.

    3. Re-run "/data/apps/dbus-serialbattery/restart.sh", if the CAN port was not added to the "config.ini" before.

CUSTOM SETTINGS: If you want to add custom settings, then check the settings you want to change in "/data/apps/dbus-serialbattery/config.default.ini"
The app "dbus-serialbattery" was already added to the directory "/opt/victronenergy/service-templates" in the config file.
The app "dbus-serialbattery_gui" was already added to the directory "/opt/victronenergy/gui" in the config file.
The app "dbus-serialbattery_gui" was already added to the directory "/var/www/venus" in the config file.

Installing QML files for GUI v1...
done.

Latest version of GUIv2 web version is already downloaded.

Latest version of GUIv2 web version is already installed.

Stop all dbus-serialbattery services...

Found 1 Bluetooth BMS in the config file!

Build-in Bluetooth is used.
Stopping bluetooth: bluetoothd.
Starting bluetooth: bluetoothd.

Installing "Jkbms_Ble" with MAC address "A4:C1:38:00:52:06" as dbus-blebattery.0

No CAN port configuration found in "/data/apps/dbus-serialbattery/config.ini".
You can ignore this, if you are using only a serial connection.

#################################
# First activation instructions #
```

### Any other information that may be helpful

[image]

### Please confirm that you have completed the following steps before submitting your support request

- [x] I have carefully read all the instructions above including the descriptions.
- [x] I have tried the `nightly` build first, if I was not already using it. This means the logs I provide are from the `nightly` build.
- [x] I have rebooted at least once after I saw the issue.
- [x] I have read the complete [changelog](https://github.com/mr-manuel/dbus-serialbattery/blob/master/CHANGELOG.md).
- [x] I have provided the logfiles in the correct format with a HUMAN READABLE timestamp by piping it through `tai64nlocal`.
- [x] I have clicked all checkboxes because I did not read the instructions.

mr-manuel, 2025-09-30:
Check the documentation.

closed 2025-09-30 by mr-manuel

MarkusPVHM, 2025-09-30:
Hi,

sorry, I am not the progammer crack, where do you think I did not consider the Doc? I do not find find my mistake....

Regards Markus